awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #391: curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Out of memory: SSL: couldn't create a context: error:140A90F1:lib(20):func(169):reason(241)

Remainfox, 2020-04-02:
hello
when i use createKinesisVideoStreamSync , report error log as follow
[kvsServer]2020-04-02 14:49:19.412 [INFO][server.cpp:CreateMsg:43] msgget 【msg key :989986817, msg id is 5210119】
[kvsServer]2020-04-02 14:49:19.440 [INFO][server.cpp:stream_send_thread:855] chn:0,start stream_send_thread PID:27342
[kvsServer]2020-04-02 14:49:19.442 [INFO][server.cpp:msg_info_thread:136] chn:0,start msg_info_thread PID:27342
[kvsServer]2020-04-02 14:49:19.448 [INFO][kvs_interface.cpp:kvsObjInit:322] accessKey:ASIAXTTC7SKI5AHG7WJN
[kvsServer]2020-04-02 14:49:19.449 [INFO][kvs_interface.cpp:kvsObjInit:323] secretKey:aCnWbphGuMBbzupqaOBVA83E3HXCmUU2ZgoZbuUe
[kvsServer]2020-04-02 14:49:19.465 [INFO][kvs_interface.cpp:kvsObjInit:324] sessionToken:FwoGZXIvYXdzEEgaDJARJ60J4qL/cB/kNCL5AoVQuYyXpJFInhepy1CggZEg5z7klsUgkI1RXn95tVAfYyTYk1hYCjkIGQe01Rwb1/hZMZsuNOjb7P3upb4ivVj7AJqL0GiugsSgTlwFPh4Vtju4dKH73+3/auQpiNuKdlj17+eguV9w10/3Fizya4GJhCPoEObuk/b9sQi/hfU7ypL4IBvKGglWzvr9kyp4HtcNNpjg4UrBoI6FLAtYEzRWIKM886c1lWW05pQOcQu43N+vy3sUMLh9p1wIwlWJeZwjlaTpxRb2xycmfQJBpKfJgrGri10XVNiMRogtgFB8g7I6idViuTZDUmXF59g7z8AkdVp00ANCk1mnKOyFruHxGVKztCNVZtul1njCxFCHSx6z1JEide1gngVkrbdKWDgJdKhUBL9UXlH+a6oU1z9BZ+V+qki7NqjQKLn3WXIULOCs3WINNyZXDWmUBpfgQn+P9kcL2wC985jU0xfq/4s7M7B8ouW2SSQb6pDuxmGudwQJ270K5U0+KLaQlvQFMi1OE1kgLdkAa74H9sROl1AA4LgCqdJeXU7spjCnAvnWXtca1InwYBOtU8dQg1I=
[kvsServer]2020-04-02 14:49:19.471 [INFO][kvs_interface.cpp:kvsObjInit:325] streamName:d980288abdfd4dad3944a12ddcb81d0181b0ff31-T8400P001953003C-T8400P001953003C-cloud
[kvsServer]2020-04-02 14:49:19.471 [INFO][kvs_interface.cpp:kvsObjInit:326] region:us-west-2
[kvsServer]2020-04-02 14:49:19.471 [INFO][kvs_interface.cpp:kvsObjInit:327] expire_time:1585813062
[kvsServer]2020-04-02 14:49:19.472 [INFO][kvs_interface.cpp:kvsObjInit:328] now_time:   1585810159
[kvsServer]2020-04-02 14:49:19.482 [INFO][kvs_interface.cpp:kvsObjInit:415]  
[kvsServer]2020-04-02 14:49:19.482 [INFO][kvs_interface.cpp:kvsObjInit:416] 000 createKinesisVideoStreamSync start
[kvsServer]2020-04-02 14:49:19.483 [INFO][kvs_interface.cpp:kvsObjInit:417]  
[kvsServer]2020-04-02 14:49:19.711 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Out of memory: SSL: couldn't create a context: error:140A90F1:lib(20):func(169):reason(241)
[kvsServer]2020-04-02 14:49:19.713 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAXTTC7SKI5AHG7WJN/20200402/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=348e016517300acc287414c2fffce094f9d3e3a0e63bf09211d5136e52b9efd0
    content-length: 101
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.7.2 Linux/3.10.14__isvp_swan_1.0__ mips
[kvsServer]2020-04-02 14:49:20.265 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Out of memory: SSL: couldn't create a context: error:140A90F1:lib(20):func(169):reason(241)
[kvsServer]2020-04-02 14:49:20.508 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAXTTC7SKI5AHG7WJN/20200402/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=348e016517300acc287414c2fffce094f9d3e3a0e63bf09211d5136e52b9efd0
    content-length: 101
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.7.2 Linux/3.10.14__isvp_swan_1.0__ mips
[kvsServer]2020-04-02 14:49:21.108 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Out of memory: SSL: couldn't create a context: error:140A90F1:lib(20):func(169):reason(241)
[kvsServer]2020-04-02 14:49:21.109 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAXTTC7SKI5AHG7WJN/20200402/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=8038207522163e36200636f9c875737ad3f22608cd4b1c248b729039b969e646
    content-length: 101
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.7.2 Linux/3.10.14__isvp_swan_1.0__ mips
[kvsServer]2020-04-02 14:49:22.093 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Out of memory: SSL: couldn't create a context: error:140A90F1:lib(20):func(169):reason(241)
[kvsServer]2020-04-02 14:49:22.172 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAXTTC7SKI5AHG7WJN/20200402/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=ce03cc75078fd0f41e2b725afbb05cb01c006321252f36f2cd6c8aabe391f6a1
    content-length: 101
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.7.2 Linux/3.10.14__isvp_swan_1.0__ mips
[kvsServer]2020-04-02 14:49:23.964 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-west-2.amazonaws.com/describeStream with result Out of memory: SSL: couldn't create a context: error:140A90F1:lib(20):func(169):reason(241)
[kvsServer]2020-04-02 14:49:24.490 [INFO][kvs_interface.cpp:kvsLogCallBack:129] createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
[kvsServer]2020-04-02 14:49:24.547 [INFO][kvs_interface.cpp:kvsLogCallBack:129] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-west-2.amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAXTTC7SKI5AHG7WJN/20200402/us-west-2/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=98235450d20c384f946b92696c54a63eceb8fdb2fe4a402b9334149af070172a
    content-length: 101
    content-type: application/json
    host: kinesisvideo.us-west-2.amazonaws.com
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.7.2 Linux/3.10.14__isvp_swan_1.0__ mips
[kvsServer]2020-04-02 14:49:24.549 [INFO][kvs_interface.cpp:kvsLogCallBack:129] continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 0 Status: 0x201500001e
[kvsServer]2020-04-02 14:49:24.549 [INFO][kvs_interface.cpp:kvsStreamErrorReportFunc:219] 222 Reported streamError callback for stream handle 8576792. Upload handle 18446744073709551615. Fragment timecode in 100ns: 0. Error status: 0x1500001e
[kvsServer]2020-04-02 14:49:24.949 [ERROR][kvs_interface.cpp:kvsObjInit:422] createKinesisVideoStreamSync failed ret:0x0000000f
[kvsServer]2020-04-02 14:49:25.425 [ERROR][kvs_interface.cpp:kvsObjInit:440] kvsObjInit Failed with status 0x0000000f
[kvsServer]2020-04-02 14:49:25.425 [ERROR][server.cpp:KvsAuthentication:1361] kvsObjInit failed
[kvsServer]2020-04-02 14:49:25.431 [INFO][server.cpp:QueuePutVideoFrame:630] chn:0,in advert video frame cnt:1,mem advert video frame cnt:0
[kvsServer]2020-04-02 14:49:26.003 [INFO][server.cpp:recive_push_thread:1510] chn:0,iContralThreadRun is to be 0
[kvsServer]2020-04-02 14:49:26.005 [INFO][server.cpp:stream_send_thread:861] chn:0,iControlThreadRun is be 0
[kvsServer]2020-04-02 14:49:26.006 [INFO][server.cpp:stream_send_thread:953] chn:0,pool_packet_num:0,in_video_cnt:73,mem_video_cnt:73,send_video_cnt:73,loss_video_cnt:0
[kvsServer]2020-04-02 14:49:26.006 [INFO][server.cpp:stream_send_thread:955] chn:0,stream send thread quit success
[kvsServer]2020-04-02 14:49:26.008 [INFO][server.cpp:msg_info_thread:175] chn:0, msg thread stop,alloc_cnt:0
[kvsServer]2020-04-02 14:49:26.008 [INFO][server.cpp:doExit:375]  
[... cut ...] 00000 
[kvsServer]2020-04-02 14:49:26.010 [INFO][server.cpp:doExit:377]  
[kvsServer]2020-04-02 14:49:26.011 [INFO][server.cpp:doExit:409]  
[kvsServer]2020-04-02 14:49:26.014 [INFO][server.cpp:doExit:410] do_exit 1111111111111111 
[kvsServer]2020-04-02 14:49:26.014 [INFO][server.cpp:doExit:411]  
[kvsServer]2020-04-02 14:49:26.015 [INFO][mempool.cpp:zx_mempool_deinit:867] mempool deinit.....
[kvsServer]2020-04-02 14:49:26.017 [INFO][server.cpp:recive_push_thread:1596] thread quit succes fd is 5

disa6302, 2020-04-02:
Hello @Remainfox , 

From the logs, it is not clear what might be causing this failure. Can you try running strace or equivalent so that we get a log of the internal syscall that might be causing it? 

Also, how frequently do you see this? Thanks!

MushMal, 2020-04-02:
@Remainfox can you also provide some more information about your platform and the settings you use? It's not clear as to why the SSL operation runs out of memory causing the API call failure.

MushMal, 2020-04-15:
I am going to resolve this for now due to staleness. Please use fresh install and attempt to repro. Include more details and re-open to track better

closed 2020-04-15 by MushMal