awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #394: Gstreamer Plugin libgstkvsink.so encounter ' Failed to hex decode the codec private data with:' error.

timetxt, 2020-04-04:
With the latest commit, I encoutered the following error in Ubuntu 18.04 TLS which is installed in a Desktop Host(Inter NUC).  This may also affect in Mingw64 as well.

OS info:

Linux desktop 5.3.0-45-generic #37~18.04.1-Ubuntu SMP Fri Mar 27 15:58:10 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Command:

GST_DEBUG=3 gst-launch-1.0 rtspsrc do-rtcp=TRUE location=rtspt://<rtsp_camera_ip>/live short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="<stream_name>" access-key=********* secret-key=********** aws-region="ap-southeast-2"

IP Camera:
https://www.amazon.com.au/gp/product/B07BF7XP6Z/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

$ git status
On branch master
Your branch is up to date with 'origin/master'.

$ git log
commit 3ae1b0224b8b3e8a6aa32b5dba8b223f158b36c0 (HEAD -> master, origin/master, origin/HEAD)
Author: chehefen <chehefen@amazon.com>
Date:   Tue Mar 31 16:44:04 2020 -0700

    add pcm support to kvssink

commit 21711fb8453631eb567f39f9424fb8fc4b8a95cf
Author: chehefen <chehefen@amazon.com>
Date:   Tue Mar 31 16:44:04 2020 -0700

    move frame ordering to producer. Apply timecode scale to frames pts before putFrame to avoid fragment overlap issue.

Error:

Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtspt://<Camera_IP>/live
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
0:00:00.902074595 13225 0x559630c7d0f0 WARN                     bin gstbin.c:2787:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
**[ERROR] [04-04-2020 11:34:22:525.218 GMT] Failed to hex decode the codec private data with: 2**
0:00:01.027624651 13225 0x7f03cc0b3800 WARN                 kvssink gstkvssink.cpp:947:gst_kvs_sink_handle_sink_event:<kvssink0> error: Failed to start stream
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: GStreamer encountered a general stream error.
Additional debug info:
/home/*****/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp(947): gst_kvs_sink_handle_sink_event (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
Failed to start stream
Execution ended after 0:00:00.125615070
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
0:00:01.029137748 13225 0x559630c2f320 WARN                 rtspsrc gstrtspsrc.c:5919:gst_rtsp_src_receive_response:<rtspsrc0> receive interrupted
Setting pipeline to NULL ...
0:00:01.029187055 13225 0x559630c2f320 WARN                 rtspsrc gstrtspsrc.c:8246:gst_rtspsrc_pause:<rtspsrc0> PAUSE interrupted
0:00:01.031148617 13225 0x559630c2f320 WARN                 rtspsrc gstrtspsrc.c:5999:gst_rtspsrc_try_send:<rtspsrc0> send interrupted
0:00:01.031192146 13225 0x559630c2f320 WARN                 rtspsrc gstrtspsrc.c:7673:gst_rtspsrc_close:<rtspsrc0> TEARDOWN interrupted
Freeing pipeline ...
[INFO ] [04-04-2020 11:34:22:530.780 GMT] Freeing Kinesis Video Stream myfirstkvs
[INFO ] [04-04-2020 11:34:22:530.860 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[WARN ] [04-04-2020 11:34:22:530.897 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty

Previous Commit which was working:
$ git log
commit 2623b87404b7e26a11012c29e448c3ccfc1cc0e0 (HEAD -> master, origin/master, origin/HEAD)
Author: chehefen <chehefen@amazon.com>
Date:   Fri Mar 27 13:47:36 2020 -0700

    update PIC and c producer hook

commit 56b0774c66cc3c596356b1a357cd30a43f844072
Author: chehefen <chehefen@amazon.com>
Date:   Wed Mar 25 12:08:05 2020 -0700

    fix issue when sample upload a file, producer start time is 0.

disa6302, 2020-04-04:
Hello @timetxt , 

Is this 100% reproducible?

timetxt, 2020-04-05:
Yes. It is 100% reproducable. 
Just download the latest commit again into a different folder and built the libgstkvssink.so again.  same error occurred. 

0:00:00.630402094 18363 0x7f7f900041e0 INFO               GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event application/x-rtp, media=(string)video, payload=(int)96, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)640033, sprop-parameter-sets=(string)"Z2QAM6wbGoAwAEB/wFuAgICgAAB9AAATiB0MAH0AAAjKBd5caGAD6AAARlAu8uFAAA\=\=\,aO44MAA\=", a-packetization-supported=(string)DH, a-rtppayload-supported=(string)DH, a-framerate=(string)20.000000, a-recvonly=(string)"", ssrc=(uint)1735451360, clock-base=(uint)58080, seqnum-base=(uint)58080, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
0:00:00.630513582 18363 0x7f7f900041e0 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<rtpptdemux0> adding pad 'src_96'
0:00:00.630710299 18363 0x7f7f900041e0 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link rtpptdemux0:src_96 and recv_rtp_src_0_1735451360_96:proxypad8
0:00:00.630763945 18363 0x7f7f900041e0 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked rtpptdemux0:src_96 and recv_rtp_src_0_1735451360_96:proxypad8, successful
0:00:00.630795501 18363 0x7f7f900041e0 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:00.630903533 18363 0x7f7f900041e0 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<manager> adding pad 'recv_rtp_src_0_1735451360_96'
0:00:00.631144123 18363 0x7f7f900041e0 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link manager:recv_rtp_src_0_1735451360_96 and recv_rtp_src_0_1735451360_96:proxypad9
0:00:00.631209551 18363 0x7f7f900041e0 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked manager:recv_rtp_src_0_1735451360_96 and recv_rtp_src_0_1735451360_96:proxypad9, successful
0:00:00.631244240 18363 0x7f7f900041e0 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:00.631369554 18363 0x7f7f900041e0 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<rtspsrc0> adding pad 'recv_rtp_src_0_1735451360_96'
0:00:00.631441071 18363 0x7f7f900041e0 INFO            GST_PIPELINE grammar.y:524:gst_parse_found_pad: trying delayed linking one pad some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
0:00:00.631491275 18363 0x7f7f900041e0 INFO        GST_ELEMENT_PADS gstutils.c:1774:gst_element_link_pads_full: trying to link element rtspsrc0:(any) to element rtph264depay0:(any)
0:00:00.631552890 18363 0x7f7f900041e0 INFO                GST_PADS gstutils.c:1035:gst_pad_check_link: trying to link rtspsrc0:recv_rtp_src_1_321325067_8 and rtph264depay0:sink
0:00:00.631734962 18363 0x7f7f900041e0 INFO        GST_ELEMENT_PADS gstutils.c:1227:gst_element_get_compatible_pad:<rtph264depay0> Could not find a compatible pad to link to rtspsrc0:recv_rtp_src_1_321325067_8
0:00:00.631792683 18363 0x7f7f900041e0 INFO                GST_PADS gstutils.c:1035:gst_pad_check_link: trying to link rtspsrc0:recv_rtp_src_0_1735451360_96 and rtph264depay0:sink
0:00:00.631903359 18363 0x7f7f900041e0 INFO                GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: rtspsrc0 and rtph264depay0 in same bin, no need for ghost pads
0:00:00.631963375 18363 0x7f7f900041e0 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link rtspsrc0:recv_rtp_src_0_1735451360_96 and rtph264depay0:sink
0:00:00.632065547 18363 0x7f7f900041e0 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked rtspsrc0:recv_rtp_src_0_1735451360_96 and rtph264depay0:sink, successful
0:00:00.632102516 18363 0x7f7f900041e0 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:00.632740016 18363 0x7f7f900041e0 INFO               GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)01640033ffe1003167640033ac1b1a803000407fc05b808080a000007d000013881d0c007d000008ca05de5c686003e8000046502ef2e1400001000568ee383000, level=(string)5.1, profile=(string)high
0:00:00.632976221 18363 0x7f7f900041e0 INFO               h264parse gsth264parse.c:1793:gst_h264_parse_update_src_caps:<h264parse0> resolution changed 3072x2048
0:00:00.633020727 18363 0x7f7f900041e0 INFO               h264parse gsth264parse.c:1817:gst_h264_parse_update_src_caps:<h264parse0> pixel aspect ratio has been changed 1/1
0:00:00.633065754 18363 0x7f7f900041e0 INFO               h264parse gsth264parse.c:1853:gst_h264_parse_update_src_caps:<h264parse0> PAR 1/1
0:00:00.633133723 18363 0x7f7f900041e0 INFO               baseparse gstbaseparse.c:4004:gst_base_parse_set_latency:<h264parse0> min/max latency 0:00:00.050000000, 0:00:00.050000000
0:00:00.633342707 18363 0x7f7f900041e0 INFO               GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)01640033ffe1003167640033ac1b1a803000407fc05b808080a000007d000013881d0c007d000008ca05de5c686003e8000046502ef2e1400001000568ee383000, level=(string)5.1, profile=(string)high, pixel-aspect-ratio=(fraction)1/1, width=(int)3072, height=(int)2048, framerate=(fraction)20/1, interlace-mode=(string)progres [... cut ...]
0:00:00.634681416 18363 0x555a54543a00 INFO              GST_STATES gstelement.c:2676:gst_element_continue_state:<h264parse0> completed state change to PAUSED
0:00:00.634710387 18363 0x555a54543a00 INFO              GST_STATES gstelement.c:2579:_priv_gst_element_state_changed:<h264parse0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:00:00.634748153 18363 0x555a54543a00 INFO              GST_STATES gstbin.c:2954:gst_bin_change_state_func:<pipeline0> child 'h264parse0' changed state to 3(PAUSED) successfully

MushMal, 2020-04-05:
@timetxt the changes that were pushed in are irrelevant to this issue. I really don't know what's going on yet and we weren't able to reproduce this yet. It seems that the CPD (Codec Private Data) that's passed in fails to be decoded by the hexDecode PIC API. I see from the log that the following string is being passed in: "01640033ffe1003167640033ac1b1a803000407fc05b808080a000007d000013881d0c007d000008ca05de5c686003e8000046502ef2e1400001000568ee383000"

I just threw together a small test in ProducerFunctionalityTest.cpp to validate the decode functionality with the latest alignment handling:

```
TEST_F(ProducerFunctionalityTest, simpleHexDecodeTest) {
    PCHAR encoded = (PCHAR) "01640033ffe1003167640033ac1b1a803000407fc05b808080a000007d000013881d0c007d000008ca05de5c686003e8000046502ef2e1400001000568ee383000";
    UINT32 size;
    BYTE buffer[10000];

    EXPECT_EQ(STATUS_SUCCESS, hexDecode(encoded, STRLEN(encoded), NULL, &size));
    EXPECT_EQ(STATUS_SUCCESS, hexDecode(encoded, STRLEN(encoded), buffer, &size));

    EXPECT_EQ(STATUS_SUCCESS, hexDecode(encoded, 0, NULL, &size));
    EXPECT_EQ(STATUS_SUCCESS, hexDecode(encoded, 0, buffer, &size));
}
```

This passes fine.

Can I ask you the following:

* How are you building the dependencies? Are you running install-script or you are pulling down the dependencies using min-install-script?
* Has anything in the RTSP source changed? Firmware update? Change in networking?
* Could you also add this test in and compile and run it with `./producer_test --gtest_filter="ProducerFunctionalityTest.simpleHexDecodeTest"`
* Could you run your pipeline with fakesource or another RTSP source https://www.wowza.com/html/mobile.html?

Thanks in advance.

timetxt, 2020-04-05:
How are you building the dependencies? Are you running install-script or you are pulling down the dependencies using min-install-script?
--- I installed ubuntu into my nuc as a fresh start. So I just installed dependencies by following the apt-get part in Dockerfile in the SDK docker demo. Then I just install with ./min-install-script
Has anything in the RTSP source changed? Firmware update? Change in networking?
--- No. I got this IP camera three days ago and it is used to testing only and it in my home LAN network.
Could you also add this test in and compile and run it with ./producer_test --gtest_filter="ProducerFunctionalityTest.simpleHexDecodeTest"
--- not sure how to do it exactly. but I will try and update it later. Not a developer.
Could you run your pipeline with fakesource or another RTSP source https://www.wowza.com/html/mobile.html?
----Sure, will update later

timetxt, 2020-04-05:
Big Bunner, failed with bad plugin. Successful with good plugin. 

Error debug logs:
0:00:09.762555187 19163 0x7f8f58074c00 INFO                GST_PADS gstutils.c:1035:gst_pad_check_link: trying to link rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink
0:00:09.762575132 19163 0x7f8f58041450 INFO                GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: rtspsrc0 and rtph264depay0 in same bin, no need for ghost pads
0:00:09.762622966 19163 0x7f8f58074c00 INFO                GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: rtspsrc0 and rtph264depay0 in same bin, no need for ghost pads
0:00:09.762637073 19163 0x7f8f58041450 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink
0:00:09.762664320 19163 0x7f8f58074c00 INFO                GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink
0:00:09.762748285 19163 0x7f8f58041450 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink, successful
0:00:09.762762638 19163 0x7f8f58074c00 INFO                GST_PADS gstpad.c:2586:gst_pad_link_full: linked rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink, successful
0:00:09.762790977 19163 0x7f8f58041450 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
0:00:09.762820795 19163 0x7f8f58074c00 INFO               GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event

(gst-launch-1.0:19163): GLib-GObject-WARNING **: 21:09:16.684: ../../../../gobject/gsignal.c:2641: instance '0x55f8a14d0220' has no handler with id '4'

(gst-launch-1.0:19163): GLib-GObject-WARNING **: 21:09:16.684: ../../../../gobject/gsignal.c:2641: instance '0x55f8a14d0220' has no handler with id '3'
0:00:09.763268921 19163 0x7f8f58041450 INFO                GST_PADS gstpad.c:4232:gst_pad_peer_query:<rtspsrc0:recv_rtp_src_0_403406234_96> pad has no peer
0:00:09.763590214 19163 0x7f8f58074c00 INFO               GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline
0:00:09.764011299 19163 0x7f8f58041450 INFO                GST_PADS gstpad.c:4232:gst_pad_peer_query:<rtspsrc0:recv_rtp_src_0_403406234_96> pad has no peer
0:00:09.764124265 19163 0x7f8f58041450 INFO                    task gsttask.c:316:gst_task_func:<rtpjitterbuffer0:src> Task going to paused
0:00:09.764133662 19163 0x7f8f58074c00 INFO               h264parse gsth264parse.c:1793:gst_h264_parse_update_src_caps:<h264parse0> resolution changed 240x160
0:00:09.764244723 19163 0x7f8f58074c00 INFO               baseparse gstbaseparse.c:4004:gst_base_parse_set_latency:<h264parse0> min/max latency 0:00:00.041666666, 0:00:00.041666666
0:00:09.764382325 19163 0x7f8f58074c00 INFO               GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline, width=(int)240, height=(int)160, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
0:00:09.764490962 19163 0x7f8f58074c00 INFO                 kvssink gstkvssink.cpp:901:gst_kvs_sink_handle_sink_event: structure is video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline, width=(int)240, height=(int)160, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true;
**[ERROR] [05-04-2020 11:09:16:685.979 GMT] Failed to hex decode the codec private data with: 2**
0:00:09.764660185 19163 0x7f8f58074c00 WARN                 kvssink gstkvssink.cpp:947:gst_kvs_sink_handle_sink_event:<kvssink0> error: Failed to start stream
0:00:09.764854476 19163 0x7f8f58074c00 INFO        GST_ERROR_SYSTEM gstelement.c:2145:gst_element_message_full_with_details:<kvssink0> posting message: GStreamer encountered a general stream error.
0:00:09.764937524 19163 0x7f8f58074c00 INFO        GST_ERROR_SYSTEM gstelement.c:2172:gst_element_message_full_with_details:<kvssink0> posted error message: GStreamer encountered a general stream error.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: GStreamer encountered a general stream error.
0:00:09.765119221 19163 0x7f8f58074c00 INFO                 kvssink gstkvssink.cpp:901:gst_kvs_sink_handle_sink_event: structure is video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline, width=(int)240, height=(int)160, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true;
Additional debug info:
[... cut ...]
        >> Available storage byte size: 134216647
        >> Allocated storage byte size: 1081
        >> Total view allocation byte size: 144080
        >> Total streams frame rate (fps): 25
        >> Total streams transfer rate (bps): 33554432 (32768 Kbps)
        >> Current view duration (ms): 0
        >> Overall view duration (ms): 0
        >> Current view byte size: 1033
        >> Overall view byte size: 1033
        >> Current frame rate (fps): 25
        >> Current transfer rate (bps): 33554432 (32768 Kbps)

timetxt, 2020-04-05:
Could you also add this test in and compile and run it with ./producer_test --gtest_filter="ProducerFunctionalityTest.simpleHexDecodeTest"

Not sure how to do it. Should I add it into 'amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeLists.txt' file? If yes, where should I put it? If not, could you point out the steps. Thanks

disa6302, 2020-04-05:
@timetxt , 

1. Add the code @MushMal posted in this file: `kinesis-video-producer/tst/ProducerFunctionalityTest.cpp`

From the native-build directory, do the following: 
2. Run the following command: `./min-install-script --build_test`. This will build all the unit tests for you. 

3. Finally, run this: `./producer_test --gtest_filter="ProducerFunctionalityTest.simpleHexDecodeTest"`

disa6302, 2020-04-05:
> Big Bunner, failed with bad plugin. Successful with good plugin.
> 
> Error debug logs:
> 0:00:09.762555187 19163 0x7f8f58074c00 INFO GST_PADS gstutils.c:1035:gst_pad_check_link: trying to link rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink
> 0:00:09.762575132 19163 0x7f8f58041450 INFO GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: rtspsrc0 and rtph264depay0 in same bin, no need for ghost pads
> 0:00:09.762622966 19163 0x7f8f58074c00 INFO GST_PADS gstutils.c:1588:prepare_link_maybe_ghosting: rtspsrc0 and rtph264depay0 in same bin, no need for ghost pads
> 0:00:09.762637073 19163 0x7f8f58041450 INFO GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink
> 0:00:09.762664320 19163 0x7f8f58074c00 INFO GST_PADS gstpad.c:2378:gst_pad_link_prepare: trying to link rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink
> 0:00:09.762748285 19163 0x7f8f58041450 INFO GST_PADS gstpad.c:2586:gst_pad_link_full: linked rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink, successful
> 0:00:09.762762638 19163 0x7f8f58074c00 INFO GST_PADS gstpad.c:2586:gst_pad_link_full: linked rtspsrc0:recv_rtp_src_1_1587901200_97 and rtph264depay0:sink, successful
> 0:00:09.762790977 19163 0x7f8f58041450 INFO GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
> 0:00:09.762820795 19163 0x7f8f58074c00 INFO GST_EVENT gstevent.c:1517:gst_event_new_reconfigure: creating reconfigure event
> 
> (gst-launch-1.0:19163): GLib-GObject-WARNING **: 21:09:16.684: ../../../../gobject/gsignal.c:2641: instance '0x55f8a14d0220' has no handler with id '4'
> 
> (gst-launch-1.0:19163): GLib-GObject-WARNING **: 21:09:16.684: ../../../../gobject/gsignal.c:2641: instance '0x55f8a14d0220' has no handler with id '3'
> 0:00:09.763268921 19163 0x7f8f58041450 INFO GST_PADS gstpad.c:4232:gst_pad_peer_query:rtspsrc0:recv_rtp_src_0_403406234_96 pad has no peer
> 0:00:09.763590214 19163 0x7f8f58074c00 INFO GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline
> 0:00:09.764011299 19163 0x7f8f58041450 INFO GST_PADS gstpad.c:4232:gst_pad_peer_query:rtspsrc0:recv_rtp_src_0_403406234_96 pad has no peer
> 0:00:09.764124265 19163 0x7f8f58041450 INFO task gsttask.c:316:gst_task_func:rtpjitterbuffer0:src Task going to paused
> 0:00:09.764133662 19163 0x7f8f58074c00 INFO h264parse gsth264parse.c:1793:gst_h264_parse_update_src_caps: resolution changed 240x160
> 0:00:09.764244723 19163 0x7f8f58074c00 INFO baseparse gstbaseparse.c:4004:gst_base_parse_set_latency: min/max latency 0:00:00.041666666, 0:00:00.041666666
> 0:00:09.764382325 19163 0x7f8f58074c00 INFO GST_EVENT gstevent.c:814:gst_event_new_caps: creating caps event video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline, width=(int)240, height=(int)160, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
> 0:00:09.764490962 19163 0x7f8f58074c00 INFO kvssink gstkvssink.cpp:901:gst_kvs_sink_handle_sink_event: structure is video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline, width=(int)240, height=(int)160, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true;
> **[ERROR] [05-04-2020 11:09:16:685.979 GMT] Failed to hex decode the codec private data with: 2**
> 0:00:09.764660185 19163 0x7f8f58074c00 WARN kvssink gstkvssink.cpp:947:gst_kvs_sink_handle_sink_event: error: Failed to start stream
> 0:00:09.764854476 19163 0x7f8f58074c00 INFO GST_ERROR_SYSTEM gstelement.c:2145:gst_element_message_full_with_details: posting message: GStreamer encountered a general stream error.
> 0:00:09.764937524 19163 0x7f8f58074c00 INFO GST_ERROR_SYSTEM gstelement.c:2172:gst_element_message_full_with_details: posted error message: GStreamer encountered a general stream error.
> ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: GStreamer encountered a general stream error.
> 0:00:09.765119221 19163 0x7f8f58074c00 INFO kvssink gstkvssink.cpp:901:gst_kvs_sink_handle_sink_event: structure is video/x-h264, stream-format=(string)avc, alignment=(string)au, codec_data=(buffer)0142c01effe100156742c01ed903c56840000003004000000c03c58b9201000468cb8cb2, level=(string)3, profile=(string)constrained-baseline, width=(int)240, height=(int)160, framerate=(fraction)24/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true;
> Additional debug info:
> /home/User/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp(947): gst_kvs_sink_handle_sink_event (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
> Failed to start stream
> Execution ended after 0:00:07.776320912
[... cut ...]
> >> Allocated storage byte size: 1081
> >> Total view allocation byte size: 144080
> >> Total streams frame rate (fps): 25
> >> Total streams transfer rate (bps): 33554432 (32768 Kbps)
> >> Current view duration (ms): 0
> >> Overall view duration (ms): 0
> >> Current view byte size: 1033
> >> Overall view byte size: 1033
> >> Current frame rate (fps): 25
> >> Current transfer rate (bps): 33554432 (32768 Kbps)

@timetxt , does this mean you see the issue with the good plugin and not with the bad plugin?

disa6302, 2020-04-05:
@timetxt ,

Thank you for the logs. We are able to reproduce the issue on Ubuntu 18.04 and working to see whats causing it. Will get back to you when we have more details on the issue.  Thank you!

a428tm, 2020-04-05:
@timetxt I ran into the same issue as well. I got around it by installing previous version that was saved on my other machine (which I installed 2 weeks ago - 3/25). Not sure if this is an available option for you or not

MushMal, 2020-04-06:
Fixed in the latest commit. Closing.

closed 2020-04-06 by MushMal