awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #395: Unable to send data to KVSSink in Gstreamer on Nvidia Jetson Nano device/Ubuntu

JasonTrue, 2020-04-05:
I've got a relatively vanilla Nvidia Jetson Nano running their Ubuntu distribution plus Kinesis Video dependencies and I'm trying what is a fairly minimal test feed that I've successfully used on a Mac or a Raspberry Pi.

On the Nano I get "Failed to hex decode the codec private data with: 2" (or sometimes ": 3") when I run this relatively simple feed. I've done less elaborate ones without this ceremony and basically get the same error. I'm not quite sure what's different about the architecture, but it seems like a bug.

The device as configured doesn't have an onboard camera; the plan is to decode an RTMP SRC generally consumed from a GoPro camera and convert it to a h264 aac feed. But because it appears to have similar symptoms without the network and protocol complexity, I'm treating this pipeline as a more minimal test case. It's blocking my ability to adopt Kinesis video in our case.

Is there something I'm missing in terms of specifying the CPD?

```
$ GST_DEBUG=4 gst-launch-1.0  videotestsrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! h264parse ! kvssink stream-name=test1234 storage-size=120 name=sink storage-size=128 aws-region=ap-northeast-1 access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_ACCESS_SECRET_KEY 
0:00:00.000131617  9166   0x5564d71400 INFO                GST_INIT gst.c:586:init_pre: Initializing GStreamer Core Library version 1.14.5
0:00:00.000203806  9166   0x5564d71400 INFO                GST_INIT gst.c:587:init_pre: Using library installed in /usr/lib/aarch64-linux-gnu
0:00:00.000237036  9166   0x5564d71400 INFO                GST_INIT gst.c:607:init_pre: Linux nvidia-streamer 4.9.140-tegra #1 SMP PREEMPT Mon Dec 9 22:47:42 PST 2019 aarch64
0:00:00.000849026  9166   0x5564d71400 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.002246396  9166   0x5564d71400 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.003019901  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize: registering 0 static plugins
0:00:00.003312614  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static: registered static plugin "staticelements"
0:00:00.003357511  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:227:gst_plugin_register_static: added static plugin "staticelements", result: 1
0:00:00.003416366  9166   0x5564d71400 INFO            GST_REGISTRY gstregistry.c:1727:ensure_current_registry: reading registry cache: /home/jason/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.042596059  9166   0x5564d71400 INFO            GST_REGISTRY gstregistrybinary.c:621:priv_gst_registry_binary_read_cache: loaded /home/jason/.cache/gstreamer-1.0/registry.aarch64.bin in 0.039106 seconds
0:00:00.042757364  9166   0x5564d71400 INFO            GST_REGISTRY gstregistry.c:1583:scan_and_update_registry: Validating plugins from registry cache: /home/jason/.cache/gstreamer-1.0/registry.aarch64.bin
0:00:00.061535268  9166   0x5564d71400 INFO            GST_REGISTRY gstregistry.c:1685:scan_and_update_registry: Registry cache has not changed
0:00:00.061579644  9166   0x5564d71400 INFO            GST_REGISTRY gstregistry.c:1762:ensure_current_registry: registry reading and updating done, result = 1
0:00:00.061599436  9166   0x5564d71400 INFO                GST_INIT gst.c:807:init_post: GLib runtime version: 2.56.4
0:00:00.061618238  9166   0x5564d71400 INFO                GST_INIT gst.c:809:init_post: GLib headers version: 2.56.4
0:00:00.061631103  9166   0x5564d71400 INFO                GST_INIT gst.c:810:init_post: initialized GStreamer successfully
0:00:00.061732616  9166   0x5564d71400 INFO            GST_PIPELINE gstparse.c:337:gst_parse_launch_full: parsing pipeline description 'videotestsrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au,width=640,height=480,framerate=30/1 ! h264parse ! kvssink stream-name=test1234 storage-size=120 name=sink storage-size=128 aws-region=ap-northeast-1 access-key=REDACTED secret-key=REDACTED '
0:00:00.063459315  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideotestsrc.so" loaded
0:00:00.063511659  9166   0x5564d71400 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "videotestsrc"
0:00:00.064020784  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseSrc@0x5564fc2f20> adding pad 'src'
0:00:00.064591628  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideoconvert.so" loaded
0:00:00.064626889  9166   0x5564d71400 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "videoconvert"
0:00:00.065101064  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x5564fb78d0> adding pad 'sink'
0:00:00.065150596  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseTransform@0x5564fb78d0> adding pad 'src'
0:00:00.067246000  9166   0x5564d71400 INFO                 x264enc gstx264enc.c:2956:plugin_init: linked against x264 build: 152
0:00:00.067646581  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstx264.so" loaded
0:00:00.067682310  9166   0x5564d71400 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "x264enc"
0:00:00.067988983  9166   0x5564d71400 INFO                 x264enc gstx264enc.c:227:gst_x264_enc_add_x264_chroma_format: 8-bit depth supported
0:00:00.068021848  9166   0x5564d71400 INFO                 x264enc gstx264enc.c:252:gst_x264_enc_add_x264_chroma_format: 10-bit depth supported
0:00:00.068184872  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@0x5564fc5b20> adding pad 'sink'
0:00:00.068237373  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstVideoEncoder@0x5564fc5b20> adding pad 'src'
0:00:00.069279580  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstvideoparsersbad.so" loaded
0:00:00.069317289  9166   0x5564d71400 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "h264parse"
0:00:00.069498178  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x5564fc9af0> adding pad 'sink'
0:00:00.069550314  9166   0x5564d71400 INFO        GST_ELEMENT_PADS gstelement.c:670:gst_element_add_pad:<GstBaseParse@0x5564fc9af0> adding pad 'src'
0:00:00.069617607  9166   0x5564d71400 INFO               baseparse gstbaseparse.c:3961:gst_base_parse_set_pts_interpolation:<GstH264Parse@0x5564fc9af0> PTS interpolation: no
0:00:00.099584419  9166   0x5564d71400 INFO      GST_PLUGIN_LOADING gstplugin.c:901:_priv_gst_plugin_load_file_for_registry: plugin "/home/jason/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/libgstkvssink.so" loaded
0:00:00.099674160  9166   0x5564d71400 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "kvssink"
0:00:00.100210316  9166   0x5564d71400 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:361:gst_element_factory_create: creating element "pipeline"
0:00:00.100457821  9166   0x5564d71400 INFO            GST_PIPELINE grammar.y:652:gst_parse_perform_link: linking some pad of GstVideoTestSrc named videotestsrc0 to some pad of GstVideoConvert named videoconvert0 (0/0) with caps "video/x-raw, width=(int)640, height=(int)480, framerate=(fraction)30/1"
[... cut ...]
0:00:10.835029922  9166   0x5564d71400 INFO         GST_REFCOUNTING gstelement.c:3328:gst_element_finalize:<pipeline0> 0x55650181b0 finalize
0:00:10.835228937  9166   0x5564d71400 INFO         GST_REFCOUNTING gstelement.c:3334:gst_element_finalize:<pipeline0> 0x55650181b0 finalize parent
0:00:10.835425190  9166   0x5564d71400 INFO                GST_INIT gst.c:1083:gst_deinit: deinitializing GStreamer
0:00:10.849385660  9166   0x5564d71400 INFO                GST_INIT gst.c:1231:gst_deinit: deinitialized GStreamer

```

MushMal, 2020-04-05:
Related to issue #394 

I tried the same CPD in the test I suggested and it works just fine. Perhaps the length is invalid.

Could I ask you to print out the content of the CPD in the error message so we can see what's actually being fed to the function to understand why it fails?:

in KinesisVideoStream.cpp LN 69 -> change the line to 

LOG_ERROR("Failed to get the size of the buffer for hex decoding the codec private data with: " << status << " hex CPD passed in is: " << pStrCpd);

disa6302, 2020-04-05:
@JasonTrue ,

Thank you for the logs. We are able to reproduce the issue on Ubuntu 18.04 and working to see whats causing it. Will get back to you when we have more details on the issue. 

Meanwhile, it would still be great if you could make the changes @MushMal suggested to get more details.

JasonTrue, 2020-04-06:
Thank you for your response

I made this change in `kinesis-video-producer/src/KinesisVideoStream.cpp b/kinesis-video-producer/src/KinesisVideoStream.cpp` then did `make clean` and re-ran ./min-install-script.

Looking at the revised logging output, I  don't see a change in the output, but it looks like the message you asked for wasn't present in the above output either.

That being said, I have seen that error message in some pipelines, so I reran one that does produce that logging data.

```
gst-launch-1.0 -m -v videotestsrc ! 'video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)30/1' ! x264enc ! h264parse ! kvssink stream-name=$STREAM_NAME storage-size=128 aws-region=ap-northeast-1 access-key=$AWS_ACCESS_KEY_ID secret-key=$AWS_ACCESS_SECRET_KEY

Got message #44 from pad "x264enc0:sink" (property-notify): GstMessagePropertyNotify, property-name=(string)caps, property-value=(GstCaps)"video/x-raw\,\ format\=\(string\)I420\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ framerate\=\(fraction\)30/1\,\ multiview-mode\=\(string\)mono\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ interlace-mode\=\(string\)progressive";
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
Got message #45 from pad "capsfilter0:sink" (property-notify): GstMessagePropertyNotify, property-name=(string)caps, property-value=(GstCaps)"video/x-raw\,\ format\=\(string\)I420\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ framerate\=\(fraction\)30/1\,\ multiview-mode\=\(string\)mono\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ interlace-mode\=\(string\)progressive";
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)640, height=(int)480, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
Got message #50 from pad "x264enc0:src" (property-notify): [ERROR] [06-04-2020 

--- AFFECTED LOGGING OUTPUT ----
01:25:19:302.446 GMT] Failed to get the size of the buffer for hex decoding the codec private data with: 3 hex CPD passed in is: 
GstMessagePropertyNotify, property-name=(string)caps, property-value=(GstCaps)"video/x-h264\,\ codec_data\=\(buffer\)0164001effe1001d6764001eacd940a03db016a0c020b4a0000003002000000791e2c5b2c001000568ebecb22c\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au\,\ level\=\(string\)3\,\ profile\=\(string\)high\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ framerate\=\(fraction\)30/1\,\ interlace-mode\=\(string\)progressive\,\ colorimetry\=\(string\)bt601\,\ chroma-site\=\(string\)jpeg\,\ multiview-mode\=\(string\)mono\,\ multiview-flags\=\(GstVideoMultiviewFlagsSet\)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono";
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)0164001effe1001d6764001eacd940a03db016a0c020b4a0000003002000000791e2c5b2c001000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
Got message #52 from pad "h264parse0:src" (property-notify): [DEBUG] [06-04-2020 01:25:19:303.211 GMT] Dropping frame with flag %u1088
GstMessagePropertyNotify, property-name=(string)caps, property-value=(GstCaps)"video/x-h264\,\ codec_data\=\(buffer\)0164001effe1001d6764001eacd940a03db016a0c020b4a0000003002000000791e2c5b2c001000568ebecb22c\,\ stream-format\=\(string\)avc\,\ alignment\=\(string\)au\,\ level\=\(string\)3\,\ profile\=\(string\)high\,\ width\=\(int\)640\,\ height\=\(int\)480\,\ pixel-aspect-ratio\=\(fraction\)1/1\,\ framerate\=\(fraction\)30/1\,\ interlace-mode\=\(string\)progressive\,\ colorimetry\=\(string\)bt601\,\ chroma-site\=\(string\)jpeg\,\ multiview-mode\=\(string\)mono\,\ multiview-flags\=\(GstVideoMultiviewFlagsSet\)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono\,\ chroma-format\=\(string\)4:2:0\,\ bit-depth-luma\=\(uint\)8\,\ bit-depth-chroma\=\(uint\)8\,\ parsed\=\(boolean\)true";
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, codec_data=(buffer)0164001effe1001d6764001eacd940a03db016a0c020b4a0000003002000000791e2c5b2c001000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)3, profile=(string)high, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
Got message #53 from element "kvssink0" (error): GstMessageError, gerror=(GError)NULL, debug=(string)"/home/jason/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp\(947\):\ gst_kvs_sink_handle_sink_event\ \(\):\ /GstPipeline:pipeline0/GstKvsSink:kvssink0:\012Failed\ to\ start\ stream";
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: GStreamer encountered a general stream error.
Additional debug info:

```

MushMal, 2020-04-06:
Fixed in the latest commit. Closing.

closed 2020-04-06 by MushMal

JasonTrue, 2020-04-06:
Verified that this use case works for me. I'll do some further investigation on my RTMPSRC issue and, if necessary, file a separate issue. Thanks so much!

rgroleau, 2020-04-22:
I had the same error on my yocto/poky platform, the "Failed to get the size of the buffer for hex decoding the codec private data", which was preventing me from getting a step further in the code.

Thanks to this fix from @MushMal the commit: 4de9bf4eab1d80eb74af4980985b0a9dfbefed28, that specific problem went away, but it came back in the next commit by @chehefen in the commit: 824de900c2d4b42a403319ace5de9d68ec88b171 which seems to precisely undo the fix.

Was that intended? Should I use a specific commit (and loose the awesome CMake improvements from the commits after that, or make by own branch or patches to get the right combo)?

MushMal, 2020-04-22:
> I had the same error on my yocto/poky platform, the "Failed to get the size of the buffer for hex decoding the codec private data", which was preventing me from getting a step further in the code.
> 
> Thanks to this fix from @MushMal in commit [4de9bf4](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/commit/4de9bf4eab1d80eb74af4980985b0a9dfbefed28), that specific problem went away, but it came back in the next commit [824de90](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/commit/824de900c2d4b42a403319ace5de9d68ec88b171) by @chehefen which seems to precisely undo the fix.
> 
> Was that intended? Should I use a specific commit (and loose the awesome CMake improvements from the commits after that, or make my own branch/fork/rebase or even patches get the right combo)?

Oops/... Let us take a look into it ASAP. This is odd, @chehefen tried to C++ify the API this is the reason he removed it.

chehefen, 2020-04-22:
> I had the same error on my yocto/poky platform, the "Failed to get the size of the buffer for hex decoding the codec private data", which was preventing me from getting a step further in the code. [...]

If you look at the api here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/KinesisVideoStream.h#L90, it is expecting a std::string reference. The original issue was caused by passing cpd to data->kinesis_video_stream->start which was freed already. Can you check if you dont have any local changes? Thanks.

rgroleau, 2020-04-22:
The problem is definitely on my end, yes, I just got confused with the commit/revert nature of the git reflog on this fix, sorry about that. I opened another ticket on the problems I was able to get to after this fix, so everything is fine here, again sorry about causing confusion.