awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #396: m3u8 to kinesis fails (but works with fakesink)

kumpelj, 2020-04-05:
How can I ingest m3u8 to Kinesis Video? 

When I run

`
gst-launch-1.0 souphttpsrc location=http://devimages.apple.com/iphone/samples/bipbop/gear4/prog_index.m3u8 ! hlsdemux !  tsdemux ! queue ! video/x-h264 ! h264parse !  kvssink stream-name="my-stream" storage-size=128 access-key="<key>" secret-key="<secret>"
`

and get the following error
`[ERROR] [05-04-2020 18:33:57:145.773 GMT] Failed to get the size of the buffer for hex decoding the codec private data with: 3
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: GStreamer encountered a general stream error.
Additional debug info:
/home/ubuntu/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp(947): gst_kvs_sink_handle_sink_event (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:`

With fakesink or filesink it works fine

disa6302, 2020-04-05:
@kumpelj , 

What is the platform you are running this on?

Can you also run the command with `GST_DEBUG=4`

MushMal, 2020-04-05:
Looks like the similar case of issue #394 

In this case, the GStreamer specified a CPD with size 1 as this is the only case where the hexDecode returns STATUS_INVALID_ARG_LEN.

Now I am wondering whether GStreamer specifies bogus CPD for Audio tracks

kumpelj, 2020-04-05:
with debug=4, find output here https://pastebin.com/EaZAPwc5

disa6302, 2020-04-05:
@kumpelj , 

Thank you for the logs. We are able to reproduce the issue on Ubuntu 18.04 and working to see whats causing it. Will get back to you when we have more details on the issue.  Thank you!

kumpelj, 2020-04-05:
@disa6302  I saw in #394  that the issue did not exist before, so I tried to checkout a previous version and with this old version it works fine!

 git checkout 56b0774c66cc3c596356b1a357cd30a43f844072

MushMal, 2020-04-06:
Fixed in the latest commit. Closing.

closed 2020-04-06 by MushMal