awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #397: Unable to play RTSP stream on ec2 instance running Microsoft Windows Server 2016 Base

Jord-Gui, 2020-04-06:
The RTSP is playable when streaming from VLC, but is returning an error when running gst-launch-1.0 to a fakesink (testing connectivity). 

gst-launch-1.0 rtspsrc do-rtcp=TRUE location=rtsp://10.99.3.11:654/00000001-0000-babe-0000-accc8e4ebf43/live ! fakesink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://10.99.3.11:654/00000001-0000-babe-0000-accc8e4ebf43/live
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
../gst-plugins-good-1.16.2/gst/rtsp/gstrtspsrc.c(5768): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.

The full debug log (GST_DEBUG=4) is below:
[gst_windows_instance_log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4435502/gst_windows_instance_log.txt)

MushMal, 2020-04-06:
I see the logs clearly state that it can’t read the source. In general, try using fakesink to figure out any problems with the pipeline first.

closed 2020-04-06 by MushMal

Jord-Gui, 2020-04-06:
Is there a reason why it is unable to read the source? The source is playable in vlc.

MushMal, 2020-04-06:
The right place to get help for gstreamer plugins is gstreamer forums. I would imagine it could be indeed a firewall issue or some port combination.

timetxt, 2020-04-07:
You forced to use TCP but your url did not change.
Try this: 
gst-launch-1.0 rtspsrc do-rtcp=TRUE **location=rtspt**://10.99.3.11:654/00000001-0000-babe-0000-accc8e4ebf43/live ! fakesink

Jord-Gui, 2020-04-07:
Thank you, that worked - everything is working now

MushMal, 2020-04-07:
Nice!