awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #400: Mingw64 compiling failed in Window Server 2016 (EC2)

timetxt, 2020-04-07:
I was testing from Amazon AMI Win Server 2016 'Windows_Server-2016-English-Full-Base-2020.01.15 (ami-076acf0f99b739999)'. I also tested from Workspace but found it was using the same Window Server 2016' Core. 

Description: 
1.When compiling plugin from Windows Server 2016, it failed at 95% compiling stage.
2. When compiling plugin from commit 2623b87 and afterwards, including the latest commit, it failed at 95% compiling stage.

Tested in Ubuntu EC2 instance and Windows 10, it was working fine in these two OS with the latest commit.

Issue was persistent. I was testing with MSYS2. Only installed git with 'pacman -S git'. And compiling was using './min-install-script'.

Error Message:

C:/msys64/mingw64/x86_64-w64-mingw32/include/_mingw.h:234: note: this is the location of the previous definition
 #define _WIN32_WINNT 0x601

[ 93%] Building CXX object CMakeFiles/gstkvssink.dir/42c7351eebad1af14771aa11892e2cf3/kinesis-video-gstreamer-plugin/src/Util/KvsSinkUtil.cpp.obj
[ 94%] Building CXX object CMakeFiles/gstkvssink.dir/42c7351eebad1af14771aa11892e2cf3/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp.obj
Assembler messages:
Fatal error: can't create CMakeFiles/kinesis_video_gstreamer_audio_video_sample_app.dir/C_/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp.obj: No such file or directory
make[2]: *** [CMakeFiles/kinesis_video_gstreamer_audio_video_sample_app.dir/build.make:63: CMakeFiles/kinesis_video_gstreamer_audio_video_sample_app.dir/C_/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_audio_video_sample_app.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/kinesis_video_gstreamer_audio_video_sample_app.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h:13,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoStreamMetrics.h:5,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoStream.h:11,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.h:5,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.h:35,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp:55:
C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/common/include/com/amazonaws/kinesis/video/common/CommonDefs.h:91: warning: "_WIN32_WINNT" redefined
         #define _WIN32_WINNT 0x0A00

In file included from C:/msys64/mingw64/x86_64-w64-mingw32/include/corecrt.h:10,
                 from C:/msys64/mingw64/x86_64-w64-mingw32/include/crtdefs.h:10,
                 from C:/msys64/mingw64/x86_64-w64-mingw32/include/stddef.h:7,
                 from C:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/8.3.0/include/stddef.h:1,
                 from C:/msys64/mingw64/include/glib-2.0/glib/gmacros.h:38,
                 from C:/msys64/mingw64/lib/glib-2.0/include/glibconfig.h:9,
                 from C:/msys64/mingw64/include/glib-2.0/glib/gtypes.h:32,
                 from C:/msys64/mingw64/include/glib-2.0/glib/galloca.h:32,
                 from C:/msys64/mingw64/include/glib-2.0/glib.h:30,
                 from C:/msys64/mingw64/include/gstreamer-1.0/gst/gst.h:27,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.h:34,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp:55:
C:/msys64/mingw64/x86_64-w64-mingw32/include/_mingw.h:234: note: this is the location of the previous definition
 #define _WIN32_WINNT 0x601

In file included from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Auth.h:12,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.h:11,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.h:9,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.h:35,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp:55:
C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp: In function 'void gst_kvs_sink_init(GstKvsSink*)':
C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Logger.h:26:10: warning: 'template<class> class std::auto_ptr' is deprecated [-Wdeprecated-declarations]
     std::auto_ptr<log4cplus::Layout> _layout(new log4cplus::PatternLayout("%D [%t] ")); \
          ^~~~~~~~
C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/Logger.h:26:10: note: in definition of macro '_LOG_CONFIGURE_CONSOLE'
     std::auto_ptr<log4cplus::Layout> _layout(new log4cplus::PatternLayout("%D [%t] ")); \
          ^~~~~~~~
C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp:658:5: note: in expansion of macro 'LOG_CONFIGURE_STDOUT'
     LOG_CONFIGURE_STDOUT("DEBUG")
     ^~~~~~~~~~~~~~~~~~~~
In file included from C:/msys64/mingw64/include/c++/8.3.0/condition_variable:43,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoStream.h:8,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/KinesisVideoProducer.h:5,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.h:35,
                 from C:/msys64/home/jasonshen/kvs-1/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gstreamer-plugin/src/gstkvssink.cpp:55:
C:/msys64/mingw64/include/c++/8.3.0/bits/unique_ptr.h:53:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
[ 95%] Linking CXX shared module libgstkvssink.dll
[ 95%] Built target gstkvssink
make: *** [Makefile:84: all] Error 2

MushMal, 2020-05-20:
Apologies for delay, We have released V3.0.0 recently which is based on CMake and platform specific package configs. Could you please try the latest?

timetxt, 2020-05-23:
Sure Will try this after getting Ubuntu build pass

disa6302, 2020-06-15:
@timetxt , 

Any updates?

MushMal, 2020-06-26:
Resolving due to staleness

closed 2020-06-26 by MushMal