awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #401: Not able to stream H264 files using the README command

CatsNipYummy, 2020-04-08:
I have been able to stream my webcam video and audio using gstreamer to the same remote video stream (named TestingStream). But when I run the H264 test as follows:

> ./kinesis_video_cproducer_video_only_sample TestingStream 10

I get the following error with an error file not found. What file is it referring to?:

> 2020-04-08 21:20:52 INFO    heapInitialize(): Initializing native heap with limit size 20971520, spill ratio 0% and flags 0x00000001
> 2020-04-08 21:20:52 INFO    heapInitialize(): Creating AIV heap.
> 2020-04-08 21:20:52 INFO    heapInitialize(): Heap is initialized OK
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
> 2020-04-08 21:20:52 INFO    createDeviceResultEvent(): Create device result event.
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
> 2020-04-08 21:20:52 INFO    createKinesisVideoStream(): Creating Kinesis Video Stream.
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): Kinesis Video Stream Info
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Stream name: TestingStream
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Content type: video/h264
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Max latency (100ns): 314500000
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Fragment duration (100ns): 20000000
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Key frame fragmentation: Yes
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Use frame timecode: Yes
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Absolute frame timecode: Yes
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Nal adaptation flags: 40
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Average bandwith (bps): 2097152
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Framerate: 120
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Buffer duration (100ns): 370000000
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Replay duration (100ns): 185000000
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Connection Staleness duration (100ns): 50000000
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Store Pressure Policy: 1
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	View Overflow Policy: 1
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Segment UUID: NULL
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Frame ordering mode: 0
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): Track list
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Track id: 1
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Track name: kvs_video_track
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
> 2020-04-08 21:20:52 DEBUG   logStreamInfo(): 	Track cpd: NULL
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
> 2020-04-08 21:20:52 INFO    writeHeaderCallback(): RequestId: b70ba36f-d216-4604-8cd5-c89627e89115
> 2020-04-08 21:20:52 DEBUG   describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.586317666893E9,"DataRetentionInHours":2,"DeviceName":"VP52M8OQZ10HOQRB","KmsKeyId":"arn:aws:kms:us-west-2:864943317271:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:864943317271:stream/TestingStream/1586317666893","StreamName":"TestingStream","Version":"wb6ZqBHb9XCbGzjOI1dD"}}
> 2020-04-08 21:20:52 INFO    describeStreamResultEvent(): Describe stream result event.
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
> 2020-04-08 21:20:52 INFO    writeHeaderCallback(): RequestId: 30245b01-c0dc-4055-9664-eb84fafca350
> 2020-04-08 21:20:52 DEBUG   getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
> 2020-04-08 21:20:52 INFO    getStreamingEndpointResultEvent(): Get streaming endpoint result event.
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
> 2020-04-08 21:20:52 INFO    getStreamingTokenResultEvent(): Get streaming token result event.
> 2020-04-08 21:20:52 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
> 2020-04-08 21:20:52 DEBUG   defaultStreamReadyCallback(): Reported streamReady callback for stream handle 140572694244240
> 2020-04-08 21:20:52 ERROR   Failed with status 0x00000009
> 
> 2020-04-08 21:20:52 INFO    freeKinesisVideoStream(): Freeing Kinesis Video stream.
> 2020-04-08 21:20:52 DEBUG   curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
> 2020-04-08 21:20:53 DEBUG   defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 140572694244240
> 2020-04-08 21:20:53 INFO    freeKinesisVideoClient(): Freeing Kinesis Video Client
> 2020-04-08 21:20:53 DEBUG   curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
> 2020-04-08 21:20:53 DEBUG   freeKinesisVideoClientInternal(): Total allocated memory 0
> 2020-04-08 21:20:53 WARN    curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

disa6302, 2020-04-08:
The sample is trying to look for H264 samples file (https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/c2f1cacb8b6123b8a11a2194e24249db69278c72/samples/KvsVideoOnlyStreamingSample.c#L73). It looks like the path is wrong. 

Thanks for bringing it to our attention. We will fix it now.

MushMal, 2020-04-08:
Thanks! You can also specify the path directly in the CMD line as the last argument

CatsNipYummy, 2020-04-08:
I ran the following:
`./kinesis_video_cproducer_video_only_sample TestingStream 10 ../kinesis-video-c-producer/samples/
`
Same error still. I also tried with the whole path and not relative. Still the same outcome. Error 0x00000009

disa6302, 2020-04-08:
Apologies
Please add ../kinesis-video-c-producer/samples/ h264SampleFrames

Thanks

MushMal, 2020-04-09:
Please specify the correct path. Closing

closed 2020-04-09 by MushMal