awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #403: Upload same video stream twice or more

wlinInspire, 2020-04-11:
When we upload video files in offline mode, it seems that Kinesis is using the current timestamp instead of using the timestamp when the video was created. It will cause problems when we accidentally upload the same file twice or more into the cloud which will have two different timestamps with the same content.

How could we ensure that the Kinesis video stream use the original timestamp associated with each frame in the file?

MushMal, 2020-04-13:
@wlinInspire could you please provide more information about your scenario

* Which asset you are using (sample app or gst-launch-1.0 with the kvs plugin)
* What file format are you uploading
* What's the actual desired result that you are expecting.

A few notes here... Most of the file formats do not have absolute timestamps. Some, like MKV allow absolute timestamps but most produced files contain relative timestamps - the fragments start from 0 and increment onwards and the frame timestamps are always encoded relative to the start of the cluster (fragment).

wlinInspire, 2020-04-13:
@MushMal 

- I have tried both `kinesis_video_gstreamer_audio_video_sample_app` and `gst-launch-1.0`. Seems neither works as expected.
- I tried both MKV and MP4.
- I would like the `producer timestamp` to be the original timestamp generated with each frame and `server timestamp` to be the timestamp when each frame hits the server. So we can always search the content by `producer timestamp`.  Does it make sense?

If you think MKV should contain the absolute timestamps, are there some settings in the KVS plugin I need to set to enable that? Thanks!

MushMal, 2020-04-13:
The problem is that your original file might not have the original timestamps. What tool do you use to dump out the original frame timecodes? For example, the MPEG by default doesn't store any absolute time stamps - all of the timestamps are relative to the beginning of the presentation - aka - most of the time the file start. Similar situation is with MKV.  If your source - the entity that produced the file, encodes the timestamps in MKV as absolute cluster timestamps then changing the sample can result in absolute timestamps.

wlinInspire, 2020-04-13:
@MushMal I used `splitmuxsink` in gstreamer to generate multiple MKV files.

MushMal, 2020-04-13:
I am not familiar with what this outputs. Perhaps, you could simply run mkvinfo -v <file.mkv> on an intermediate file to dump out the cluster infos and see what timestamps does it use.

wlinInspire, 2020-04-13:
@MushMal  mkinfo shows the following. There is a field `Data` in the head but the cluster starts with 0s.
```
+ EBML head
|+ Doc type: matroska
|+ Doc type version: 2
|+ Doc type read version: 2
+ Segment, size 2801381
|+ Seek head (subentries will be skipped)
|+ Segment information
| + Segment UID: 0x76 0x4e 0xef 0x9f 0x68 0x4e 0xc6 0x58 0x82 0x46 0xcb 0x14 0xc2 0x34 0xcb 0xeb
| + Timestamp scale: 1000000
| + Duration: 11.622s (00:00:11.622)
| + Multiplexing application: GStreamer matroskamux version 1.14.5
| + Writing application: GStreamer Matroska muxer
| + Date: Mon Apr 13 19:03:04 2020 UTC
|+ Segment tracks
| + A track
|  + Track number: 1 (track ID for mkvmerge & mkvextract: 0)
|  + Track type: video
|  + Track UID: 4893042141254354873
|  + Default duration: 33.333ms (30.000 frames/fields per second for a video track)
|  + Name: Video
|  + Video track
|   + Pixel width: 1280
|   + Pixel height: 720
|   + Interlaced: 2
|  + Codec ID: V_MPEG4/ISO/AVC
|  + CodecPrivate, length 45 (h.264 profile: High 4:2:2 @L3.1)
| + A track
|  + Track number: 2 (track ID for mkvmerge & mkvextract: 1)
|  + Track type: audio
|  + Track UID: 12798541518792981045
|  + Default duration: 23.220ms (43.066 frames/fields per second for a video track)
|  + Name: Audio
|  + Audio track
|   + Sampling frequency: 44100
|   + Channels: 2
|  + Codec ID: A_AAC
|  + CodecPrivate, length 5
|+ Cluster
| + Cluster timestamp: 0.000s
| + SimpleBlock (track number 2, 1 frame(s), timestamp 0.000s = 00:00:00.000)
|  + Frame with size 24
| + SimpleBlock (track number 2, 1 frame(s), timestamp 0.023s = 00:00:00.023)
|  + Frame with size 6
| + SimpleBlock (track number 2, 1 frame(s), timestamp 0.046s = 00:00:00.046)
|  + Frame with size 594
| + SimpleBlock (track number 2, 1 frame(s), timestamp 0.070s = 00:00:00.070)
|  + Frame with size 763
```

MushMal, 2020-04-13:
OK, indeed it seems that the plugin you are using to create the file is using relative timestamps - this is the norm for most files.

In your case, if you start with a set of files and would like to preserve the timestamp, you need to somehow either generate the absolute timestamps in the files or store the producer time when the file is created somewhere and use that as the base for the timestamps: 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L80

wlinInspire, 2020-04-14:
@MushMal  Thanks! It makes sense to me now. I can modify the timestamp in the sample code above to whatever I want

closed 2020-04-14 by wlinInspire

MushMal, 2020-04-14:
Yep, the core of the SDK is pretty flexible. Once you know how to configure/work with it you can use it to your advantage :)