awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #404: How to adjust fragment duration

ekcheng, 2020-04-12:
Is there way to adjust the default fragment duration without rebuilding the code?

I see here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/824de900c2d4b42a403319ace5de9d68ec88b171/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L33 that there is a default fragment length of 2 seconds for the kinesis_video_gstreamer_sample_app. 

Is the only way to adjust that value to modify kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp and rebuild?  

Thanks.

disa6302, 2020-04-12:
@ekcheng ,

Yes. There is no command line argument you can give if that is what you mean. You will have to set the parameter in the code and rebuild

ekcheng, 2020-04-12:
Ok, thanks for the quick response! On a related note, how does this interplay with the `DEFAULT_STREAM_FRAMERATE` parameter (which I see is set to 25)?

For example, if the true frame rate of the source is actually 5 frames per second but I leave the default framerate unchanged at 25, will the gstreamer sample app effectively provide a fragment length that is 5x longer than expected (because it was assuming a frame rate 5x faster than truth)?

ekcheng, 2020-04-13:
Hello, 

I just tried adjusting the `DEFAULT_FRAGMENT_DURATION_MILLISECONDS` parameter here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/824de900c2d4b42a403319ace5de9d68ec88b171/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L33

to both `1000` (1 second) and `10000` (10 seconds) but it seems to have no effect on the reported Fragment Duration in the AWS KVS web console. 

I did this by modifying the `Dockerfile` to overwrite the `kinesis_gstreamer_sample_app.cpp` file in the docker image, instead replacing it with my file containing the modified fragment duration before it runs the build script, e.g.:

```
RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git                                                                     
# Overwrite with file containing custom fragment duration prior to building
COPY kinesis_video_gstreamer_sample_app.cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/                                         18                                                                                
# ================ Build the Producer SDK (CPP) ===============================    =============================                                                  
                                                                               
WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native    -build/                                                                        
RUN  chmod a+x ./min-install-script                                             23 RUN ./min-install-script                                                       
RUN ./java-install-script   
```

I then ran the `rtspdockertest` per the example here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docker_native_scripts/README.md

However, the fragment duration reported in the AWS console is always 2.1 seconds. 

Am I modifying the right value?

Thanks.

MushMal, 2020-04-13:
@ekcheng the fragment rate and fragment duration are not affecting the source of the media. Let me cover in details:

fragment rate or FPS: The SDK is using the value to optimize the temporal storage requirements. This is intended to be used as a loose range. For example, for video, you can leave this as the default and it would work fine. However, if you are streaming some custom high-frame rate (for example raw audio when you have 48K samples a second), then specifying this value will result in proper temporal storage. Most of the people don't change this value.

fragment duration: If the low-level SDK is configured to do the fragmentation itself (no key-frame fragmentation) then the SDK will produce a fragment on a key-frame after this time elapses. For example, if you have an audio stream, all of the frames are key frames. The application can specify a key-frame fragmentation as FALSE and set the fragment duration to something manageable like 2 seconds. The higher-level assets like GStreamer plugin or samples rely on the media pipeline to produce key-frames at their intervals and set the key-frame fragmentation as TRUE. In this case, the upstream encoder configuration dictates the fragment duration.

Please let us know if you have more questions. Resolve the issue if you have enough information.

ekcheng, 2020-04-15:
Thank you, @MushMal. 

My video source is a standard IP camera, streaming h264 over rtsp. I am using the `start_rtsp_in_docker.sh` to stream this to KVS. 

To change my fragment duration, do I understand correctly from your answer that I should set this parameter https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/824de900c2d4b42a403319ace5de9d68ec88b171/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L35 to `TRUE` and simply adjust my the keyframe frequency at my source?

MushMal, 2020-04-15:
@ekcheng yes, you would want to drive the fragmentation from your source indeed. Once the source specifies the frame being a key-frame then the SDK would cause fragmentation on that key-frame. As you said, in this case, you need to adjust the fragment duration at your source camera - each would have it's own way to configure it. In case of video, it's also called GoP (Group-of-pictures) or an I-frame interval - depending on the manufacturer. Some specify the period in seconds, while others specify the number of frames per I-frame.

ekcheng, 2020-04-15:
Got it. Thank you!

closed 2020-04-15 by ekcheng