awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #405: SEI NALU causes fragment decoding error in the console

farleylai, 2020-04-14:
The use case is to stream a video clip saved in mp4 to KVS in loops.

First, the NALU inside mp4 is said to be in AVCC format and there seems like no easy way to convert to annex b format for KVS given the SPS/PPS need to be extracted from the `extradata` header?

Therefore, the fallback is to convert the mp4 into an annex b format bitstream for pyAV to read frame by frame. However, the very first frame comes with the `SEI` as the beginning NALU such that the KVS console reports `fragment decoding error` as follows:

[image]

After manually removing the `SEI`, the KVS console finally works with key frames always composed of SPS/PPS/IDR. Is this expected?

MushMal, 2020-04-14:
@farleylai can you provide a little more information on how you source the MP4? Are you using GStreamer plugin? Is it possible that the CPD auto-extraction fails to parse the Annex-B format you produced to extract the SPS/PPS? Do you have the debug logs form the producer I can look at?

In general, if you have AvCC format frames, you don't need to convert them to Annex-B and set the Annex-B -> AvCC adaptation flag for the SDK to convert Annex-B back to AvCC.

If, however, you are using AvCC directly with no adaptation flags, you will need to somehow supply the CPD directly. This is why it depends on how you extract the contents from the MP4 file.

farleylai, 2020-04-14:
@MushMal Thanks for the clarification.

The original recording is from YouTube saved in MPEG4 .avi and then transcoded to H.264 mp4.
When the mp4 is exported to the annex-b format through ffmpeg (`-bsf h264_mp4toannexb`), there is an `SEI` NALU inserted in the very beginning key frame just before SPS/PPS.  I also noticed, there is no SEI generated if transcoded from `.avi` to `.h264` bitstream, only from `.mp4` but if that SEI can be inserted, it is likely allowed and defined in the spec to do so.

The log messages are attached below FYI. There is an error message I just noticed:

```
mkvgenAdaptCodecPrivateData(): Failed extracting video configuration from SPS with 32000015.
```

It is just a bit surprising that the `SEI` affects the extracting of SPS.
Otherwise, the fragments seem persisted eventually without other issues.

```
020-04-14 18:35:06 DEBUG   heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2020-04-14 18:35:06 DEBUG   heapInitialize(): Creating AIV heap.
2020-04-14 18:35:06 DEBUG   heapInitialize(): Heap is initialized OK
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-04-14 18:35:06 DEBUG   createDeviceResultEvent(): Create device result event.
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-04-14 18:35:06 DEBUG   createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-04-14 18:35:06 DEBUG   writeHeaderCallback(): RequestId: 347f7b37-d857-4664-bca7-2bee50dd9957
2020-04-14 18:35:06 DEBUG   describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.586861380409E9,"DataRetentionInHours":6,"DeviceName":"JY9757ACX1EVE7NM","KmsKeyId":"arn:aws:kms:us-east-1:734908081819:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:734908081819:stream/H264Bitstream/1586861380409","StreamName":"H264Bitstream","Version":"qwBcV5LhZPF7zCW5N046"}}
2020-04-14 18:35:06 DEBUG   describeStreamResultEvent(): Describe stream result event.
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2020-04-14 18:35:06 DEBUG   writeHeaderCallback(): RequestId: 999abf70-2909-4fbf-82fb-db3521568b8c
2020-04-14 18:35:06 DEBUG   getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
2020-04-14 18:35:06 DEBUG   getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2020-04-14 18:35:06 DEBUG   getStreamingTokenResultEvent(): Get streaming token result event.
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-04-14 18:35:06 DEBUG   defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94412491056921
[04/14/2020 14:35:06] WARNING: SEI from key frames for KVS
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
2020-04-14 18:35:06 DEBUG   putStreamResultEvent(): Put stream result event.
2020-04-14 18:35:06 DEBUG   notifyDataAvailable(): Failed to un-pause curl with error: 43
2020-04-14 18:35:06 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2020-04-14 18:35:06 DEBUG   mkvgenAdaptCodecPrivateData(): Failed extracting video configuration from SPS with 32000015.
2020-04-14 18:35:06 DEBUG   mkvgenAdaptCodecPrivateData(): Failed extracting video configuration from SPS with 32000015.
2020-04-14 18:35:06 DEBUG   logStreamMetric(): Kinesis Video client and stream metrics: 
        Overall storage byte size: 134217728
        Available storage byte size: 133949923
        Allocated storage byte size: 267805
        Total view allocation byte size: 691272
        Total streams frame rate (fps): 120
        Total streams transfer rate (bps): 16777216 (16384 Kbps)
        Current view duration (ms): 0
        Overall view duration (ms): 0
        Current view byte size: 0
        Overall view byte size: 0
        Current frame rate (fps): 120.000000
        Current transfer rate (bps): 16777216 (16384 Kbps)
Sent key frame[0] of duration 0.067s to KVS with timestamp 1586889306.534s at 1586889306.534s
Sent frame[1] of duration 0.067s to KVS with timestamp 1586889306.601s at 1586889306.549s
Sent frame[2] of duration 0.067s to KVS with timestamp 1586889306.667s at 1586889306.610s
2020-04-14 18:35:06 DEBUG   postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-04-14 18:35:06 DEBUG   writeHeaderCallback(): RequestId: e6633e8c-ed07-99bd-b173-3a16e7ec363d
2020-04-14 18:35:06 DEBUG   postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-04-14 18:35:06 DEBUG   postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[... cut ...]
Sent frame[22] of duration 0.067s to KVS with timestamp 1586889308.001s at 1586889307.947s
2020-04-14 18:35:07 DEBUG   postWriteCallback(): Curl post body write function for stream with handle: H264Bitstream and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":1586889306534,"FragmentNumber":"91343852333181625511328191147209448739898942754"}
```

As for the AvCC format, setting the adaptation flags to `NAL_ADAPTATION_FLAG_NONE` and prepending the SPS/PPS in AvCC from the file format container should do the trick, right?

MushMal, 2020-04-14:
There are a couple of issues you are touching

1) When saving the file to avi - what's the actual bitstream? I assume it's h264 indeed but what format is it in? If it's in proper AvCC then you don't need to transcode to MP4 but just transpackage which is a lot cheaper to do.
2) If the original h264 has b-frames then you might still need to transcode to get rid of the initial b-frames.
3) the failure you reported is not when extracting the CPD. The SDK has a logic to extract width and height and furnish MKV elements for Video with those. It's the getVideoWidthAndHeightFromH264SPS call that's failing - is caused by our internal logic of processing the SPS. It expects the first NALU to be SPS NALU:

CHK(read == SPS_NALU_67 || read == SPS_NALU_27, STATUS_MKV_INVALID_H264_H265_SPS_NALU);

Perhaps we can make it smarter at some stage - please let us know.

4) KVS core streaming is really agnostic of the frames or type of frames you are sending. There are few pieces of the architecture that have intimate knowledge of the frames. One of those is the console playback or HLS/MPEG-DASH. These operate on AvCC format of H264 or H265 encoded frames. If your end-to-end doesn't care, you can feed in any format and get the same format out. If you start with AvCC format, all you need to do is to supply the CPD. It can be either done in the TrackInfo in the initial call to CreateStream.StreamCaps or you can create the stream and before you put any frames into it, you can call StreamFormatChanged API with the CPD - this is done as some media pipelines will not have info on the codec (as they haven't yet configured the encoder) so they have a way to supply the CPD later but before any frames are processed.

farleylai, 2020-04-14:
1. The very original YouTube video is downloaded as `webm` and then transcoded to `.avi`. The `webm` format contains `vp9` (Profile 0) and the `avi` format contains `mpeg4` (Simple Profile) reported by `ffprobe`. Therefore, the transcoding is necessary to generate H.264 bitstream required by KVS.

2. Yes, the transcoding is set to generate zero b-frames

3. It seems like `SEI` is not essential for the core decoding but can surely come with SPS/PPS in the bitstream as described in Advances in Multimedia Information Processing - PCM 2006, page 345:

[image]

So safely disregarding `SEI` that may show before `SPS` would be preferred in case of accidental surprises.

4. Thanks for the hints. I would try to give those APIs a shot next time.

MushMal, 2020-04-15:
This is informative @farleylai , thank you very much. We will evaluate the SEI NALU skipping. The main issue is that all of the auxiliary functionality like extracting CPD, getting width and height are not going to support the full range of MPEG specification anyway and we need to make hard choices at times as to what exactly we are going to support and what we simply can't. For example, in our case, we only support a single SPS and a single PPS although the MPEG spec allows for multiple entries. 

Are you able to automatically remove the SEI? Another alternative is to leave it in place and proactively specify the CPD before streaming.

farleylai, 2020-04-15:
@MushMal Sure, make your best decisions!
So far, we just manually skip the SEI by its size. Not too much work since it is just the first NALU in the key frame. We would definitely consider specifying the CPD to leave things intact soon.

closed 2020-04-15 by farleylai

MushMal, 2020-04-15:
Perfect! I just created a new issue in PIC: https://github.com/awslabs/amazon-kinesis-video-streams-pic/issues/43

Will track and resolve as soon as possible

farleylai, 2020-04-28:
In case of confusion with awslabs/amazon-kinesis-video-streams-pic#43,
I am asking whether the decoding error is gone because the SEI is skipped given the following video input where the SEI is not the very first NALU:
- key frame with SPS/PPS/SEI but no IDR followed by
- key frame with SPS/PPS/SPS/PPS/IDR

In this case, the KVS console player does not return or show any errors.
However, if the SEI is the very first NALU of a key frame (original input of the issue), the decoding error remains.
So it seems like the KVS console player does expect the very first NALU of a key frame must be SPS.