awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #406: gstreamer-plugin-install-script fails

wlinInspire, 2020-04-14:
It seems the following code will copy the files to themselves, causing running `./gstreamer-plugin-install-script` to fail.

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/824de900c2d4b42a403319ace5de9d68ec88b171/kinesis-video-native-build/gstreamer-plugin-install-script#L74

MushMal, 2020-04-14:
Thanks for reporting, it's not immediately clear to me what this line was attempting to achieve as from what I understand the binaries are being built into the native build directory. Also, we really appreciate pull requests :)

wlinInspire, 2020-04-14:
@MushMal Thanks. I don't think I have permission to create a branch

MushMal, 2020-04-15:
@wlinInspire , sorry for the delay. The master branch is "locked" so nobody can push directly into it. The process is to make your changes locally, create a local branch `git checkout -b <branch_name>` then push the code into the remote `git push` (it will suggest the full path). The push process will output the link to the PR. Click the link and publish the PR.

Thanks again!

wlinInspire, 2020-04-19:
@MushMal Thanks. Seems the whole build system is now switched to cmake. Close this issue for now. Will issue PR if I still have the same issue

closed 2020-04-19 by wlinInspire