awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #409: missing kinesis-video-native-build

jasdeepsd, 2020-04-17:
I was following the instructions on the following page to build the c++ producer on the raspberry pi. https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producersdk-cpp-rpi.html#producersdk-cpp-rpi-iam

It says to run the automated build scripts in kinesis-video-native-build, but that folder doesnt exist. I was wondering if there was a new way of installing this on the raspberry pi?

disa6302, 2020-04-17:
Hello @jasdeepsd ,

We changed the folder structure yesterday.
Please follow the instructions on the github page. We are yet to update the AWS docs with the current information.

MushMal, 2020-04-17:
Please follow the Readme in GitHub to get it built. We will be updating the docs. Please open a separate issue if you encounter issues with CMake based scripts

closed 2020-04-17 by MushMal