awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #411: Cmake libopenssl failed

jasdeepsd, 2020-04-17:
When running the cmake i get this issue:
Error 1
make[1]: * [CMakeFiles/Makefile2:73: CMakeFiles/project_libopenssl.dir/all] Error 2
make: * [Makefile:84: all] Error 2
CMake Error at CMake/Utilities.cmake:55 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:69 (build_dependency)

I couldn't seem to find anything online for this issue.

disa6302, 2020-04-17:
@jasdeepsd ,

There is not much information in whatever output you posted. Can you post the rest of the lines before this?
That will have some information on the issue.

jasdeepsd, 2020-04-17:
here's the full error
[image]

closed 2020-04-18 by jasdeepsd