awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #412: Issue with gstreamer sample app

jasdeepsd, 2020-04-18:
I managed to build everything but I get the issue:
-bash: ./kinesis_video_gstreamer_sample_app: No such file or directory

is there a step that I missed?

disa6302, 2020-04-18:
Where are you running it from?

jasdeepsd, 2020-04-18:
I installed it a raspberry pi

disa6302, 2020-04-18:
Please run it from the build directory

Also, make sure you run cmake with -DBUILD_GSTREAMER_PLUGIN=ON option

jasdeepsd, 2020-04-18:
Couldnt find the executable and ran make with that option, but it didn't help.

disa6302, 2020-04-18:
@jasdeepsd ,

Please run cmake with the option and not make

Also, for reference, please check the github readme. It clearly mentions all the flags you can use with cmake. Thanks!

jasdeepsd, 2020-04-18:
That was my bad, yes i ran it with make. GUess ill try the other options

disa6302, 2020-04-18:
Also, make sure to check if cmake completes successfully. Let us know if you have any other questions!

closed 2020-04-18 by MushMal