awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #415: Build fails on MacOS 10.14.6

joydeepsaha05, 2020-04-20:
I was trying to build the project locally on my mac. Ran the `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON` command and get an error saying 
> Can't open perl script "/Configure": No such file or directory

The complete output is:

```
$ cmake .. -DBUILD_GSTREAMER_PLUGIN=ON                   
-- Kinesis Video PIC path is /Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-pic/
-- Kinesis Video C Producer path is /Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-producer-c/
-- Kinesis Video WebRTC Client path is /Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libopenssl
Scanning dependencies of target project_libopenssl
[ 11%] Creating directories for 'project_libopenssl'
[ 22%] Performing download step (git clone) for 'project_libopenssl'
Cloning into 'project_libopenssl'...
Checking out files: 100% (23576/23576), done.
Note: checking out 'OpenSSL_1_1_0l'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 7ea5bd2b52 Prepare for 1.1.0l release
[ 33%] No patch step for 'project_libopenssl'
[ 44%] Performing update step for 'project_libopenssl'
[ 55%] Performing configure step for 'project_libopenssl'
usage: dirname path
Operating system: x86_64-apple-darwinDarwin Kernel Version 18.7.0: Mon Feb 10 21:08:45 PST 2020; root:xnu-4903.278.28~1/RELEASE_X86_64
WARNING! If you wish to build 32-bit library, then you have to
         invoke 'KERNEL_BITS=32 /config  --prefix=/Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local --openssldir=/Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local'.
Can't open perl script "/Configure": No such file or directory
Can't open perl script "/Configure": No such file or directory
This system (darwin64-x86_64-cc) is not supported. See file INSTALL for details.
make[2]: *** [build/src/project_libopenssl-stamp/project_libopenssl-configure] Error 1
make[1]: *** [CMakeFiles/project_libopenssl.dir/all] Error 2
make: *** [all] Error 2
CMake Error at CMake/Utilities.cmake:55 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:73 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "/Users/joydeep/Development/AWS Random Code/Rekognition Kinesis Streaming/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
```

I also tried running the cmake command from a clean build (using `git clean -fxd`), but got the same error output. Are there any other requirements I'm missing?

disa6302, 2020-04-20:
@joydeepsaha05 , 

You are probably missing autoconf. Can you run `brew install autoconf` to see if the error resolves? Thanks!

joydeepsaha05, 2020-04-21:
@disa6302 I already had autoconf installed on my system. However, I ran `brew reinstall autoconf` to reinstall autoconf v2.69. Still get the same error.

disa6302, 2020-04-21:
@joydeepsaha05 ,

I am pretty clueless on why this might be happening. But, a couple of things you can try: 
1. Check perl version (`perl -v`). For OpenSSL 1_1_0 and higher, minimum required perl version is 5.10.
2. It also looks like there are some known issues with Mac OSX where ./configure conveniently ignores darwin64-x86_64-cc as a platform. To confirm this theory, what you could do is, try running `./Configure darwin64-x86_64-cc` by pulling libopenssl from github. If this does not work, then it is platform related. 
3. Another thing you could try is adding this line and then running the cmake command. (`export KERNEL_BITS=64`).

Hope this helps!

mjtk-ktjm, 2020-04-24:
Thanks for the notes disa6302@

I read them in the wrong order, and so tried steps 1 and 2, which did enable me to make the project and utilize it in the blog post.

disa6302, 2020-04-24:
Glad it worked!

closed 2020-04-28 by joydeepsaha05

JoeyYang2010, 2024-05-27:
How to run ./Configure, I can not find Configure file, please help have a check.