awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #416: createDefaultCallbacksProviderWithIotCertificate FailFail

hsinghaowang, 2020-04-21:
version: 2caf6760a289cdc628c712a88f6ed508c8f5a085
OS: arm-linux-3.3
RAM: 128MB 

Call createDefaultCallbacksProviderWithIotCertificate() failed.
The same parameters work in webrtc createLwsIotCredentialProvider().

```
[ 502][E]blockingCurlCall(): Curl perform failed for url https://c1y7ffp4ezkxi4.credentials.iot.us-west-2.amazonaws.com/role-aliases/KvsCameraIoTRoleAlias/credentials with result Timeout was reached : Operation timed out after 4900 milliseconds with 0 out of 0 bytes received
[ 502][W]curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[ 502][E]
../../src/kvs_producer/src/KvsAacAudioVideoStreamingSample.cpp(352):
Failed with status 0x15000011
```

MushMal, 2020-04-21:
Need debug logs and more details. If you are using createLwsIotCredentialProvider then I assume you must have been using the webRTC SDK assets, whereas createDefaultCallbackProviderWithIoTCertificate applies to C Producer SDK.

C Producer operates on the streams whereas the webRTC Signaling Client is operating on the channel primitives. Check whether you have set the proper IoT permissions.

hsinghaowang, 2020-04-21:
We have 2 executable files: webrtc_master and kvs_producer.

webrtc_master used to work well until last Friday. The error is the same as https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c/issues/330

kvs_producer works well with createDefaultCallbacksProviderWithAwsCredentials(). We want it to work with createDefaultCallbacksProviderWithIotCertificate() too.

MushMal, 2020-04-21:
@hsinghaowang I am not sure how the IoT credentials issue you highlighted here relates with ICE/TURN connectivity that's tracked in the mentioned issue or what has it got to do with kvs_producer and what these binaries really are (if I guess, webrtc_master must be referring to the master sample in the webrtc or based on it and kvs_producer might be referring to video only sample???)

Include debug logs so we can try to help.

closed 2020-04-22 by hsinghaowang

henryChmn, 2020-04-22:
The debugs from webcam:
[ipcam_debug_verbose_log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4515521/ipcam_debug_verbose_log.txt)

henryChmn, 2020-04-22:
The debug logs from test page viewer,
[webrtc2.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4515532/webrtc2.log)

MushMal, 2020-04-22:
@henryChmn not sure whether these logs are intended for this issue.

henryChmn, 2020-04-23:
yes, it's for webrtc.

MushMal, 2020-04-23:
@henryChmn this is a KVS CPP repository, it's not WebRTC. 

At any rate, looking at ipcam_debug_verbose_log.txt I can see that you are able to establish connectivity with the Signaling service which means that your IoT integration for WebRTC works fine. However, it seems that there is an issue with the TURN connection. I am not sure yet what's wrong with that - likely a bug that was fixed earlier. Could you please update to the latest codebases both. Could you please file an issue with WebRTC repository?