awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #417: LD_LIBRARY_PATH on Mac OS

jelling, 2020-04-21:
I am getting the errror `WARNING: erroneous pipeline: no element "kvssink"` which I believe is due to not setting LD_LIBRARY_PATH appropriately.

However, all attempts to set LD_LIBRARY_PATH fail as it seems to be a protected environment variable in Mac OS. 

Has anyone run into this?

jelling, 2020-04-22:
This [Mulesoft gives a concise explanation of the issue](https://help.mulesoft.com/s/article/Variables-LD-LIBRARY-PATH-DYLD-LIBRARY-PATH-are-ignored-on-MAC-OS-if-System-Integrity-Protect-SIP-is-enable). 

They recommend turning off System Integrity Protection (SIP) but I'm not quite ready to do that. Is there another way to workaround this?

MushMal, 2020-04-22:
I assume you are using Mac OS Catalina and it fails due to the SIP. I haven't tested it on Catalina yet. Not sure if there are easy ways other than disabling SIP.

However, prior doing that. Have you tried specifying GST_PLUGIN_PATH?

jelling, 2020-04-22:
@MushMal yeah GST_PLUGIN_PATH is specified. 

C++ is outside my usual wheelhouse but would it be possible to create a symbolic link inside of the directories in the MacOS LD_LIBRARY_PATH?

jelling, 2020-04-22:
I went ahead and disabled SIP and successfully reran the build but it still can't find kvssink:

[image]

MushMal, 2020-04-22:
Just to confirm, you do specify the path correctly? using 'pwd' points to it correct?
Nothing stands out immediately as to why it's not loading except the incorrect path. Could you also set 

`export GST_DEBUG=5`

to get more debug spew from GStreamer?

chehefen, 2020-04-22:
when you run `which gst-launch-1.0` what does it return? can you try also export GST_PLUGIN_PATH_1_0 to the same path?

jelling, 2020-04-22:
Yes paths are set to same dir as from where I'm running and gst-launch path set to /usr/local/bin/gst-launch-1.0:

[image]

gst-launch path is set to which points to `/usr/local/bin/gst-launch-1.0 ` `../Cellar/gstreamer/1.16.2/bin/gst-launch-1.0

Bottom part of debug output:

```
0:00:04.028845000  6638 0x7fa3dcc0cc00 DEBUG        GST_PERFORMANCE gstminiobject.c:428:GstMiniObject *gst_mini_object_make_writable(GstMiniObject *): copy GstCaps miniobject 0x7fa3de0049e0 -> 0x7fa3de005000
0:00:04.028856000  6638 0x7fa3dcc0cc00 DEBUG               GST_PADS gstpad.c:4077:gboolean gst_pad_query(GstPad *, GstQuery *):<h264parse0:sink> sent query 0x7fa3de004b20 (caps), result 1
0:00:04.028863000  6638 0x7fa3dcc0cc00 DEBUG               GST_CAPS gstutils.c:3065:GstCaps *gst_pad_query_caps(GstPad *, GstCaps *):<h264parse0:sink> query returned video/x-h264
0:00:04.028874000  6638 0x7fa3dcc0cc00 DEBUG               GST_CAPS gstpad.c:2278:gboolean gst_pad_link_check_compatible_unlocked(GstPad *, GstPad *, GstPadLinkCheck):<vtenc_h264_hw0:src> src caps video/x-h264, width=(int)[ 1, 2147483647 ], height=(int)[ 1, 2147483647 ], framerate=(fraction)[ 0/1, 2147483647/1 ], stream-format=(string)avc, alignment=(string)au
0:00:04.028940000  6638 0x7fa3dcc0cc00 DEBUG               GST_CAPS gstpad.c:2280:gboolean gst_pad_link_check_compatible_unlocked(GstPad *, GstPad *, GstPadLinkCheck):<h264parse0:sink> sink caps video/x-h264
0:00:04.028949000  6638 0x7fa3dcc0cc00 DEBUG               GST_CAPS gstpad.c:2298:gboolean gst_pad_link_check_compatible_unlocked(GstPad *, GstPad *, GstPadLinkCheck): caps are compatible
0:00:04.028958000  6638 0x7fa3dcc0cc00 INFO                GST_PADS gstpad.c:2586:GstPadLinkReturn gst_pad_link_full(GstPad *, GstPad *, GstPadLinkCheck): linked vtenc_h264_hw0:src and h264parse0:sink, successful
0:00:04.029022000  6638 0x7fa3dcc0cc00 INFO               GST_EVENT gstevent.c:1579:GstEvent *gst_event_new_reconfigure(void): creating reconfigure event
0:00:04.029034000  6638 0x7fa3dcc0cc00 DEBUG              GST_EVENT gstevent.c:309:GstEvent *gst_event_new_custom(GstEventType, GstStructure *): creating new event 0x7fa3dd04d960 reconfigure 61441
0:00:04.029043000  6638 0x7fa3dcc0cc00 DEBUG              GST_EVENT gstpad.c:5698:GstFlowReturn gst_pad_send_event_unchecked(GstPad *, GstEvent *, GstPadProbeType):<vtenc_h264_hw0:src> have event type reconfigure event: 0x7fa3dd04d960, time 99:99:99.999999999, seq-num 15, (NULL)
0:00:04.029049000  6638 0x7fa3dcc0cc00 INFO               GST_EVENT gstpad.c:5813:GstFlowReturn gst_pad_send_event_unchecked(GstPad *, GstEvent *, GstPadProbeType):<vtenc_h264_hw0:src> Received event on flushing pad. Discarding
0:00:04.029088000  6638 0x7fa3dcc0cc00 DEBUG                GST_BUS gstbus.c:316:gboolean gst_bus_post(GstBus *, GstMessage *):<bus1> [msg 0x7fa3de03c820] posting on bus structure-change message: 0x7fa3de03c820, time 99:99:99.999999999, seq-num 16, element 'sink', GstMessageStructureChange, type=(GstStructureChangeType)GST_STRUCTURE_CHANGE_TYPE_PAD_LINK, owner=(GstElement)"\(vtenc_h264_hw\)\ vtenc_h264_hw0", busy=(boolean)false;
0:00:04.029101000  6638 0x7fa3dcc0cc00 DEBUG                    bin gstbin.c:3720:void gst_bin_handle_message_func(GstBin *, GstMessage *):<pipeline0> [msg 0x7fa3de03c820] handling child sink message of type structure-change
0:00:04.029109000  6638 0x7fa3dcc0cc00 DEBUG                    bin gstbin.c:958:void bin_remove_messages(GstBin *, GstObject *, GstMessageType):<h264parse0:sink> deleting message 0x7fa3de03c920 of type structure-change (types 0x00001000)
0:00:04.029245000  6638 0x7fa3dcc0cc00 DEBUG                GST_BUS gstbus.c:351:gboolean gst_bus_post(GstBus *, GstMessage *):<bus1> [msg 0x7fa3de03c820] dropped
0:00:04.029257000  6638 0x7fa3dcc0cc00 DEBUG       GST_ELEMENT_PADS gstutils.c:1904:gboolean gst_element_link_pads_full(GstElement *, const gchar *, GstElement *, const gchar *, GstPadLinkCheck): linked pad vtenc_h264_hw0:src to pad h264parse0:sink
WARNING: erroneous pipeline: no element "kvssink"
```

chehefen, 2020-04-23:
you can do `gst-inspect-1.0 appsink` and in the plugin details you will see where the file is. Most likely in /usr/local/lib/gstreamer-1.0/. You can copy the libgstkvssink.so there and you wont need the GST_PLUGIN_PATH and GST_PLUGIN_PATH_1_0 environment variable anymore.

jelling, 2020-04-23:
Thanks. This got me further but I then hit an issue with finding `openssl` which I assume is totally unrelated. Closing this issue.

closed 2020-04-23 by jelling