awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #418: how to debug step machine issues

rgroleau, 2020-04-22:
On my specific yocto/poky platform, when using omxh264enc with kvssink the stream "curl" connectivity works (I see valid stream info in the logs).

But after apparently starting the stream it just ends abruptly with a weird "Dropping frame with flag" debug message from kvssink and some "DEBUG - stepStateMachine()" plain message.

And finally the whole pipeline crashes with a "Bus error (core dumped)" most likely coming from omx.

How can I debug such a stream?

Do you think this is purely related to the omx encoder (which works fine when encoding for example to a matroskamux .mkv file sink)? Or could it be related to the encoding (NV12 instead of I420)? 

Could it be some inline-headers messing up the buffer hex content?

Note that on this platform video and audio encoding works fine (when not using kvssink), including rtp and rtsp.

Anything pointer would help me here, thanks!
```
0:00:00.000304637 3622 0x8560b600 INFO                GST_INIT gst.c:507:init_pre: Initializing GStreamer Core Library version 1.12.2
0:00:00.000694274 3622 0x8560b600 INFO                GST_INIT gst.c:508:init_pre: Using library installed in /usr/lib
0:00:00.000761826 3622 0x8560b600 INFO                GST_INIT gst.c:528:init_pre: Linux 3.18.71 #1 SMP PREEMPT Tue Apr 21 19:07:42 UTC 2020 armv7l
0:00:00.002251103 3622 0x8560b600 INFO                GST_INIT gstmessage.c:127:_priv_gst_message_initialize: init messages
0:00:00.005634450 3622 0x8560b600 INFO                GST_INIT gstcontext.c:84:_priv_gst_context_initialize: init contexts
0:00:00.010468116 3622 0x8560b600 INFO                GST_INIT gst.c:727:init_post: GLib runtime version: 2.52.3
0:00:00.010534887 3622 0x8560b600 INFO                GST_INIT gst.c:729:init_post: GLib headers version: 2.52.3
0:00:00.010586033 3622 0x8560b600 INFO                GST_INIT gst.c:730:init_post: initialized GStreamer successfully
0:00:00.430674938 3622 0x8560b600 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/gstreamer-1.0/libgstkvssink.so" loaded
0:00:00.995548551 3622 0x8560b600 INFO      GST_PLUGIN_LOADING gstplugin.c:843:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/gstreamer-1.0/libgstomx.so" loaded
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2020-04-22 11:11:39 [2822303072] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2020-04-22 11:11:39 [2822303072] INFO - heapInitialize(): Creating AIV heap.
2020-04-22 11:11:39 [2822303072] INFO - heapInitialize(): Heap is initialized OK
2020-04-22 11:11:39 [2822303072] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-04-22 11:11:39 [2822303072] DEBUG - getSecurityTokenHandler invoked
2020-04-22 11:11:39 [2822303072] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1587525099776076936 Expiration: 0
2020-04-22 11:11:39 [2822303072] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-04-22 11:11:39 [2822303072] INFO - createDeviceResultEvent(): Create device result event.
2020-04-22 11:11:39 [2822303072] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-04-22 11:11:39 [2822303072] DEBUG - clientReadyHandler invoked
2020-04-22 11:11:39 [2822303072] INFO - try creating stream
2020-04-22 11:11:39 [2822303072] INFO - Creating Kinesis Video Stream demo-camera
2020-04-22 11:11:39 [2822303072] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Stream name: demo-camera 
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Content type: video/h264 
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Framerate: 25
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): Track list
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Track id: 1
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2020-04-22 11:11:39 [2822303072] DEBUG - logStreamInfo(): 	Track cpd: NULL
2020-04-22 11:11:39 [2822303072] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-04-22 11:11:39 [2822303072] INFO - writeHeaderCallback(): RequestId: 187a92c1-88f5-4c44-91db-52dcefcb7101
2020-04-22 11:11:40 [2666673248] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.582809285215E9,"DataRetentionInHours":1,"DeviceName":null,"KmsKeyId":"arn:aws:kms:us-east-1:123456789012:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:123456789012:stream/demo-camera/1234567890123","StreamName":"demo-camera","Version":"vRfiDhdCDNN8Z1eu7jDx"}}
2020-04-22 11:11:40 [2666673248] INFO - describeStreamResultEvent(): Describe stream result event.
2020-04-22 11:11:40 [2666673248] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[... 120 lines cut ...]
OMXCORE: qc_omx_component_get_parameter 0x84441374, 0x844068bc , 33554433
OMXCORE: qc_omx_component_get_parameter 0x84441374, 0x99bfed2c , 100663312
OMXCORE: qc_omx_component_fill_this_buffer 0x84441374, 0x9e6abcf0
OMXCORE: qc_omx_component_fill_this_buffer 0x84441374, 0x9e6abd40
OMXCORE: qc_omx_component_fill_this_buffer 0x84441374, 0x9e6abd90
OMXCORE: qc_omx_component_fill_this_buffer 0x84441374, 0x9e6abde0
OMXCORE: qc_omx_component_empty_this_buffer 0x84441374, 0x9e60a900
OMXCORE: qc_omx_component_empty_this_buffer 0x84441374, 0x9e60a950
OMXCORE: qc_omx_component_fill_this_buffer 0x84441374, 0x9e6abcf0
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2020-04-22 11:11:42 [2579493984] DEBUG - Dropping frame with flag %u9280
2020-04-22 11:11:42 [2579493984] DEBUG - pts: 15875251023934983, dts: 4043679, duration: 0, size: 48535, trackId: 1, isKey: 1
2020-04-22 11:11:42 [2579493984] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
2020-04-22 11:11:42 [2579493984] INFO - putStreamResultEvent(): Put stream result event. New upload handle 0
2020-04-22 11:11:42 [2579493984] DEBUG - stepStateMachine()
0:00:06.561665638 3621 0x9e6b2cf0 ERROR                    omx gstomx.c:1438:gst_omx_port_release_buffer:<omxh264enc-omxh264enc0> Releasing buffer 0x841a6960 (0x99fe4000) to avc port 1
0:00:06.564906172 3621 0x8437d7b0 ERROR                    omx gstomx.c:1438:gst_omx_port_release_buffer:<omxh264enc-omxh264enc0> Releasing buffer 0x841a69c0 (0x9a3e6000) to avc port 0
[... cut ...]
0:00:06.604877064 3621 0x97401c60 ERROR                    omx gstomx.c:675:FillBufferDone:<omxh264enc-omxh264enc0> avc port 1 filled buffer 0x841a6960 (0x99fe4000), nAllocLen(1382400), nOffset(0), nFilledLen(21979), nSize(80)
0:00:06.605998892 3621 0x9e6b2cf0 ERROR                    omx gstomx.c:1438:gst_omx_port_release_buffer:<omxh264enc-omxh264enc0> Releasing buffer 0x841a6960 (0x99fe4000) to avc port 1
0:00:06.635105573 3621 0x97401c60 ERROR                    omx gstomx.c:675:FillBufferDone:<omxh264enc-omxh264enc0> avc port 1 filled buffer 0x841a6978 (0x9a136000), nAllocLen(1382400), nOffset(0), nFilledLen(19958), nSize(80)
Bus error (core dumped)
```

rgroleau, 2020-04-22:
Some extra logs here, from dmesg I get:
```
[ 1406.155585] msm_vidc:  err: State not recognized
[ 1406.159291] msm_vidc:  err: Failed to move from state: 16 to 13
[ 1406.165476] msm_vidc:  err: Failed to move inst: d5fd2000, cap = 9 to state: 15
[ 1406.172605] msm_vidc: info: Closed video instance: d5fd2000
[ 1406.231794] ------------[ cut here ]------------
[ 1406.235394] WARNING: CPU: 2 PID: 3691 at kernel-source/drivers/media/v4l2-core/videobuf2-core.c:2124 __vb2_queue_cancel+0x44/0x130()
```

MushMal, 2020-04-22:
@rgroleau it seems that something in gstreamer pipeline is crashing in your case. I would recommend replacing kvs plugin with fakesink in the exact same pipeline and try it over again.

The dropped frame happens at the source - this is likely the first few frames which don't contain frame info. GStreamer pipeline could push samples with no frame data - those could be codec private data, etc... The code that handles it is here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/f99eba5f764102e8b63cde696512c2a426e407dc/src/gstreamer/gstkvssink.cpp#L1073

BTW, there is a bug in the printout that uses C++ streams with C printf style parameterization. You are welcome to submit a PR to fix the formatting :)

Regarding the states. The PIC (core business logic that C++ wraps) has an FSA which drives the formal state machine states. The states are somewhat described in the code so you can see but in general, there is new -> describe -> (opt create) -> get endpoint -> ready (pause until the first frame is produced) -> stream -> streaming state.

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/bffd345c4a66fa9a8faf1dd15be15e99d5b67b72/src/client/src/Stream.h#L24-L34

rgroleau, 2020-04-22:
Thanks for the pointers, it really helped! So I first tried the fakesink that you suggested, this works perfectly, no bug, no problem (just as I expected since the capture source and the omxh264enc plugins work perfectly fine when for instance I capture and save a video to a filesink such as matroskamux .mkv or even the jpeg single frame capture plugins work fine too, all using the exact same pipeline specifications, format, etc.).

So this really looks like something original (clearly in the "good plugin" category of course) that kvssink is doing by pausing/stopping/restarting the stream at some unexpected places (unexpected only from the capture source and omxh264enc plugins point of view of course, this seems very legit).

Right now the states that I see are an "uninit" state sent/caused by kvssink through the whole pipeline when "pausing" until the first frame is produced. I will see if I can hack the capture source and omxh264enc plugins to support going from state 9 (start_done) straight to 16 (uninit) and then back to 15 (stop_done) by forcing going again through state 2 and 3 (init and init_done) everytime the "normal" states flow is circumvented.

I don't get why an "uninit" is sent though. Maybe I would be better to just remove the uninit state sent/caused by kvssink in the first place? Maybe I could do that by just "not pausing" until the first frame is produced? Don't we loose some frames by not capturing for a few milliseconds anyway, shouldn't we just go ahead and capture/encode/encapsulate/upload those few milliseconds too?

Note that the device I am working on "starts capturing video" when motion is detected, so those first few frames are the most important thing there is for our product (since this is usually where/when we see the thing/car/human/animal/movement/whatever that caused the streaming event anyways), so in any case dropping them by pausing for an additional delay is maybe less optimal for us. I realize "not pausing" is most likely creating a latency issue here when we start uploading already-late frames right out of the gate, but like I said, if it is the case, I might want those frames anyways for their very-important/most-significant content in our situation.

I will continue trying a few things and I will keep you updated, thanks again for your help! This does not look like a real bug from kvssink right now anyways, just an unexpected-from-my-specific-plugins-perspective behavior.

MushMal, 2020-04-22:
@rgroleau I am still not sure what's causing this. kvssink doesn't uninit the pipeline. It can generate an error though that will tear down the pipeline.

Can we get debug logs for the kvssink? Also, try to enable the GST debug logs by specifying GST_DEBUG=4 env variable (I think you are doing that already). Let's capture some longer logs and attach it to the issue.

rgroleau, 2020-04-23:
So I found it: kvssink causes an alignment trap on my platform. Which forces the various plugins to uninit, and then of course going from uninit to any other state is a no go for them. I will try compiling with various switches to try and find which pointer access is causing this alignment trap (starting with debug compilation, then -fsanitize=address and -fsanitize=undefined) or maybe just to a -Wall and go through all the warnings one by one, whatever gets me faster to it. Most likely something like a pointer to a double that is not a multiple of 8 or a __attribute__((packed)) missing on some struct somewhere.

MushMal, 2020-04-23:
Oh NICE!!! OK, here is what you can do. Specify ALIGNED_MEMORY_MODEL compile variable and re-compile PIC. 

https://github.com/awslabs/amazon-kinesis-video-streams-pic#note-on-alignment

The rest should be aligned.

rgroleau, 2020-04-23:
That was it! That parameter fixed it! Thanks @MushMal for all your help!

closed 2020-04-23 by rgroleau