awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #419: Stream upload fail when stream name not same as thing name

henryChmn, 2020-04-22:
We test KvsAacAudioVideoStreamingSample with iot credential fail when stream name not the same with things name, the response error:
2020-04-22 06:59:39 WARN    blockingCurlCall(): Curl call response failed with http status 403
2020-04-22 06:59:39 WARN    curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2020-04-22 06:59:39 WARN    Failed with status 0x15000011

Is this normal for SDK? 
Thanks

MushMal, 2020-04-22:
It’s not really an issue with the SDK per-say but rather how the trust as well as the permissions are established and validated. The failure is 403 which is access denied that is returned by the backend.

Please reopen if you need more info

closed 2020-04-22 by MushMal

henryChmn, 2020-04-22:
Thanks, got it.