awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #420: Folders & Files missing - C++ Producer SDK on Windows

aeperaltan, 2020-04-22:
Hello,

We have followed the documentation found at https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-win.html  to use the C++ Producer SDK on Windows but it refers to folders and files that no longer exist on GitHub. Per the documentation, there should be /min-install-script  for MinGW and vs-buildtools-install.bat file for Visual Studio. However, both files and the folder in which they are supposed to be do not exist on GitHub.

Please confirm if this has been moved to another location.

Also, we have the same problem for MAC OS documentation on Github.

Thanks,
Amador

MushMal, 2020-04-22:
In the latest commits we moved towards CMake based build system. The documentation hasn't been updated yet.

Could you please follow the build steps highlighted in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/README.md

Include some details as to what's failing so we can help debugging.

JuanCarlos-Arce, 2020-04-23:
Hi,

Following the steps of the README.md on Windows and compiling with VS2017 I receive an error during the 

> cmake ..

while the library compiles **project_libopenssl**. The error that I have is:

> -- Building for: Visual Studio 15 2017
-- The C compiler identification is MSVC 19.16.27038.0
-- The CXX compiler identification is MSVC 19.16.27038.0
....
....
This version of amazon-kinesis-video-streams-producer-sdk-cpp\open-source\libopenssl\build\src\project_libopenssl\config.com is not compatible with the version of Windows you're running. Check your computer's system information  and then contact the software publisher.

Hope it helps

disa6302, 2020-05-03:
Closing this issue based on Mushmal's comment. Feel free to open a new issue if you face problems with the new repo.

closed 2020-05-03 by disa6302