awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #426: Instructions in README don't work

mwildehahn, 2020-04-26:
Given the following Dockerfile:

```
FROM ubuntu:18.04

RUN apt-get upgrade
RUN apt-get update

RUN apt-get install -y git && \
    apt-get install -y pkg-config && \
    apt-get install -y cmake && \
    apt-get install -y automake && \
    rm -rf /var/lib/apt/lists/*
WORKDIR /opt/

RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
RUN mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake ..
```

It results in:

```
/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus/missing: line 81: aclocal-1.16: command not found
WARNING: 'aclocal-1.16' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
Makefile:934: recipe for target 'aclocal.m4' failed
make[3]: *** [aclocal.m4] Error 127
make[2]: *** [build/src/project_log4cplus-stamp/project_log4cplus-build] Error 2
CMakeFiles/project_log4cplus.dir/build.make:111: recipe for target 'build/src/project_log4cplus-stamp/project_log4cplus-build' failed
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/project_log4cplus.dir/all' failed
make[1]: *** [CMakeFiles/project_log4cplus.dir/all] Error 2
make: *** [all] Error 2
Makefile:83: recipe for target 'all' failed
CMake Error at CMake/Utilities.cmake:55 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:77 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
The command '/bin/sh -c mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake ..' returned a non-zero code: 1
```

Is there something I'm missing?

mwildehahn, 2020-04-26:
I updated the Dockerfile to this:

```
FROM ubuntu:18.04

RUN apt-get upgrade
RUN apt-get update

RUN apt-get update && apt-get -y --no-install-recommends install \
    vim \
    wget \
    git \
    build-essential \
    pkg-config \
    cmake \
    automake \
    liblog4cplus-dev \
    libcurl4-openssl-dev \
    ca-certificates \
    libssl-dev

RUN apt-get install -y libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
    gstreamer1.0-plugins-base gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
    gstreamer1.0-libav libgstrtspserver-1.0-dev

WORKDIR /opt/

RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
RUN mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF
```

But this now fails with:

```
Step 8/8 : RUN mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF
 ---> Running in 75183c63232e
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video PIC path is /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-pic/
-- Kinesis Video C Producer path is /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-producer-c/
-- Kinesis Video WebRTC Client path is /opt/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Found OpenSSL: /usr/lib/x86_64-linux-gnu/libcrypto.so (found version "1.1.1")
-- Checking for module 'libcurl'
--   Found libcurl, version 7.58.0
CMake Error at CMakeLists.txt:158 (add_subdirectory):
  The source directory

    /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-producer-c

  does not contain a CMakeLists.txt file.

-- Checking for module 'gstreamer-app-1.0'
--   Found gstreamer-app-1.0, version 1.14.5
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Jsmn
    linked by target "KinesisVideoProducer" in directory /opt/amazon-kinesis-video-streams-producer-sdk-cpp

-- Configuring incomplete, errors occurred!
See also "/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".
The command '/bin/sh -c mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF' returned a non-zero code: 1
```

Sean-Der, 2020-04-26:
Hey @mwildehahn 

Can you update the `git clone` include the `--recursive`?

For your first Dockerfile I am not sure, does `build-essential` pull in `aclocal-1.16` ?

mwildehahn, 2020-04-26:
Yup, that isn't in the README but I realized that was missing from reading it in some other instructions.

I think this is the correct Dockerfile:

```
FROM ubuntu:20.04

RUN apt-get upgrade
RUN apt-get update

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get -y --no-install-recommends install \
    vim \
    wget \
    git \
    pkg-config \
    cmake \
    automake \
    ca-certificates

RUN apt-get install -y apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
    gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x \
    gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio

WORKDIR /opt/

RUN git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
RUN mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON
```

After everything builds I'm having issues registering the kvssink with gts-launch though:

```
root@223e9dca3677:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build# ls
CMakeCache.txt       kinesis_video_gstreamer_audio_video_sample_app  libKinesisVideoProducer.so
CMakeFiles           kinesis_video_gstreamer_sample_app              libgstkvssink.so
Makefile             kinesis_video_gstreamer_sample_multistream_app
cmake_install.cmake  kvscproducer
root@223e9dca3677:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build# gst-launch-1.0 -v rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$app/$name" ! flvdemux name=demux demux.video ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=$name storage-size=512
WARNING: erroneous pipeline: no element "kvssink"
```

How do I register the kvssink with gstreamer?

mwildehahn, 2020-04-26:
Ah I was missing exports from these instructions: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html. Will try that and see.

mwildehahn, 2020-04-26:
Hm I adjusted those exports to match the updated directory structure:

```
export LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/lib:$LD_LIBRARY_PATH
export PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/downloads/local/bin:$PATH
export GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH
```

and it still isn't able to find it

Sean-Der, 2020-04-26:
In your build directory do you have `libgstkvssink.so`?

If I execute

    LD_LIBRARY_PATH=`pwd`/../open-source/local/lib GST_PLUGIN_PATH=`pwd` gst-inspect-1.0 kvssink

I get the plugin details. The plugin may be blacklisted, try removing `rm -rf ~/.cache/gstreamer-1.0/`

I am working on updating the docs, they are in a PR now. Should be done today!

mwildehahn, 2020-04-27:
Thanks for the responses today!

This was the final Dockerfile which got everything working:

```
FROM ubuntu:20.04
​
RUN apt-get upgrade
RUN apt-get update
​
ENV DEBIAN_FRONTEND=noninteractive
​
RUN apt-get -y --no-install-recommends install \
    vim \
    wget \
    git \
    pkg-config \
    cmake \
    automake \
    ca-certificates
​
RUN apt-get install -y libgstreamer1.0-0 gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
    gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x \
    gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio \
    libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
    gstreamer1.0-plugins-base gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
    gstreamer1.0-libav libgstrtspserver-1.0-dev
​
WORKDIR /opt/
​
RUN git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
RUN mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON; make
​
ENV LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH
ENV PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH
ENV GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH
```

Sean-Der, 2020-04-27:
Happy to help @mwildehahn!

I also pushed a PR that builds autoconf/automake for us. We have been getting lots of PRs that this is a sticking point, so that should make it easier in the future :)

Closing, but if you still have issues please re-open!

closed 2020-04-27 by Sean-Der