awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #427: Unable to build following readme - 'aclocal-1.16' is missing on your system

js1972, 2020-04-27:
After issuing this command `cmake .. -DBUILD_GSTREAMER_PLUGIN=ON` I end up with the following error:

```
[ 66%] Performing build step for 'project_log4cplus'
/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus/missing: line 81: aclocal-1.16: command not found
WARNING: 'aclocal-1.16' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
Makefile:934: recipe for target 'aclocal.m4' failed
make[3]: *** [aclocal.m4] Error 127
CMakeFiles/project_log4cplus.dir/build.make:111: recipe for target 'build/src/project_log4cplus-stamp/project_log4cplus-build' failed
make[2]: *** [build/src/project_log4cplus-stamp/project_log4cplus-build] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/project_log4cplus.dir/all' failed
make[1]: *** [CMakeFiles/project_log4cplus.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
CMake Error at CMake/Utilities.cmake:55 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:77 (build_dependency)

-- Configuring incomplete, errors occurred!
```

I have already installed autotools.

In a previous version you used to have a script called min-install-script and this worked well.

Sean-Der, 2020-04-27:
@js1972 What version of `autotools` do you have? Some distros provide distinct versions because they break compatibility.

I am going to update our build system to build autotools in-tree, this has caused a lot of headache for people. sorry about that!

Sean-Der, 2020-04-27:
Hey @js1972 can you try out the PR I just pushed ^

Instead of using the global automake/autoconf we are just going to bundle it.

js1972, 2020-04-28:
I've just managed to get it working from the branch "2018...." (the old way). Can I just git clone the master branch to test the build without effecting my other installation? What I'm after is kvssink and this produces a worrking version.

What are the reasons and differences to the new way of installing the kinesis producer library?

Sean-Der, 2020-04-28:
If you get a chance I would pull that PR, you are going to be missing out on a lot of bug fixes.

> What are the reasons and differences to the new way of installing the kinesis producer library?
I am probably forgetting some, but these were my motivations.

* We are splitting out everything into unique repos. We released a C WebRTC implementation, and we didn't want to slow people down who only wanted to use one of our products.
* portability issues/bugs with shell scripts. We would break this accidentally
* Way more LoC to maintain. The commit to move everything was `649 insertions(+), 19742 deletions(-)`
* We added lots of tooling (ASAN, LSAN, TSAN, MSAN, Code Coverage and CI) stuff that we didn't have before. Adding this to all our existing scripts would have been a lot of work, and we would have broke things.

Sorry for the breakage, but we are just trying to make the repo better. I appreciate the feedback, and I hope you can use the PR! I promise we are acting on your feedback, if you stick to old version it will be harder to make `HEAD` better :)

js1972, 2020-04-28:
@Sean-Der thanks. The build works well now using PR 429. However I'm unable to use the kvssink component. I get the error `No such element or plugin 'kvssink'` but I have already set `export GST_PLUGIN_PATH=`pwd`/build`.

In the old versions you needed to also set  an env var called LD_LIBRARY_PATH - is that still required but just not documented? The directory it used to point to is now gone. It was: amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib

Sean-Der, 2020-04-28:
Oh yes I missed that sorry, only documented under the platform specific instructions

```LD_LIBRARY_PATH=`pwd`/open-source/lib``` also make sure to clean your plugin blacklist. `~/.cache/gstreamer-1.0/` will update the README if that fixes it for you.

js1972, 2020-04-28:
gst-inspect-1.0 still can't seem to find kvssink, but after I clear the blacklist I see the following output:

```
$ gst-inspect-1.0 kvssink

(gst-plugin-scanner:3072): GStreamer-WARNING **: 14:17:48.290: Failed to load plugin '/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/kvscproducer/libcproducer.so': /home/jason/dev/amazon-kinesis-video-streams-prod
ucer-sdk-cpp/open-source/local/lib/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by /usr/lib/aarch64-linux-gnu/libssl.so.1.1)

(gst-plugin-scanner:3072): GStreamer-WARNING **: 14:17:48.295: Failed to load plugin '/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/libKinesisVideoProducer.so': /home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by /usr/lib/aarch64-linux-gnu/libssl.so.1.1)

(gst-plugin-scanner:3072): GStreamer-WARNING **: 14:17:48.296: Failed to load plugin '/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so': /home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by /usr/lib/aarch64-linux-gnu/libssl.so.1.1)

No such element or plugin 'kvssink'
```

I'd previously set:
```
GST_PLUGIN_PATH=/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build
LD_LIBRARY_PATH=/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/open-source/local/lib
```

js1972, 2020-04-28:
ok solved it. Had to add local into you directory list for the LD_LIBRARY_PATH and additionally add `/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/kvscproducer`

Thankyou for all your help here. Seems to be installed properly based on your PR. I'll now test streaming up to kinesis.

Has anything changed in kvssink itself?

js1972, 2020-04-28:
Have tested streaming from a h.264 rtsp camera up into kinesis video with this version and all works great! Thankyou @Sean-Der for your amazingly quick support. Very much appreciated.

closed 2020-04-28 by js1972