awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #430: Conflict when packaging with nginx

mwildehahn, 2020-04-28:
I'm trying to hook up the nginx-rtmp module to support accepting RTMP and then using gstreamer to publish to Kinesis Video Stream.

I got this docker image working which installs and builds everything correctly:

```
FROM ubuntu:20.04

RUN apt-get upgrade
RUN apt-get update

ENV DEBIAN_FRONTEND=noninteractive

# Install base packages
RUN apt-get -y --no-install-recommends install \
    vim \
    wget \
    git \
    pkg-config \
    cmake \
    automake \
    ca-certificates \
    openssl \ 
    libssl-dev

# Install gstreamer
RUN apt-get install -y libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
    gstreamer1.0-plugins-base gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly \
    gstreamer1.0-libav libgstrtspserver-1.0-dev

# Download and decompress Nginx
RUN apt-get install -y nginx libnginx-mod-rtmp

WORKDIR /opt/

# Build kvssink
RUN git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
RUN mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build; cd amazon-kinesis-video-streams-producer-sdk-cpp/build; cmake .. -DBUILD_GSTREAMER_PLUGIN=ON; make

ENV LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH
ENV PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH
ENV GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH

# Forward logs to Docker
RUN ln -sf /dev/stdout /var/log/nginx/access.log && \
    ln -sf /dev/stderr /var/log/nginx/error.log

# Set up config file
COPY nginx.conf /etc/nginx/nginx.conf

EXPOSE 1935
CMD ["nginx", "-g", "daemon off;"]
```

But when running it, there is a conflict with openssl:

```
root@7b47ce00f402:/opt# nginx  -g "daemon off;"
nginx: /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libssl.so.1.1: version `OPENSSL_1_1_1' not found (required by nginx)
```

Any ideas on how to get around this?

MushMal, 2020-04-28:
Curious if the order in setting LD_LIBRARY_PATH matters.

mwildehahn, 2020-04-28:
This is all I have in the LD_LIBRARY_PATH right now:

```
root@5a7415c212ea:/opt# echo $LD_LIBRARY_PATH
/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:
```

mwildehahn, 2020-04-28:
I've never worked with cpp before, but I'm curious if there is a way to package kvssink by itself so I can copy it over into another container and not have any of these build dependencies around?

mwildehahn, 2020-04-28:
Also, when just trying to run gstreamer, seeing this:

```
root@12ab9d9eb062:/opt# gst-launch-1.0 -v rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$app/$name" ! flvdemux name=demux demux.video ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink stream-name=$name storage-size=512

(gst-plugin-scanner:10): GStreamer-WARNING **: 22:28:04.992: Failed to load plugin '/usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstcurl.so': /opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by /lib/x86_64-linux-gnu/libssh.so.4)
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
ERROR - Unable to create Rotating Credential provider. Error status: 0x9
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:kvssink0: Could not initialize supporting library.
Additional debug info:
/opt/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1374): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:kvssink0:
Failed to init kvs producer. Error: Unable to create Rotating Credential provider. Error status: 0x9
2020-04-28 22:28:05 [140208872154176] Setting pipeline to NULL ...
Freeing pipeline ...
```

Sean-Der, 2020-04-28:
So I think the best thing you can do is not build dependencies automatically, and instead use the system libraries. You can configure that use CMake.

It looks like nginx is linked against a different version of OpenSSL, and you are going to trip over the ABI differences. You could also maybe change the version of libraries we use! You can get into that in CMake/Dependencies.

disa6302, 2020-04-29:
@mwildehahn , can you also add this path to LD_LIBRARY_PATH? 
`/usr/local/lib`, assuming the openssl on your ubuntu system is installed here and see if it solves your problem

Regarding kvssink package, as far as I know, we do not provide anyway to package kvssink by itself, @MushMal might have a better idea this. 

Regarding the log4cplus error: Try adding the `log-config=samples/kvs_log_configuration` to the end of your `gst-launch-1.0`. That should fix the error.

mwildehahn, 2020-04-29:
:yay: updating openssl fixed the issue. Do you want me to open up a PR for that? 

Also, thank you @disa6302 , that worked! 

Y'all are the best.

MushMal, 2020-04-29:
RE: The packaging. kvssink is a dynamic library which has a dependency on GStreamer libraries and CPP producer dynamic library which in turn has a dependency on log4cplus and C producer dynamic library, which, in turn depends on libCurl, openSSL and libCrypto. It is possible to ship the kvs sink and let it link to the system versions of the libraries but it is likely to result in mismatched versions. 

An easier way to distribute the kvs plugin would be if we publish this plugin into one of the plugin libraries (Good plugins). This is something we are looking at but no ETA yet unfortunately.

disa6302, 2020-04-30:
Closing this issue since the problem is fixed.

closed 2020-04-30 by disa6302

samjosephkpl, 2020-05-15:
I got the same issue when I tried to run `./kvs_gstreamer_sample` after `make`. I had to add `<sdk_directory>/open-source/local/lib` to LD_LIBRARY_PATH to fix it. 

`export LD_LIBRARY_PATH=`pwd`/open-source/local/lib:$LD_LIBRARY_PATH` 

Just writing it down here, if it helps others.