awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #431: Unable to get RTSP streaming with H 265 Video Encoding with error "Pad link failed"

dvsantana, 2020-04-28:
I'm using the producer library in a Raspberry Pi 3 to stream from RTSP source.

I tried using the sample multistream app (kinesis_video_gstreamer_sample_multistream_app) to send a H265 video source to AWS KVS. 

When I try to send the stream with H265, 1080p, VBR, and 15fps, I get the message "New RTSP source found: / Source Link FAILED" error twice and no information about the streams. It sends the Source Link FAILED from two different sources, as if it tries to connect twice. 

However, when I send the stream with H 264, 1080p, VBR, and 15fps, I get  "New RTSP source found: / Source linked", and right after:  "New RTSP source found: / Source Link FAILED", even though I am only working with one stream. Still, this way, the stream still gets sent to AWS and can be seen on the AWS Management Console.

Then, I tried using the sample app (kinesis_video_gstreamer_sample_app), to see if there was any other information on the error, saw the following error and tried the fix from issue #130 .
```sh
"Pad link failed
Error received from element udpsrc1: Internal data stream error."
```

I changed the kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp and ran the ./min-install-script:

```c++
#include <gst/gst.h>
#include <gst/app/gstappsink.h>
#include <string.h>
#include <gst/rtsp/gstrtsptransport.h>
#include <chrono>
#include <Logger.h>
#include "KinesisVideoProducer.h"
#include <vector>
#include <stdlib.h>
#include <mutex>
#include <IotCertCredentialProvider.h>

(...)
(...)

// configure rtspsrc
    g_object_set(G_OBJECT (source),
                 "location", data->rtsp_url.c_str(),
                 "short-header", true, // Necessary for target camera
		 "protocols", GST_RTSP_LOWER_TRANS_UDP, 
                 NULL);
```
However I still got the same error.

-------------------------

I'm aware that I won't be able to see the stream on AWS Management Console, as it's not yet supported, however, there is no stream being sent whatsoever.

So, my questions are:
1) Is it possible to send the H265 Stream to KVS using this producer SDK?
2) Can it be done using the sample multistream app or the sample app?
2.1) If so, what do I have to do to make it work?
2.2) If it's not possible, how can it be done?

Below are the DEBUG logs from both tries using the sample app:

Console Log - H265
```sh

[INFO ] [28-04-2020 22:44:21:603.871 GMT] Using region: us-west-2
[INFO ] [28-04-2020 22:44:21:604.146 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [28-04-2020 22:44:21:604.224 GMT] No session token was detected.
[INFO ] [28-04-2020 22:44:21:615.704 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [28-04-2020 22:44:21:615.995 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [28-04-2020 22:44:21:616.069 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [28-04-2020 22:44:21:616.255 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [28-04-2020 22:44:21:616.416 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [28-04-2020 22:44:21:616.515 GMT] getSecurityTokenHandler invoked
[DEBUG] [28-04-2020 22:44:21:616.740 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1588113861616598212 Expiration: 0
[INFO ] [28-04-2020 22:44:21:616.850 GMT] New credentials expiration is 1588117461
[DEBUG] [28-04-2020 22:44:21:616.969 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [28-04-2020 22:44:21:617.051 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [28-04-2020 22:44:21:617.143 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[... 174 lines cut ...]
[DEBUG] [28-04-2020 23:11:08:565.470 GMT] postReadCallback(): Wrote 41295 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-04-2020 23:11:08:568.777 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [28-04-2020 23:11:08:886.796 GMT] postReadCallback(): Wrote 5891 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-04-2020 23:11:08:888.033 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [28-04-2020 23:11:09:246.301 GMT] postWriteCallback(): Curl post body write function for stream with handle: xxxxxxx-stream and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":aaaaaaaaaa,"FragmentNumber":"bbbbbbbbbb"}

[DEBUG] [28-04-2020 23:11:09:247.030 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [28-04-2020 23:11:09:247.679 GMT] Reporting fragment ack received. Ack timecode aaaaaaaaaa
[DEBUG] [28-04-2020 23:11:09:248.309 GMT] postReadCallback(): Wrote 7087 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-04-2020 23:11:09:249.254 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [28-04-2020 23:11:09:253.636 GMT] postWriteCallback(): Curl post body write function for stream with handle: xxxxxxx-stream and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":cccccccccc,"FragmentNumber":"ddddddddddd"}

[DEBUG] [28-04-2020 23:11:09:254.939 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [28-04-2020 23:11:09:256.339 GMT] Reporting fragment ack received. Ack timecode cccccccccc
[DEBUG] [28-04-2020 23:11:09:358.558 GMT] postWriteCallback(): Curl post body write function for stream with handle: xxxxxxx-stream and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":aaaaaaaaaa,"FragmentNumber":"bbbbbbbbbb"}

[DEBUG] [28-04-2020 23:11:09:360.134 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [28-04-2020 23:11:09:360.447 GMT] Reporting fragment ack received. Ack timecode aaaaaaaaaa
[DEBUG] [28-04-2020 23:11:09:360.790 GMT] postReadCallback(): Wrote 2899 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-04-2020 23:11:09:361.472 GMT] postReadCallback(): Pausing CURL read for upload handle: 0
[DEBUG] [28-04-2020 23:11:10:363.216 GMT] postReadCallback(): Wrote 16486 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [28-04-2020 23:11:10:365.202 GMT] postReadCallback(): Pausing CURL read for upload handle: 0

(...)
```

MushMal, 2020-04-29:
@dvsantana the issue really is about how to configure the RTSP plugin rather the KVS itself. The KVS does support H265 ingestion and playback in the console (as long as your browser supports the MSE with H265).

Could you please enable GStreamer debug logs to understand why the Pads are not being negotiated?

Set GST_DEBUG=4 env var like so

`export GST_DEBUG=4`

dvsantana, 2020-04-29:
@MushMal, thanks for the quick reply!
The log was very long this time:
```sh
[INFO ] [29-04-2020 12:53:28:849.811 GMT] Using region: us-west-2
[INFO ] [29-04-2020 12:53:28:850.179 GMT] Using aws credentials for Kinesis Video Streams
[INFO ] [29-04-2020 12:53:28:850.255 GMT] No session token was detected.
[INFO ] [29-04-2020 12:53:28:862.214 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [29-04-2020 12:53:28:862.558 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [29-04-2020 12:53:28:862.634 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [29-04-2020 12:53:28:862.795 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [29-04-2020 12:53:28:862.965 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [29-04-2020 12:53:28:863.090 GMT] getSecurityTokenHandler invoked
[DEBUG] [29-04-2020 12:53:28:863.363 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1588164808863194936 Expiration: 0
[INFO ] [29-04-2020 12:53:28:863.487 GMT] New credentials expiration is 1588168408
[DEBUG] [29-04-2020 12:53:28:863.642 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [29-04-2020 12:53:28:863.730 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [29-04-2020 12:53:28:863.842 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [29-04-2020 12:53:28:863.922 GMT] clientReadyHandler invoked
[DEBUG] [29-04-2020 12:53:28:864.124 GMT] Client is ready
[INFO ] [29-04-2020 12:53:28:864.656 GMT] Creating Kinesis Video Stream xxxxx-stream
[INFO ] [29-04-2020 12:53:28:864.883 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [29-04-2020 12:53:28:864.970 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [29-04-2020 12:53:28:865.038 GMT] logStreamInfo(): 	Stream name: xxxxx-stream 
[DEBUG] [29-04-2020 12:53:28:865.105 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [29-04-2020 12:53:28:865.169 GMT] logStreamInfo(): 	Content type: video/h264 
[DEBUG] [29-04-2020 12:53:28:865.236 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [29-04-2020 12:53:28:865.301 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [29-04-2020 12:53:28:865.366 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [29-04-2020 12:53:28:865.430 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [29-04-2020 12:53:28:865.493 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [29-04-2020 12:53:28:865.558 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [29-04-2020 12:53:28:865.623 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [29-04-2020 12:53:28:865.688 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [29-04-2020 12:53:28:865.752 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [29-04-2020 12:53:28:865.817 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [29-04-2020 12:53:28:865.883 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [29-04-2020 12:53:28:865.949 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [29-04-2020 12:53:28:866.015 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [29-04-2020 12:53:28:866.080 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [29-04-2020 12:53:28:866.144 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [29-04-2020 12:53:28:866.207 GMT] logStreamInfo(): Track list
[DEBUG] [29-04-2020 12:53:28:866.271 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [29-04-2020 12:53:28:866.333 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [29-04-2020 12:53:28:866.397 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [29-04-2020 12:53:28:866.461 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [29-04-2020 12:53:28:866.524 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [29-04-2020 12:53:28:866.876 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [29-040:00:00.000547187 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;31m            GST_INIT gst.c:586:init_pre:[00m Initializing GStreamer Core Library version 1.14.4
0:00:00.000822760 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;31m            GST_INIT gst.c:587:init_pre:[00m Using library installed in /usr/lib/arm-linux-gnueabihf
0:00:00.000924427 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;31m            GST_INIT gst.c:607:init_pre:[00m Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l
0:00:00.002469843 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;31m            GST_INIT gstmessage.c:127:_priv_gst_message_initialize:[00m init messages
0:00:00.005786925 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;31m            GST_INIT gstcontext.c:84:_priv_gst_context_initialize:[00m init contexts
0:00:00.007263747 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;36m  GST_PLUGIN_LOADING gstplugin.c:317:_priv_gst_plugin_initialize:[00m registering 0 static plugins
0:00:00.007927809 [332m 1051[00m  0x1e89e00 [36mINFO   [00m [00;01;36m  GST_PLUGIN_LOADING gstplugin.c:225:gst_plugin_register_static:[00m registered static plugin "staticelements"
[... cut ...]
[INFO ] [29-04-2020 12:53:32:013.450 GMT] Freeing Kinesis Video Stream xxxxx-stream
[INFO ] [29-04-2020 12:53:32:013.746 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [29-04-2020 12:53:32:013.859 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [29-04-2020 12:53:32:416.102 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [29-04-2020 12:53:32:416.367 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

```

MushMal, 2020-04-29:
It's not really clear to me what's causing the TEARDOWN. Seems like the rtsp module. Might need more eyes on it. @chehefen any ideas?

dvsantana, 2020-04-29:
From the sample_app, it seems like it looks only for h264 encoding. Could that be the reason?

disa6302, 2020-05-03:
@dvsantana , 
Have you tried giving `video/h265` as the value for content-type?

dvsantana, 2020-05-03:
@disa6302 , 
I have,
I also changed the other occurences of 264 to 265 to see if I would get anything different, and I got another error, so maybe it is not allowed

dvsantana, 2020-05-04:
This is the error I get when I change every occurrence of 264 to 265 (7 occurrences) in the sample multistream app.

There is a line which might be a problem, that contains:
`logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC `

```sh
[INFO ] [04-05-2020 00:23:46:062.028 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [04-05-2020 00:23:46:062.259 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [04-05-2020 00:23:46:062.296 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [04-05-2020 00:23:46:062.377 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [04-05-2020 00:23:46:062.485 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [04-05-2020 00:23:46:062.559 GMT] getSecurityTokenHandler invoked
[DEBUG] [04-05-2020 00:23:46:062.660 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1588551826062609694 Expiration: 0
[INFO ] [04-05-2020 00:23:46:062.725 GMT] New credentials expiration is 1588555426
[DEBUG] [04-05-2020 00:23:46:062.780 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [04-05-2020 00:23:46:062.814 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [04-05-2020 00:23:46:062.861 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [04-05-2020 00:23:46:062.895 GMT] clientReadyHandler invoked
[DEBUG] [04-05-2020 00:23:46:062.930 GMT] Client is ready
[INFO ] [04-05-2020 00:23:46:063.033 GMT] Creating Kinesis Video Stream stream-xxxx
[INFO ] [04-05-2020 00:23:46:063.074 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [04-05-2020 00:23:46:063.108 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [04-05-2020 00:23:46:063.130 GMT] logStreamInfo(): 	Stream name: stream-xxxx 
[DEBUG] [04-05-2020 00:23:46:063.156 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [04-05-2020 00:23:46:063.179 GMT] logStreamInfo(): 	Content type: video/h265 
[DEBUG] [04-05-2020 00:23:46:063.204 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [04-05-2020 00:23:46:063.227 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [04-05-2020 00:23:46:063.252 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [04-05-2020 00:23:46:063.275 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [04-05-2020 00:23:46:063.352 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [04-05-2020 00:23:46:063.437 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [04-05-2020 00:23:46:063.504 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [04-05-2020 00:23:46:063.571 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [04-05-2020 00:23:46:063.637 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [04-05-2020 00:23:46:063.704 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [04-05-2020 00:23:46:063.771 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [04-05-2020 00:23:46:063.839 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [04-05-2020 00:23:46:063.905 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [04-05-2020 00:23:46:063.981 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [04-05-2020 00:23:46:064.049 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [04-05-2020 00:23:46:064.115 GMT] logStreamInfo(): Track list
[DEBUG] [04-05-2020 00:23:46:064.160 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [04-05-2020 00:23:46:064.225 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [04-05-2020 00:23:46:064.290 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [04-05-2020 00:23:46:064.357 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [04-05-2020 00:23:46:064.422 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [04-05-2020 00:23:46:064.660 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [04-05-2020 00:23:47:087.514 GMT] writeHeaderCallback(): RequestId: afbde1b4-8129-4129-8b65-c6bb31bc5c36
[DEBUG] [04-05-2020 00:23:47:087.954 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.586210606267E9,"DataRetentionInHours":1,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:yyyyy:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:yyyyy:stream/stream-xxxx/aaaaaaa","StreamName":"stream-xxxx","Version":"a5TPP3tumlG7Lwer37IW"}}
[INFO ] [04-05-2020 00:23:47:097.630 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [04-05-2020 00:23:47:097.891 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [04-05-2020 00:23:47:098.115 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [04-05-2020 00:23:47:098.427 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [04-05-2020 00:23:48:121.377 GMT] writeHeaderCallback(): RequestId: 044469e3-38cb-4675-ba69-0cf64f74bb29
[... cut ...]
[INFO ] [04-05-2020 00:23:48:588.353 GMT] Freeing Kinesis Video Stream stream-xxxx
[INFO ] [04-05-2020 00:23:48:588.462 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[DEBUG] [04-05-2020 00:23:48:588.533 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [04-05-2020 00:23:48:990.681 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [04-05-2020 00:23:48:990.884 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down

```

dvsantana, 2020-05-04:
The stream-format was causing the issue, it seems.
For this, I switched the capsfilter to:

```sh
        GstCaps *h265_caps = gst_caps_new_simple("video/x-h265",
                                                 "stream-format", G_TYPE_STRING, "hvc1", //or hev1 or byte-stream
                                                 "alignment", G_TYPE_STRING, "au",
                                                 NULL);
```

Will run a few more tests and will close the issue if everything works as it should.

disa6302, 2020-05-04:
Sure @dvsantana ! Thanks.

dvsantana, 2020-05-04:
It still isn't working. It looks like the stream is being sent, but somehow it is not working.

```sh
[INFO ] [04-05-2020 22:24:30:524.897 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [04-05-2020 22:24:30:525.103 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [04-05-2020 22:24:30:525.138 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [04-05-2020 22:24:30:525.222 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [04-05-2020 22:24:30:525.318 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [04-05-2020 22:24:30:525.394 GMT] getSecurityTokenHandler invoked
[DEBUG] [04-05-2020 22:24:30:525.500 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1588631070525449262 Expiration: 0
[INFO ] [04-05-2020 22:24:30:525.566 GMT] New credentials expiration is 1588634670
[DEBUG] [04-05-2020 22:24:30:525.622 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [04-05-2020 22:24:30:525.656 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [04-05-2020 22:24:30:525.703 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [04-05-2020 22:24:30:525.736 GMT] clientReadyHandler invoked
[DEBUG] [04-05-2020 22:24:30:525.771 GMT] Client is ready
[INFO ] [04-05-2020 22:24:30:525.872 GMT] Creating Kinesis Video Stream stream-xxxx
[INFO ] [04-05-2020 22:24:30:525.914 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [04-05-2020 22:24:30:525.947 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [04-05-2020 22:24:30:525.972 GMT] logStreamInfo(): 	Stream name: stream-xxxx 
[DEBUG] [04-05-2020 22:24:30:525.998 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [04-05-2020 22:24:30:526.028 GMT] logStreamInfo(): 	Content type: video/h265 
[DEBUG] [04-05-2020 22:24:30:526.054 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [04-05-2020 22:24:30:526.080 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [04-05-2020 22:24:30:526.104 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [04-05-2020 22:24:30:526.129 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [04-05-2020 22:24:30:526.152 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [04-05-2020 22:24:30:526.176 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [04-05-2020 22:24:30:526.201 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [04-05-2020 22:24:30:526.226 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [04-05-2020 22:24:30:526.250 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [04-05-2020 22:24:30:526.275 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [04-05-2020 22:24:30:526.300 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [04-05-2020 22:24:30:526.325 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [04-05-2020 22:24:30:526.349 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [04-05-2020 22:24:30:526.373 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [04-05-2020 22:24:30:526.397 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [04-05-2020 22:24:30:526.421 GMT] logStreamInfo(): Track list
[DEBUG] [04-05-2020 22:24:30:526.444 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [04-05-2020 22:24:30:526.467 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [04-05-2020 22:24:30:526.490 GMT] logStreamInfo(): 	Codec id: HEVC
[DEBUG] [04-05-2020 22:24:30:526.514 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [04-05-2020 22:24:30:526.538 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [04-05-2020 22:24:30:526.736 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [04-05-2020 22:24:31:545.296 GMT] writeHeaderCallback(): RequestId: e2541e9b-3c74-421d-9bb9-4bb27812575c
[DEBUG] [04-05-2020 22:24:31:545.663 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.586210606267E9,"DataRetentionInHours":1,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:yyyyyy:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:yyyyyy:stream/stream-xxxx/zzzzzzz","StreamName":"stream-xxxx","Version":"a5TPP3tumlG7Lwer37IW"}}
[INFO ] [04-05-2020 22:24:31:555.256 GMT] describeStreamResultEvent(): Describe stream result event.
[WARN ] [04-05-2020 22:24:31:555.437 GMT] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ] [04-05-2020 22:24:31:555.503 GMT] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG] [04-05-2020 22:24:31:555.577 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [04-05-2020 22:24:32:561.514 GMT] writeHeaderCallback(): RequestId: 4ea2bc89-7afa-4d7d-98b4-217e15a3cfee
[DEBUG] [04-05-2020 22:24:32:561.806 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-aaaaaaaa.kinesisvideo.us-west-2.amazonaws.com"}
[... cut ...]
[WARN ] [04-05-2020 22:25:08:890.570 GMT] continuousRetryStreamErrorReportHandler(): Reporting stream error. Errored timecode: 248820000 Status: 0x25200005e
```

With the following line in the GST_DEBUG=4

`0:00:57.043961846  2061   0xbcbef0 FIXME           rtph265depay gstrtph265depay.c:1278:gst_rtp_h265_depay_process:<depay> Assuming DONL field is not present`

From this, there is no support for hvc1. But when I use the other possibilities, which are hev1 and byte-stream, I get an error as well, as if it is not supported.

chehefen, 2020-05-04:
rtph265depay should support hvc1 as described in its template:
```
GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstRTPBaseDepayload
                         +----GstRtpH265Depay

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      application/x-rtp
                  media: video
             clock-rate: 90000
          encoding-name: H265

  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-h265
          stream-format: hvc1
              alignment: au
      video/x-h265
          stream-format: byte-stream
              alignment: { (string)nal, (string)au }
```

dvsantana, 2020-05-05:
@chehefen yes, that's where I got it from, but I'm still getting that FIXME. Should I just ignore it? It looks like the stream is transmitting.

zhiyua-git, 2020-05-05:
Hi @dvsantana ,

From your previous log, seems you were hitting ERROR ACK max fragment duration exceeded, which means your stream doesn't provide I-frame for 10 seconds. Can you check if you can tune the rtsp source or through the gstreamer pipeline to generate I-frame more often? Thanks.

dvsantana, 2020-05-05:
@zhiyua-git , I've fixed that already, but it sure was the problem in the previous log, thanks for noticing. Now I'm able to send the stream, however, I'm still getting that FIXME. It doesn't seem to affect the stream since I'm getting all the confirmations on the logs: 

```sh
[DEBUG] [04-05-2020 22:57:02:465.016 GMT] postReadCallback(): Wrote 61149 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [04-05-2020 22:57:02:768.620 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [04-05-2020 22:57:03:085.066 GMT] postReadCallback(): Wrote 40095 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [04-05-2020 22:57:03:317.636 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [04-05-2020 22:57:05:805.983 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [04-05-2020 22:57:06:542.113 GMT] Kinesis Video client and stream metrics
    >> Overall storage byte size: 134217728
    >> Available storage byte size: 132060775
    >> Allocated storage byte size: 2156953
    >> Total view allocation byte size: 144080
    >> Total streams frame rate (fps): 14
    >> Total streams transfer rate (bps): 13491080 (13174 Kbps)
    >> Current view duration (ms): 3002
    >> Overall view duration (ms): 5162
    >> Current view byte size: 1575304
    >> Overall view byte size: 2155153
    >> Current frame rate (fps): 14.7956
    >> Current transfer rate (bps): 13491080 (13174 Kbps)
[DEBUG] [04-05-2020 22:57:07:444.277 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
```

Also, I'm not able to see it in the KVS console, but that was expected as it says it only allows H264, or in my application, but that is probably my fault (it was working with H264).

zhiyua-git, 2020-05-05:
@dvsantana , you will need to also update the codec id https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/824de900c2d4b42a403319ace5de9d68ec88b171/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L46 to "**V_MPEGH/ISO/HEVC**" as backend relies on it to check if it is h264 or h265. And you can try to verify the playback on https://aws-samples.github.io/amazon-kinesis-video-streams-media-viewer/.

dvsantana, 2020-05-05:
> to "**V_MPEGH/ISO/HEVC**" as backend relies on it to check if it is h264 or h265. And you can try to verify the playback on https://aws-samples.github.io/amazon-kinesis-video-streams-media-viewer/.

@zhiyua-git I changed it, the FIXME is still happening `0:00:57.043961846 2061 0xbcbef0 FIXME rtph265depay gstrtph265depay.c:1278:gst_rtp_h265_depay_process:<depay> Assuming DONL field is not present`

To verify that playback, to I need to be on a specific browser?

dvsantana, 2020-05-05:
Isn't this supposed to be working? Or am I doing something wrong?

[image]

zhiyua-git, 2020-05-05:
You can switch different combinations of streaming protocols and players, some of them might not support h265 well.

dvsantana, 2020-05-05:
Tried them all, the only one that got close to working was:

[image]

zhiyua-git, 2020-05-05:
Can you specify env var KVS_DEBUG_DUMP_DATA_FILE_DIR to a directory and check the output mkv to see if it is playable in VLC player? If it plays just fine, please send a short one to Kinesis-video-support@amazon.com so we can take a close look into it to understand what is failing. Thanks.

dvsantana, 2020-05-05:
@zhiyua-git thank you so much for all your help!

I tried setting the env var `export KVS_DEBUG_DUMP_DATA_FILE_DIR=`pwd`/dump
`
But after that, I got the following warning: `[WARN ] [05-05-2020 12:53:10:484.966 GMT] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x00000009 decoding timestamp: 3091829 presentation timestamp: 15886831904840956`

dvsantana, 2020-05-05:
@zhiyua-git thank you so much for your help!

I was able to see the mkv on the VLC player, so I sent the output mkv to the email above.

MixMasterMitch, 2020-05-11:
@dvsantana H.265 playback is not supported in Chrome. We put your MKV to a stream and viewed it in Safari via the Media Stream Viewer test page without issue.

However, if I take the HLS URL and paste it directly into the Safari URL bar, it does not play.

A helpful debugging tool is Apple's `mediastreamvalidator`. It revealed a couple things:

```
Error: Unsupported HEVC profile_space
--> Detail:  profile_space: 1 ('A')
--> Source:  https://b-647daf39.kinesisvideo.us-west-2.amazonaws.com/hls/v1/getHLSMasterPlaylist.m3u8
--> Compare: getHLSMediaPlaylist.m3u8?TrackNumber=1
```
^ I am not sure why this is a problem since the media is able to be decoded.

```
Error: Playlist vs segment duration mismatch
--> Detail:  Segment duration 0.9930, Playlist duration: 3.0660
--> Source:  getHLSMediaPlaylist.m3u8?TrackNumber=1 - getMP4MediaFragment.mp4?FragmentNumber=91343852333200022166869999398375715807955315381&TrackNumber=1
```
^I don't know how the segment duration is being calculated by Apple, but I can see that the frame timecodes are not at a consistent interval, so that is probably causing the Apple tool to have issues.

dvsantana, 2020-05-11:
@MixMasterMitch, thank you so much for your answer, it cleared things a bit here. I tried in different browsers, but for the safari I had to use the iOS safari, so that could be the reason why I couldn't get it to work.

As for the other issues I don't know how the segment and the playlist could have different duration, but I don't think that is a main issue. 

Again, thank you all for taking your time and helping me, I learned a lot on the way.
I will close this issue now, as the main problem was resolved already.

closed 2020-05-11 by dvsantana