awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #433: Nginx -> gstreamer -> kinesis

mwildehahn, 2020-04-29:
I don't think this is related to this library, but hoping y'all can point me in the right direction on this. I'm trying to ingest an RTMP stream from something like OBS (https://obsproject.com/) into Kinesis. Since Kinesis doesn't directly support RTMP, I'm running nginx with the nginx-rtmp module.

I finally got everything installed and hooked up correctly but with the following nginx file:

```
load_module "modules/ngx_rtmp_module.so";
error_log /var/log/nginx/error.log debug;

events {}

rtmp {
    server {
        listen 1935;

        application stream {
            live on;

            exec /opt/kvs-producer $app $name;
            exec_kill_signal term;
        }
    }
}
```

and the following `kvs-producer` file:

```
#!/bin/bash

on_die ()
{
    echo terminated >> /tmp/$1-$2.log
    # kill all children
    pkill -KILL -P $$
}

trap 'on_die' TERM

echo started >> /tmp/$1-$2.log

LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH /usr/bin/gst-launch-1.0 -v rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" ! flvdemux name=demux demux.video ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=512 aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" >> /tmp/$1-$2.log &

wait
```

I'm just seeing this loop of logs when streaming:

```Setting pipeline to PAUSED ...
[INFO ] [29-04-2020 06:50:03:275.910 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [29-04-2020 06:50:03:275.958 GMT] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ] [29-04-2020 06:50:03:275.974 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [29-04-2020 06:50:03:276.000 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [29-04-2020 06:50:03:276.030 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [29-04-2020 06:50:03:276.051 GMT] getSecurityTokenHandler invoked
[DEBUG] [29-04-2020 06:50:03:276.074 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1588143003276066300 Expiration: 0
[DEBUG] [29-04-2020 06:50:03:276.102 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [29-04-2020 06:50:03:276.114 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [29-04-2020 06:50:03:276.136 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [29-04-2020 06:50:03:276.150 GMT] clientReadyHandler invoked
[INFO ] [29-04-2020 06:50:03:276.255 GMT] try creating stream
[INFO ] [29-04-2020 06:50:03:276.329 GMT] Creating Kinesis Video Stream test2
[INFO ] [29-04-2020 06:50:03:276.352 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [29-04-2020 06:50:03:276.505 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [29-04-2020 06:50:03:276.531 GMT] logStreamInfo(): 	Stream name: test2
[DEBUG] [29-04-2020 06:50:03:276.543 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [29-04-2020 06:50:03:276.554 GMT] logStreamInfo(): 	Content type: video/h264
[DEBUG] [29-04-2020 06:50:03:276.568 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [29-04-2020 06:50:03:276.577 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [29-04-2020 06:50:03:276.591 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [29-04-2020 06:50:03:276.608 GMT] logStreamInfo(): 	Use frame timecode: Yes
[... 102 lines cut ...]
[DEBUG] [29-04-2020 06:50:10:953.453 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [29-04-2020 06:50:10:954.452 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [29-04-2020 06:50:10:954.502 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [29-04-2020 06:50:10:954.544 GMT] getStreamingTokenHandler invoked
[DEBUG] [29-04-2020 06:50:10:954.610 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1588143010954599700 Expiration: 18446744073709551615
[INFO ] [29-04-2020 06:50:10:954.642 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [29-04-2020 06:50:10:954.674 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [29-04-2020 06:50:10:954.714 GMT] streamReadyHandler invoked
Stream is ready
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Execution ended after 0:00:00.081006000
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
[INFO ] [29-04-2020 06:50:11:038.449 GMT] Freeing Kinesis Video Stream test2
[INFO ] [29-04-2020 06:50:11:038.489 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG] [29-04-2020 06:50:11:038.516 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
```

The stream gets created in kinesis but no frames ever get sent. Do you know if this is an issue with the RTMP module or am I invoking gstreamer incorrectly?

I pulled some of this context from: https://forums.aws.amazon.com/thread.jspa?threadID=284176

MushMal, 2020-04-29:
@mwildehahn It's not immediately clear why the GStreamer tears down the pipeline. Could you set GST_DEBUG=4 env variable to make GStreamer to spew debug logs so we can see if anything is misconfigured?

mwildehahn, 2020-04-29:
I added that but looks like it is the same output as above (which includes `[DEBUG]` logs). I added some additional logs and rebuilt:

```
[WRAPPER_DEBUG] STARTED
Setting pipeline to PAUSED ...
[INFO ] [29-04-2020 16:26:54:694.463 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [29-04-2020 16:26:54:694.501 GMT] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ] [29-04-2020 16:26:54:694.517 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [29-04-2020 16:26:54:694.543 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [29-04-2020 16:26:54:694.566 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[DEBUG] [29-04-2020 16:26:54:694.586 GMT] getSecurityTokenHandler invoked
[DEBUG] [29-04-2020 16:26:54:694.608 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1588177614694600800 Expiration: 0
[DEBUG] [29-04-2020 16:26:54:694.635 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[INFO ] [29-04-2020 16:26:54:694.651 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [29-04-2020 16:26:54:694.672 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
[DEBUG] [29-04-2020 16:26:54:694.686 GMT] clientReadyHandler invoked
[INFO ] [29-04-2020 16:26:54:694.735 GMT] try creating stream
[INFO ] [29-04-2020 16:26:54:694.762 GMT] Creating Kinesis Video Stream test2
[INFO ] [29-04-2020 16:26:54:694.780 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG] [29-04-2020 16:26:54:694.795 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [29-04-2020 16:26:54:694.809 GMT] logStreamInfo(): 	Stream name: test2
[DEBUG] [29-04-2020 16:26:54:694.822 GMT] logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
[DEBUG] [29-04-2020 16:26:54:694.836 GMT] logStreamInfo(): 	Content type: video/h264
[DEBUG] [29-04-2020 16:26:54:694.850 GMT] logStreamInfo(): 	Max latency (100ns): 600000000
[DEBUG] [29-04-2020 16:26:54:694.864 GMT] logStreamInfo(): 	Fragment duration (100ns): 20000000
[DEBUG] [29-04-2020 16:26:54:694.877 GMT] logStreamInfo(): 	Key frame fragmentation: Yes
[DEBUG] [29-04-2020 16:26:54:694.891 GMT] logStreamInfo(): 	Use frame timecode: Yes
[DEBUG] [29-04-2020 16:26:54:694.904 GMT] logStreamInfo(): 	Absolute frame timecode: Yes
[DEBUG] [29-04-2020 16:26:54:694.918 GMT] logStreamInfo(): 	Nal adaptation flags: 0
[DEBUG] [29-04-2020 16:26:54:694.931 GMT] logStreamInfo(): 	Average bandwith (bps): 4194304
[DEBUG] [29-04-2020 16:26:54:694.945 GMT] logStreamInfo(): 	Framerate: 25
[DEBUG] [29-04-2020 16:26:54:694.958 GMT] logStreamInfo(): 	Buffer duration (100ns): 1200000000
[DEBUG] [29-04-2020 16:26:54:694.972 GMT] logStreamInfo(): 	Replay duration (100ns): 400000000
[DEBUG] [29-04-2020 16:26:54:694.985 GMT] logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
[DEBUG] [29-04-2020 16:26:54:694.999 GMT] logStreamInfo(): 	Store Pressure Policy: 1
[DEBUG] [29-04-2020 16:26:54:695.012 GMT] logStreamInfo(): 	View Overflow Policy: 1
[DEBUG] [29-04-2020 16:26:54:695.026 GMT] logStreamInfo(): 	Segment UUID: NULL
[DEBUG] [29-04-2020 16:26:54:695.040 GMT] logStreamInfo(): 	Frame ordering mode: 0
[DEBUG] [29-04-2020 16:26:54:695.053 GMT] logStreamInfo(): Track list
[DEBUG] [29-04-2020 16:26:54:695.066 GMT] logStreamInfo(): 	Track id: 1
[DEBUG] [29-04-2020 16:26:54:695.079 GMT] logStreamInfo(): 	Track name: kinesis_video
[DEBUG] [29-04-2020 16:26:54:695.093 GMT] logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
[DEBUG] [29-04-2020 16:26:54:695.106 GMT] logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [29-04-2020 16:26:54:695.120 GMT] logStreamInfo(): 	Track cpd: NULL
[DEBUG] [29-04-2020 16:26:54:695.272 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[INFO ] [29-04-2020 16:26:54:916.674 GMT] writeHeaderCallback(): RequestId: 18f3040d-f7ae-40fb-9d88-d67a92e82ef1
[DEBUG] [29-04-2020 16:26:54:916.758 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.588143003893E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:934411350416:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:934411350416:stream/test2/1588143003893","StreamName":"test2","Version":"mxQ9LjWWxXhuou4LlHg2"}}
[INFO ] [29-04-2020 16:26:54:917.468 GMT] describeStreamResultEvent(): Describe stream result event.
[DEBUG] [29-04-2020 16:26:54:917.497 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
[INFO ] [29-04-2020 16:26:55:123.517 GMT] writeHeaderCallback(): RequestId: 1f448aa1-5d6a-4f43-8de1-da2e4f28c816
[DEBUG] [29-04-2020 16:26:55:123.602 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ] [29-04-2020 16:26:55:124.275 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [29-04-2020 16:26:55:124.320 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[DEBUG] [29-04-2020 16:26:55:124.349 GMT] getStreamingTokenHandler invoked
[... cut ...]  streaming to kinesis?

Any suggestions on more places to add logging to see what is going on?

I simplified my invocation of gstreamer to just:

`/usr/bin/gst-launch-1.0 -v rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=512 aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" restart-on-error=0`

mwildehahn, 2020-04-29:
I played with the gstreamer invocation and ended up with this:

```
LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH GST_DEBUG=4 DEBUG_DUMP_FRAME_INFO=1 \
    /usr/bin/gst-launch-1.0 -v \
    rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" \
    ! flvdemux name=demux demux.video \
    ! h264parse \
    ! video/x-h264, format=avc,alignment=au \
    ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=512 aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" restart-on-error=0 \
    >> /tmp/$1-$2.log &
```

This started streaming to kinesis, which is great! But it only happened after I hit stream in OBS several times. Here are the full logs: https://gist.github.com/mwildehahn/8b1cee9cb49fe91f6ae55198309d1a8b. It creates the stream successfully when it gets initialized, and then just loops through restarting numerous times. Eventually it starts picking up packets and forwarding them to kinesis: https://gist.github.com/mwildehahn/8b1cee9cb49fe91f6ae55198309d1a8b#file-debug-gstreamer-L857. 

I thought this might be some buffer setting, but it doesn't seem related to the length of how long I'm streaming, more in re-publishing numerous times?

mwildehahn, 2020-04-29:
I've been trying to reproduce how to get this to work, still can't see a pattern. It only seems to work if I start/stop streaming repeatedly, then trigger:

```
[WRAPPER_DEBUG] STARTED
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, stream-format=(string)avc, width=(int)1792, height=(int)1120, framerate=(fraction)60/1, codec_data=(buffer)0164002affe100196764002aacd94070023684000003000400000301e23c60c65801000468efbcb0, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, alignment=(string)au, profile=(string)high, level=(string)4.2
[INFO ] [29-04-2020 21:59:47:415.874 GMT] kinesisVideoStreamFormatChanged(): Stream format changed.
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-h264, format=(string)avc, alignment=(string)au, stream-format=(string)avc, width=(int)1792, height=(int)1120, framerate=(fraction)60/1, codec_data=(buffer)0164002affe100196764002aacd94070023684000003000400000301e23c60c65801000468efbcb0, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)high, level=(string)4.2
[DEBUG] [29-04-2020 21:59:47:416.009 GMT] Stream started
/GstPipeline:pipeline0/GstKvsSink:kvssink0.GstPad:video_0: caps = video/x-h264, format=(string)avc, alignment=(string)au, stream-format=(string)avc, width=(int)1792, height=(int)1120, framerate=(fraction)60/1, codec_data=(buffer)0164002affe100196764002aacd94070023684000003000400000301e23c60c65801000468efbcb0, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)high, level=(string)4.2
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-h264, stream-format=(string)avc, width=(int)1792, height=(int)1120, framerate=(fraction)60/1, codec_data=(buffer)0164002affe100196764002aacd94070023684000003000400000301e23c60c65801000468efbcb0, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, alignment=(string)au, profile=(string)high, level=(string)4.2
[DEBUG] [29-04-2020 21:59:47:416.207 GMT] Handling buffer
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, stream-format=(string)avc, width=(int)1792, height=(int)1120, framerate=(fraction)60/1, codec_data=(buffer)0164002affe100196764002aacd94070023684000003000400000301e23c60c65801000468efbcb0
[DEBUG] [29-04-2020 21:59:47:416.224 GMT] Dropping frame with flag %u1088
[DEBUG] [29-04-2020 21:59:47:435.324 GMT] Handling buffer
[DEBUG] [29-04-2020 21:59:47:435.369 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 21:59:47:435.395 GMT] pts: 15881975874353636, dts: 20000, duration: 0, size: 39899, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 21:59:47:435.429 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
```

```
[INFO ] [29-04-2020 21:59:47:415.874 GMT] kinesisVideoStreamFormatChanged(): Stream format changed.
```

which then results in:

```
[DEBUG] [29-04-2020 21:59:47:416.009 GMT] Stream started
```

It seems to be attempting to push to kinesis at least, but seeing a lot of warnings around dropped frames:

```
[DEBUG] [29-04-2020 22:15:51:725.454 GMT] Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 534861509
	>> Allocated storage byte size: 2009403
	>> Total view allocation byte size: 6080
	>> Total streams frame rate (fps): 744
	>> Total streams transfer rate (bps): 19447000 (18991 Kbps)
	>> Current view duration (ms): 1833
	>> Overall view duration (ms): 1850
	>> Current view byte size: 1964440
	>> Overall view byte size: 2004027
	>> Current frame rate (fps): 744.193
	>> Current transfer rate (bps): 19447000 (18991 Kbps)
[DEBUG] [29-04-2020 22:15:51:728.260 GMT] Handling buffer
[DEBUG] [29-04-2020 22:15:51:728.349 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 22:15:51:728.366 GMT] pts: 15881976836003636, dts: 961490000, duration: 0, size: 59152, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 22:15:51:737.871 GMT] Handling buffer
[DEBUG] [29-04-2020 22:15:51:737.910 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 22:15:51:737.925 GMT] pts: 15881976835663636, dts: 961650000, duration: 0, size: 24188, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 22:15:51:770.398 GMT] Handling buffer
[DEBUG] [29-04-2020 22:15:51:770.451 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 22:15:51:770.476 GMT] pts: 15881976835503636, dts: 961820000, duration: 0, size: 69, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 22:15:51:781.846 GMT] Handling buffer
[DEBUG] [29-04-2020 22:15:51:781.884 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 22:15:51:781.901 GMT] pts: 15881976835833636, dts: 961990000, duration: 0, size: 286, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 22:15:51:789.975 GMT] Handling buffer
[DEBUG] [29-04-2020 22:15:51:790.023 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 22:15:51:790.047 GMT] pts: 15881976836663636, dts: 962150000, duration: 0, size: 57224, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 22:15:51:835.480 GMT] Handling buffer
[DEBUG] [29-04-2020 22:15:51:835.514 GMT] Kinesis Put Frame
[DEBUG] [29-04-2020 22:15:51:835.535 GMT] pts: 15881976836333636, dts: 962320000, duration: 0, size: 24502, trackId: 1, isKey: 0
[DEBUG] [29-04-2020 22:15:51:835.592 GMT] Handling buffer
[... cut ...]
[DEBUG] [29-04-2020 22:15:51:942.483 GMT] droppedFrameReportHandler invoked
[WARN ] [29-04-2020 22:15:51:942.497 GMT] Reported droppedFrame callback for stream handle 94235860402416. Dropped frame timecode in 100ns: 944820000
[WARN ] [29-04-2020 22:15:51:942.511 GMT] viewItemRemoved(): Reporting a dropped frame/fragment.
[DEBUG] [29-04-2020 22:15:51:942.525 GMT] droppedFrameReportHandler invoked
[WARN ] [29-04-2020 22:15:51:942.539 GMT] Reported droppedFrame callback for stream handle 94235860402416. Dropped frame timecode in 100ns: 944990000
```

disa6302, 2020-04-29:
@mwildehahn , 

I am not quite sure whats happening. I will have to spend some time to figure out what might be going on. Will get back as soon as I have something. Thank you for your patience in the meantime!

mwildehahn, 2020-04-29:
Thank you! Let me know if I can provide anything else.

mwildehahn, 2020-04-30:
I was able to get logs: https://gist.github.com/mwildehahn/58aaa5fcf4a75b3f64fbbf6f5bef29f9

I think this might be an issue with my gstreamer pipeline. I'm trying to route both audio and video to kinesis and using:

```
LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH GST_DEBUG=4 DEBUG_DUMP_FRAME_INFO=1 \
    /usr/bin/gst-launch-1.0 -v \
    rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" \
    ! flvdemux name=demux demux.video \
    ! h264parse \
    ! video/x-h264, format=avc,alignment=au \
    ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=1024 name=sink \
        aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" restart-on-error=0 \
    demux.audio ! audioconvert ! avenc_aac ! sink. &>> /tmp/$1-$2.log &
```

mwildehahn, 2020-04-30:
Ah! I'm so close but this is my first time using gstreamer so I must be missing something about setting this up.

I updated to this:

```
LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH GST_DEBUG=4 DEBUG_DUMP_FRAME_INFO=1 \
    /usr/bin/gst-launch-1.0 -v \
    rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" \
    ! flvdemux name=demux demux.video ! queue ! decodebin ! h264parse \
    ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=1024 name=sink \
        aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" restart-on-error=0 \
    demux.audio ! queue ! decodebin ! audioconvert ! avenc_aac ! sink. &>> $LOG_FILE &
```

based off some recommendations here: https://github.com/matthew1000/gstreamer-cheat-sheet/blob/master/rtmp.md.

This is the new kvs.log (which I hadn't been creating previously because I hadn't configured the `log-config`): https://gist.github.com/mwildehahn/ed18c2af2c876e9b84a1f357ede859ba

These are the gst logs for this stream: https://gist.github.com/mwildehahn/d76a91c92f67daf95f47f89b596ec71b

I'm no longer getting the errors for `not-linked` and it says "Stream started", but even with `DEBUG_DUMP_FRAME_INFO=1`, I'm not seeing any of the frame info being logged like I was able to see intermittently before. The process just seems to hang in "Stream started" and I don't see any frames being put to kinesis.

I know this is probably an issue with the gstreamer plugin and not the kvssink, but any guidance would be much appreciated! Thank you!

disa6302, 2020-04-30:
@mwildehahn , 

From the logs, I see this error: 
`0:00:01.163964500    34 0x55aa35abacc0 INFO        GST_ELEMENT_PADS gstelement.c:912:gst_element_get_static_pad: no such pad 'sink' in element "rtmpsrc"`

This is what is probably causing issues. Because, it looks like the other pads are getting created just fine.

mwildehahn, 2020-04-30:
Hm, I thought that might be due to it not being available there but being dynamically linked later.

I made some tweaks to the gstreamer invocation and ended up with this:

```
LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH GST_DEBUG=4 DEBUG_DUMP_FRAME_INFO=1 \
    /usr/bin/gst-launch-1.0 -vm \
    rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" \
    ! flvdemux name=demux demux.video \
    ! h264parse \
    ! video/x-h264, format=avc,alignment=au \
    ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=512 name=kvs \
        aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" restart-on-error=0 \
    demux.audio ! decodebin ! kvs. &>> $LOG_FILE &
```

Here are the resulting logs: https://gist.github.com/mwildehahn/abdd1eee54acbf75fb5001800a612cf1

This consistently works for video! Which is awesome! But I'm not sure if it is working for audio. In the AWS console I downloaded the video clip as an mp4 and there was no sound. Is it possible to see if I'm not sending the audio along correctly?

Thank you for all your help!

disa6302, 2020-05-03:
@mwildehahn , 

Apologies for the delayed response. Is your audio pipeline working now?

mwildehahn, 2020-05-05:
It still isn't playing audio through the kinesis console, I just see video. Any steps to debug?

zhiyua-git, 2020-05-05:
Hi @mwildehahn ,

You can use env var KVS_DEBUG_DUMP_DATA_FILE_DIR to set a directory and gather the mkv generated by kinesis video producer SDK, and check if it plays audio and video in VLC player, if it plays fine, please send the file to kinesis-video-support@amazon.com so we can take a close look. Thanks.

mwildehahn, 2020-05-05:
Thanks, just sent an email with logs + mkv that was generated. There is no sound in the mkv file which makes me think something is wrong with the audio pipeline. I've tried making some adjustments to it but they all seem to prevent video + audio, where the current configuration at least allows for videos.

MushMal, 2020-05-05:
Thanks @mwildehahn. We are looking at the info and will let you know asap.

MushMal, 2020-05-05:
Here is a preliminary info:

* The generated MKV has a single video track - no audio as noted.
* The logs indicate that the StreamInfo is created with a single video track.
* No errors in the streaming (Good), however, initially, we get a latency pressure event and re-set the connection and the new connection handles the streaming OK with no buffer.
* Initially, the audio pad fails to load so the KVS plugin gets video pad only - likely the reason why there is video track only.
0:00:00.650923400 [333m   26[00m 0x56294f024600 [36mINFO   [00m [00;01;37;41m    GST_ELEMENT_PADS gstutils.c:1771:gst_element_link_pads_full:[00m trying to link element demux:audio to element decodebin0:(any)
0:00:00.650940100 [333m   26[00m 0x56294f024600 [36mINFO   [00m [00;01;37;41m    GST_ELEMENT_PADS gstelement.c:912:gst_element_get_static_pad:[00m no such pad 'audio' in element "demux"
0:00:00.650957200 [333m   26[00m 0x56294f024600 [36mINFO   [00m [00;04m             default gstutils.c:2161:gst_element_link_pads_filtered:[00m Could not link pads: demux:audio - decodebin0:(null)
0:00:00.650976800 [333m   26[00m 0x56294f024600 [36mINFO   [00m [00;01;37;41m       
* Seems that the audio gets prepared async and gets ready before the pipeline is fully moved to playing state - aka - when the first frame arrives. However, it seems that the KVS plugin has already initialized and is in READY state itself.

chehefen, 2020-05-05:
Hi @mwildehahn, I see these two lines in your latest log
```
0:00:00.410279100    38 0x556fbc245600 INFO        GST_ELEMENT_PADS gstutils.c:1771:gst_element_link_pads_full: trying to link element demux:audio to element decodebin0:(any)
0:00:00.410295600    38 0x556fbc245600 INFO        GST_ELEMENT_PADS gstelement.c:912:gst_element_get_static_pad: no such pad 'audio' in element "demux"
```
It seems that the demuxer did not detect any audio in you rtmp stream. Can you double check if you are producing audio? Thanks.

mwildehahn, 2020-05-05:
Here are the nginx logs from nginx-rtmp: https://gist.github.com/mwildehahn/cfbc6e357ae3b36370d2f8fd422cab2f. It looks like it is receiving audio from these lines:

```
2020/05/05 20:45:22 [debug] 17#17: *1 RTMP mheader fmt=0 audio (8) time=0+0 mlen=4 len=0 msid=1
2020/05/05 20:45:22 [debug] 17#17: *1 RTMP recv audio (8) csid=4 timestamp=0 mlen=4 msid=1 nbufs=1
2020/05/05 20:45:22 [debug] 17#17: *1 nhandlers: 5
2020/05/05 20:45:22 [debug] 17#17: *1 calling handler 0
2020/05/05 20:45:22 [debug] 17#17: *1 codec: aac header af001210
2020/05/05 20:45:22 [debug] 17#17: *1 codec: aac header profile=2, sample_rate=44100, chan_conf=2
2020/05/05 20:45:22 [debug] 17#17: malloc: 0000562618D49040:4214
2020/05/05 20:45:22 [debug] 17#17: *1 calling handler 1
2020/05/05 20:45:22 [debug] 17#17: *1 calling handler 2
2020/05/05 20:45:22 [debug] 17#17: *1 create: stream_begin msid=1
2020/05/05 20:45:22 [debug] 17#17: malloc: 0000562618D4A0C0:4214
2020/05/05 20:45:22 [debug] 17#17: *1 RTMP prep user (4) fmt=0 csid=2 timestamp=0 mlen=6 msid=0 nbufs=1
2020/05/05 20:45:22 [debug] 17#17: *1 live: set active=1
2020/05/05 20:45:22 [debug] 17#17: *1 live: audio packet timestamp=0
2020/05/05 20:45:22 [debug] 17#17: *1 RTMP prep audio (8) fmt=1 csid=6 timestamp=0 mlen=4 msid=1 nbufs=1
```

Here are the updated KVS logs: https://gist.github.com/mwildehahn/61ffde87a858397c83be0596b9640b4d

It looks like it is able to do a delayed link for audio: 
```
0:00:01.444743100    19 0x55ab019606a0 INFO            GST_PIPELINE grammar.y:520:gst_parse_found_pad: trying delayed linking one pad pad  audio of GstFlvDemux named demux to some pad of GstDecodeBin named decodebin0
0:00:01.444761500    19 0x55ab019606a0 INFO        GST_ELEMENT_PADS gstutils.c:1771:gst_element_link_pads_full: trying to link element demux:audio to element decodebin0:(any)
0:00:01.444770400    19 0x55ab019606a0 INFO        GST_ELEMENT_PADS gstelement.c:915:gst_element_get_static_pad: found pad demux:audio
0:00:01.444785400    19 0x55ab019606a0 INFO                GST_PADS gstutils.c:1034:gst_pad_check_link: trying to link demux:audio and decodebin0:sink
0:00:01.444809500    19 0x55ab019606a0 INFO                GST_PADS gstutils.c:1587:prepare_link_maybe_ghosting: demux and decodebin0 in same bin, no need for ghost pads
0:00:01.444827000    19 0x55ab019606a0 INFO                GST_PADS gstpad.c:2377:gst_pad_link_prepare: trying to link demux:audio and decodebin0:sink
0:00:01.444845300    19 0x55ab019606a0 INFO                GST_PADS gstpad.c:2585:gst_pad_link_full: linked demux:audio and decodebin0:sink, successful
```

I can provide a docker setup with a couple files to reproduce this if it is helpful!

chehefen, 2020-05-05:
Can you try this patch? [patch.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4583826/patch.txt)
Also I think you should replace decodebin with aacparse after demux.audio because kvssink dont accept raw audio frames.

mwildehahn, 2020-05-06:
Updated logs after applying patch and swapping to `aacparse`: https://gist.github.com/mwildehahn/b23aea687e27173c732313a38dcb258d

It looks like something crashed, no mkv file is generated and it doesn't say anything is happening after "Stream is ready".

I haven't seen this before:

```
0:00:00.496800500    25 0x55dbb8125e40 INFO                 kvssink gstkvssink.cpp:902:gst_kvs_sink_handle_sink_event: structure is audio/mpeg, mpegversion=(int)4, framed=(boolean)true, stream-format=(string)raw, rate=(int)44100, channels=(int)2, codec_data=(buffer)1210, level=(string)2, base-profile=(string)lc, profile=(string)lc;
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 25' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

chehefen, 2020-05-06:
can you run it in debugger and see where it crashed?

mwildehahn, 2020-05-06:
Those logs I linked to are from running with `GST_DEBUG=4 DEBUG_DUMP_FRAME_INFO=1` , are there other debugger flags I can set?

mwildehahn, 2020-05-06:
Oh, run with `gbd` you mean? I’ve never done that before, let me try.

chehefen, 2020-05-06:
yh, once it crashed do `thread backtrace`

mwildehahn, 2020-05-06:
Hm. I'm not really familiar with this development flow so apologies if I'm doing something wrong here.

When I run: `gdb gst-launch-1.0 33`, I get:
```
# gdb gst-launch-1.0 33
GNU gdb (Ubuntu 9.1-0ubuntu1) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
--Type <RET> for more, q to quit, c to continue without paging--

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from gst-launch-1.0...
(No debugging symbols found in gst-launch-1.0)
Attaching to program: /usr/bin/gst-launch-1.0, process 33
[New LWP 34]
[New LWP 35]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

__lll_lock_wait (futex=futex@entry=0x55dc873cbf50, private=0) at lowlevellock.c:52
52	lowlevellock.c: No such file or directory.
(gdb)
(gdb) quit
A debugging session is active.

	Inferior 1 [process 33] will be detached.

Quit anyway? (y or n) Quit
(gdb) quit
A debugging session is active.

	Inferior 1 [process 33] will be detached.

Quit anyway? (y or n) EOF [assumed Y]
Detaching from program: /usr/bin/gst-launch-1.0, process 33
[Inferior 1 (process 33) detached]
```

which I'm guessing is actually that it can't find the core dump?

I'm running this from the same directory as where I have the command running so not sure why it can't find it.

These are the logs from invoking the pipeline where it starts to error:

```
Caught SIGSEGV
#0  futex_abstimed_wait_cancelable
#1  __pthread_cond_wait_common
#2  __pthread_cond_timedwait
#3  0x00007ff58107dda6 in defaultConditionVariableWait ()
#4  0x00007ff58107acdc in kinesisVideoStreamDefaultWaitConditionVariable ()
#5  0x00007ff58105a18f in createKinesisVideoStreamSync ()
#6  0x00007ff58104d7c2 in com::amazonaws::kinesis::video::KinesisVideoProducer::createStreamSync(std::unique_ptr<com::amazonaws::kinesis::video::StreamDefinition, std::default_delete<com::amazonaws::kinesis::video::StreamDefinition> >) ()
#7  0x00007ff5810d8d67 in create_kinesis_video_stream(_GstKvsSink*) ()
#8  0x00007ff5810d9152 in kinesis_video_stream_init(_GstKvsSink*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) ()
#9  0x00007ff5810dd893 in gst_kvs_sink_change_state(_GstElement*, GstStateChange) () at /opt/aws-kvs-sdk/build/libgstkvssink.so
#10 0x00007ff5820309f2 in gst_element_change_state ()
#11 0x00007ff582031139 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#12 0x00007ff58200d1b8 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#13 0x00007ff5820309f2 in gst_element_change_state ()
#14 0x00007ff582031139 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#15 0x000055dc85891616 in  ()
#16 0x00007ff581c570b3 in __libc_start_main (main=
#17 0x000055dc85891d2e in  ()
Spinning.  Please run 'gdb gst-launch-1.0 33' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

mwildehahn, 2020-05-06:
Not sure if this is helpful:
```
(gdb) thread apply all bt

Thread 3 (Thread 0x7f109a388700 (LWP 80)):
#0  0x00007f10bc0f196f in __GI___poll (fds=0x55e38ef9c400, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
#1  0x00007f10bc2431ae in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f10bc2432e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007f10bc243331 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007f10bc26c931 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007f10bc1d7609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007f10bc0fe103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 2 (Thread 0x7f109abaa700 (LWP 79)):
#0  0x00007f10bc0bc361 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7f109aba8970, rem=0x7f--Type <RET> for more, q to quit, c to continue without paging--
109aba8980) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:48
#1  0x00007f10bc0c1eb7 in __GI___nanosleep (requested_time=<optimized out>, remaining=<optimized out>) at nanosleep.c:27
#2  0x00007f10bc26e3a7 in g_usleep () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x000055e38e5a6b1a in  ()
#4  0x00007f10bc1e33c0 in <signal handler called> () at /lib/x86_64-linux-gnu/libpthread.so.0
#5  0x00007f10bb3ffd96 in com::amazonaws::kinesis::video::KinesisVideoStream::start(unsigned char const*, unsigned long, unsigned long) () at /opt/aws-kvs-sdk/build/libKinesisVideoProducer.so
#6  0x00007f10bb3ffccd in com::amazonaws::kinesis::video::KinesisVideoStream::start(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long) () at /opt/aws-kvs-sdk/build/libKinesisVideoProducer.so
#7  0x00007f10bb48736f in gst_kvs_sink_handle_sink_event(_GstCollectPads*, _GstCollectData*, _GstEvent*, void*) () at /opt/aws-kvs-sdk/build/libgstkvssink.so
#8  0x00007f10bbdbda4f in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#9  0x00007f10bc3f5bb4 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#10 0x00007f10bc3f6164 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#11 0x00007f10bc3f65da in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#12 0x00007f10bc3f4020 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#13 0x00007f10bc3fed71 in gst_pad_push_event () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#14 0x00007f10bac67d58 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstaudioparsers.so
#15 0x00007f10bac6b03b in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstaudioparsers.so
#16 0x00007f10bbd96fd1 in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#17 0x00007f10bc3f5bb4 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#18 0x00007f10bc3f6164 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#19 0x00007f10bc3f65da in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#20 0x00007f10bc3f4020 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#21 0x00007f10bc3fed71 in gst_pad_push_event () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#22 0x00007f10bb763d96 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstflv.so
#23 0x00007f10bb765c66 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstflv.so
#24 0x00007f10bc3f4eff in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#25 0x00007f10bc3f6f61 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#26 0x00007f10bc3fdd73 in gst_pad_push () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#27 0x00007f10bbdacf95 in  () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#28 0x00007f10bc42c107 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#29 0x00007f10bc26d1d4 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007f10bc26c931 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007f10bc1d7609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#32 0x00007f10bc0fe103 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7f10bbdff440 (LWP 78)):
#0  __lll_lock_wait (futex=futex@entry=0x55e38ef96740, private=0) at lowlevellock.c:52
#1  0x00007f10bc1da131 in __GI___pthread_mutex_lock (mutex=0x55e38ef96740) at ../nptl/pthread_mutex_lock.c:115
#2  0x00007f10bbdbff2c in gst_collect_pads_start () at /lib/x86_64-linux-gnu/libgstbase-1.0.so.0
#3  0x00007f10bb4898bc in gst_kvs_sink_change_state(_GstElement*, GstStateChange) () at /opt/aws-kvs-sdk/build/libgstkvssink.so
#4  0x00007f10bc3dc9f2 in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#5  0x00007f10bc3dd139 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#6  0x00007f10bc3b91b8 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#7  0x00007f10bc3dc9f2 in gst_element_change_state () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#8  0x00007f10bc3dd139 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#9  0x000055e38e5a4616 in  ()
#10 0x00007f10bc0030b3 in __libc_start_main (main=0x55e38e5a4080, argc=31, argv=0x7ffe13225708, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffe132256f8) at ../csu/libc-start.c:308
#11 0x000055e38e5a4d2e in  ()
```

chehefen, 2020-05-06:
not sure why it crashed there. Can you try this pipeline? It worked for me with the patch.
```
gst-launch-1.0 -v videotestsrc is-live=TRUE ! videoconvert ! x264enc bframes=0 speed-preset=veryfast key-int-max=30 bitrate=512 ! video/x-h264,stream-format=byte-stream,alignment=au,profile=baseline ! h264parse ! kvssink stream-name="my-stream" access-key="access-key" secret-key="secret-key" storage-size=512 name=sink audiotestsrc is-live=TRUE ! audioconvert ! audioresample ! avenc_aac ! aacparse ! sink.
```

mwildehahn, 2020-05-06:
Hm that pipeline works for me but still getting crashes with mine 😞 . I uploaded this repo: https://github.com/mwildehahn/nginx-rtmp-kinesis with a container if you are able to try that out.

chehefen, 2020-05-07:
Sorry but I'm not very familiar with OBS to make it work.

chehefen, 2020-05-07:
on top of the patch, can you change createStreamSync to createStream here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L385

mwildehahn, 2020-05-08:
Thanks for continuing to help me debug, I really appreciate it. Here are the logs after making that patch: https://gist.github.com/mwildehahn/bc47fca8d3f672268d260f953eddca2b. I don't see the exception anymore, but it just seems to hang.

I'm going to test deploying this to a server and ingesting another RTMP stream since I'm not familiar with OBS either, but I'm not optimistic that is going to fix it.

chehefen, 2020-05-08:
can you try this pipeline?
```
LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib:$LD_LIBRARY_PATH PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/bin:$PATH GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build:$GST_PLUGIN_PATH GST_DEBUG=4 DEBUG_DUMP_FRAME_INFO=1 \
    /usr/bin/gst-launch-1.0 -vm \
    rtmpsrc name=rtmpsrc blocksize=1024 do-timestamp=true location="rtmp://localhost:1935/$1/$2" \
    ! flvdemux name=demux demux.video \
    ! queue \
    ! h264parse \
    ! video/x-h264, format=avc,alignment=au \
    ! kvssink log-config=/opt/kvs-log-config stream-name=$2 storage-size=512 name=kvs \
        aws-region="${AWS_REGION}" access-key="${AWS_ACCESS_KEY}" secret-key="${AWS_SECRET_KEY}" restart-on-error=0 \
    demux.audio ! queue ! aacparse ! kvs. &>> $LOG_FILE &
```

disa6302, 2020-06-15:
@mwildehahn , 

Any updates? Does the pipeline @chehefen suggested work for you?

MushMal, 2020-06-26:
Resolving due to staleness. Please feel free to cut a new issue with details if needed

closed 2020-06-26 by MushMal