awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #434: Build failing in Redhat Linux

sumanghosh12, 2020-04-30:
-- Begin building dependencies.
-- openssl already built
-- curl already built
-- Finished building dependencies.
**-- Checking for module 'libcurl'
--   No package 'libcurl' found**
CMake Error at /usr/local/share/cmake-3.6/Modules/FindPkgConfig.cmake:424 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/local/share/cmake-3.6/Modules/FindPkgConfig.cmake:597 (_pkg_check_modules_internal)
  CMakeLists.txt:93 (pkg_check_modules)

disa6302, 2020-04-30:
@sumanghosh12 , 

Have you installed libcurl4-openssl-dev on your system?

sumanghosh12, 2020-04-30:
@disa6302  thanks- After installing libcurl4-openssl-dev, I am able to proceed further.

Now I am stuck with gstreamer package. Are you using Redhat? How is you install it?

I followed this link but no success.
https://gstreamer.freedesktop.org/download/redhat.html

Error details:
 yum install gstreamer-universe
Loaded plugins: product-id, search-disabled-repos, subscription-manager
http://gstreamer.net/pkg/fedora/7Server/x86_64/yum/gst/repodata/repomd.xml: [Errno 14] HTTPS Error 404 - Not Found
Trying other mirror.
To address this issue please refer to the below knowledge base article

https://access.redhat.com/articles/1320623

If above article doesn't help to resolve this issue please open a ticket with Red Hat Support.

 One of the configured repositories failed (Red Hat Linux 7Server - x86_64 - GStreamer),
 and yum doesn't have enough cached data to continue. At this point the only
 safe thing yum can do is fail. There are a few ways to work "fix" this:

     1. Contact the upstream for the repository and get them to fix the problem.

     2. Reconfigure the baseurl/etc. for the repository, to point to a working
        upstream. This is most often useful if you are using a newer
        distribution release than is supported by the repository (and the
        packages for the previous distribution release still work).

     3. Run the command with the repository temporarily disabled
            yum --disablerepo=gstreamer ...

     4. Disable the repository permanently, so yum won't use it by default. Yum
        will then just ignore the repository until you permanently enable it
        again or use --enablerepo for temporary usage:

            yum-config-manager --disable gstreamer
        or
            subscription-manager repos --disable=gstreamer

     5. Configure the failing repository to be skipped, if it is unavailable.
        Note that yum will try to contact the repo. when it runs most commands,
        so will have to try and fail each time (and thus. yum will be be much
        slower). If it is a very temporary problem though, this is often a nice
        compromise:

            yum-config-manager --save --setopt=gstreamer.skip_if_unavailable=true

failure: repodata/repomd.xml from gstreamer: [Errno 256] No more mirrors to try.
http://gstreamer.net/pkg/fedora/7Server/x86_64/yum/gst/repodata/repomd.xml: [Errno 14] HTTPS Error 404 - Not Found

disa6302, 2020-04-30:
@sumanghosh12 , 

Can you install the following packages using yum and let me know if it works?
1. ibgstreamer1.0-dev 
2. libgstreamer-plugins-base1.0-dev 
3. gstreamer1.0-plugins-base-apps
4. gstreamer1.0-plugins-bad 
5. gstreamer1.0-plugins-good 
6. gstreamer1.0-plugins-ugly 
7. gstreamer1.0-tools

sumanghosh12, 2020-05-01:
@disa6302  - 
Not able to install any of them in redhat linux. 
Thanks,
Suman

sumanghosh12, 2020-05-01:
@disa6302 Able to successfully build the the code. I used yum install gstreamer*
When I execute gst-launch-1.0, I am getting the following error
**ERROR - WARNING: erroneous pipeline: no element "kvssink"** 
I am running it from build directory. I am not able to locate : libgstkvssink.so anywhere in my machine.

gst-launch-1.0 rtspsrc location="rtsp://<RTSP_URL>" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="SOME_STREAM_NAME" storage-size=512  access-key="XXXXXXXXXXXXX" secret-key="XXXXXXXXXXXXXXXXXXX" aws-region="us-east-1"

disa6302, 2020-05-01:
@sumanghosh12 ,

The lib will be available in the build directory. 

Please ensure you ran the following cmake command:

`cmake .. -DBUILD_GSTREAMER_PLUGIN=ON`

If this passes successfully, you can run `make`. Once this goes through, the library should be located in the build directory. You also need to run `export GST_PLUGIN_PATH=$PWD/build`

sumanghosh12, 2020-05-01:
@disa6302 
**Success**
I am able to send the live stream from camera to Kinesis Video stream.

I do see this error in the very first line:**log4cplus:ERROR could not open file ./kvs_log_configuration**? 
How do I stop the pipeline?

sumanghosh12, 2020-05-01:
@disa6302 - Thanks. 

Getting this error:

Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
eating Kinesis Video Client
 INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
 INFO - heapInitialize(): Creating AIV heap.
 INFO - heapInitialize(): Heap is initialized OK
 DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
 DEBUG - getSecurityTokenHandler invoked
 DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: xxxxxxxxxxxxxxxxxxxxxxx Expiration: 0
 DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
 INFO - createDeviceResultEvent(): Create device result event.
 DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
 DEBUG - clientReadyHandler invoked
 INFO - try creating stream
 INFO - Creating Kinesis Video Stream xxxxxxxxxxxxxxxxxxxxxxx
 INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
 DEBUG - logStreamInfo(): Kinesis Video Stream Info
 DEBUG - logStreamInfo():  Stream name: xxxxxxxxxxxxxxxxxxxxxxx
 DEBUG - logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME
 DEBUG - logStreamInfo():  Content type: video/h264
 DEBUG - logStreamInfo():  Max latency (100ns): 600000000
 DEBUG - logStreamInfo():  Fragment duration (100ns): 20000000
 DEBUG - logStreamInfo():  Key frame fragmentation: Yes
 DEBUG - logStreamInfo():  Use frame timecode: Yes
 DEBUG - logStreamInfo():  Absolute frame timecode: Yes
 DEBUG - logStreamInfo():  Nal adaptation flags: 0
 DEBUG - logStreamInfo():  Average bandwith (bps): 4194304
 DEBUG - logStreamInfo():  Framerate: 25
 DEBUG - logStreamInfo():  Buffer duration (100ns): 1200000000
 DEBUG - logStreamInfo():  Replay duration (100ns): 400000000
 DEBUG - logStreamInfo():  Connection Staleness duration (100ns): 600000000
 DEBUG - logStreamInfo():  Store Pressure Policy: 1
 DEBUG - logStreamInfo():  View Overflow Policy: 1
 DEBUG - logStreamInfo():  Segment UUID: NULL
 DEBUG - logStreamInfo():  Frame ordering mode: 0
 DEBUG - logStreamInfo(): Track list
 DEBUG - logStreamInfo():  Track id: 1
 DEBUG - logStreamInfo():  Track name: kinesis_video
 DEBUG - logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
 DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
 DEBUG - logStreamInfo():  Track cpd: NULL
 DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
 INFO - writeHeaderCallback(): RequestId: xxxxxxxxxxxxxxxxxxxxxxx
 DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.xxxxxxxxxxxxxxxxxxxxxxx,"DataRetentionInHours":24,"DeviceName":null,"KmsKeyId":"xxxxxxxxxxxxxxxxxxxxxxx","MediaType":null,"Status":"ACTIVE","StreamARN":"xxxxxxxxxxxxxxxxxxxxxxx","StreamName":"xxxxxxxxxxxxxxxxxxxxxxx","Version":"xxxxxxxxxxxxxxxxxxxxxxx"}}
2020-04-30 22:17:52 [139690704680704] INFO - describeStreamResultEvent(): Describe stream result event.
2020-04-30 22:17:52 [139690704680704] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2020-04-30 22:17:52 [139690704680704] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2020-04-30 22:17:52 [139690704680704] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2020-04-30 22:17:52 [139690704680704] INFO - writeHeaderCallback(): RequestId: xxxxxxxxxxxxxxxxxxxxxxx
2020-04-30 22:17:53 [139690696288000] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"xxxxxxxxxxxxxxxxxxxxxxx"}
2020-04-30 22:17:53 [139690696288000] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2020-04-30 22:17:53 [139690696288000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2020-04-30 22:17:53 [139690696288000] DEBUG - getStreamingTokenHandler invoked
2020-04-30 22:17:53 [139690696288000] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: xxxxxxxxxxxxxxxxxxxxxxx Expiration: xxxxxxxxxxxxxxxxxxxxxxx
2020-04-30 22:17:53 [139690696288000] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2020-04-30 22:17:53 [139690696288000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-04-30 22:17:53 [139690696288000] DEBUG - streamReadyHandler invoked
2020-04-30 22:17:53 [139690696288000] Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://xxxxxxxxxxxxxxxxxxxxxxx
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1: Internal data stream error.
Additional debug info:
gstbasesrc.c(2950): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:02.193318139
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream XXXXXXXXXXXXXXXXXXXXXXXXXXXXX
 INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
 DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty

MushMal, 2020-05-01:
@sumanghosh12 seems like you have an issue with your pipeline, specifically with the rtsp plugin in UDP mode.

Try enabling GStreamer debug logs by setting GST_DEBUG environment variable like so

`export GST_DEBUG=4`

and re-running the pipeline as recommended here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/193#issuecomment-474093882

sumanghosh12, 2020-05-01:
@disa6302  I am able to send live stream now from my Linux server. Thank you.

closed 2020-05-01 by sumanghosh12