awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #435: Files structure mismatch from the aws documentation

nik13, 2020-04-30:
Hi Team, 

Looks like the whole file structure and files have changed from the latest push. There is no docker_native_scripts folder and files. Where do I exactly find them?

disa6302, 2020-04-30:
@nik13 ,

We have removed docker related folder from this repo. 

The repo went through a revamp recently and we are working on getting the documentation fixed in aws as well. 

As far as dockerscripts, we are working on adding to our repos. There is no ETA on it yet. 

Meanwhile, if you would like a reference the docker scripts, you could look at version[ 2.1.0](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/releases/tag/2.1.0) of the sdk and that will have the directory you are looking for.

disa6302, 2020-05-03:
Hello @nik13 , 

We will be adding the docker scripts to https://github.com/aws-samples/amazon-kinesis-video-streams-demos. Closing this for now. Feel free to re-open the issue in the future.

closed 2020-05-03 by disa6302