awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #437: Error "No such element or plugin 'kvssink'

darkeyeai, 2020-05-01:
Hi, 

I am following the documentation on https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/gs-send-data.html exactly as written, with the exception of downloading the SDK, which I have done via this GitHub page. I am having two issues:

1. **kinesis-video-native-build** directory does not exist. I cannot locate it.
2. I receive this error: _No such element or plugin 'kvssink'_ 
when I run `gst-inspect-1.0 kvssink`

Any help is greatly appreciated.
Thank You.

disa6302, 2020-05-01:
@darkeyeai , 

Please follow the instructions in the github page. We have moved to a CMake build system and do not have the native-build directory anymore. Thanks!

darkeyeai, 2020-05-01:
Thank you for the quick response. That solves issue one.

I have followed the directions completely on the github page, but am still receiving the error:
**No such element or plugin 'kvssink**

Thanks in advance.

disa6302, 2020-05-01:
Can you check if you have the plugin library in the build directory? You need to run export `GST_PLUGIN_PATH=$PWD/build`. Make sure your `cmake` and `make` ran through with no errors. 

Also, ensure you run `cmake` with `-DBUILD_GSTREAMER_PLUGIN=ON`

darkeyeai, 2020-05-01:
Got it working, thank you. I was running `cmake` with `-DBUILD_GSTREAMER_PLUGIN=ON`in a wrong directory.

Thanks Divya!

closed 2020-05-01 by darkeyeai