awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #438: CMake fails with -DBUILD_DEPENDENCIES=OFF

bpawlik, 2020-05-04:
Cmake fails with -DBUILD_DEPENDENCIES=OFF

Apparently JSMN lib is missing, but I don't know of any existing apt-get package including it. Perhaps this one should be always build from source?

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
JSMN_LIBRARY (ADVANCED)
    linked by target "kvsCommonCurl" in directory amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-producer-c
Jsmn
    linked by target "KinesisVideoProducer" in directory amazon-kinesis-video-streams-producer-sdk-cpp

```

disa6302, 2020-05-04:
@bpawlik ,

You are right. There is no apt-get for this. It should always be built from source. I would still suggest you to run `cmake` with the flag ON, so that these packages are taken care of. Also, this flag is ON by default.

bpawlik, 2020-05-04:
Thanks for quick response.

The issue is that I have other components that rely on openssl and I get version mismatch at runtime. One option would be to link against KVS repo, but it's ugly and adds unnecessary complexity. Elegant way would be that they all use the same system level library.

disa6302, 2020-05-04:
@bpawlik ,

You can do that. For jsmn, you can run the steps in the Libjsmn specific cmake file which you find in the CMake directory. That should work.

bpawlik, 2020-05-04:
There are two issues if I do that:

1) There is a bug in CMakeLists.txt file: find_library will fails cause this line `set(OPEN_SRC_INSTALL_PREFIX ${KINESIS_VIDEO_OPEN_SOURCE_SRC}/local)` is hidden behind BUILD_DEPENDENCIES if-clause

2) Submodule amazon-kinesis-video-streams-producer-c should switch from find_package to find_library with PATHS pointing to open-source/local

bpawlik, 2020-05-05:
FYI

I hacked the solution in the following way:
- built Jsmn from sources
- copy libs to amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libs
- copy headers to amazon-kinesis-video-streams-producer-sdk-cpp/open-source/headers
- `export OPEN_SOURCE_PATH=/path/to/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/`
- build KVS with `cmake .. -DCMAKE_PREFIX_PATH=$OPEN_SOURCE_PATH -DBUILD_DEPENDENCIES=OFF`

mwildehahn, 2020-05-05:
Slightly related to this: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/430. I upgraded openssl to get it working. Is it worth doing a PR for this? Not sure what downstream side effects this has for older builds.

disa6302, 2020-05-05:
@bpawlik ,
Good to know you made it work! Feel free to open a PR with your changes @bpawlik !

zhiyua-git, 2020-05-10:
Closing as the original issue is resolved. PR of the fixes/hacks is welcome!

closed 2020-05-10 by zhiyua-git