awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #440: Failed to load plugin '../amazon-kinesis-video-streams-producer-sdk-cpp/build/kvscproducer/libcproducer.so'

js1972, 2020-05-06:
Occasionally when using the plugin in a gstreamer pipeline and even with gst-inspect-1.0 I get this warning:
```
(gst-plugin-scanner:28351): GStreamer-WARNING **: 10:49:41.290: Failed to load plugin '/home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/kvscproducer/libcproducer.so': /home/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/build/kvscproducer/libcproducer.so: undefined symbol: globalGetCompilerInfo
```

It only happens every now and then and doesnt seem to effect the plugin - that I can tell...

MushMal, 2020-05-06:
GET_COMPILER_INFO is the macro that's mapped to an extern function pointer globalGetCompilerInfo which is defined in PIC utils/Version.c and is set to defaultGetCompilerInfo which then gets #def-ed to whichever specific compiler string is. This is used to set the version information in the user agent string.

I really don't understand what could cause this problem and would love to learn a little more about this. We don't do dynamic symbol loading in our library chain so it's all compile-time defined and the OS loader should load and satisfy these links. 

In this case, the gst plugin which is a dynamic library (.so) gets loaded by the GStreamers inspect utility dynamically (using sym loading) which, in turn will cause the loader to load the linked .so libraries automatically. These will be the CPP producer library, which links to C producer library which links to PIC utilities statically.

Perhaps, for some reason the OS loader fails but I don't understand why. It would be nice to get more repro steps.

js1972, 2020-05-07:
I'm not concerned as gst-launch-1.0 and gst-inspect-1.0 still work perfectly well with kvssink... and it is just a warning. I just wanted your feedback to ensure its not a critical thing??

MushMal, 2020-05-07:
This particular api is not critical in itself but I just don’t understand what’s going on with the symload for GStreamers gst-inspect. My recommendation would be to try to get as much info as you can. We will also keep an eye on it

MushMal, 2020-05-20:
I am resolving this for now. Please reopen if you get more info

closed 2020-05-20 by MushMal