awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #446: Releases should contain binaries of sample applications to quickly get started

prabhatsharma, 2020-05-10:
Releases should contain binaries of sample applications to quickly get started.  Compiling sample just to stream a video for testing this has resulted in me spending at least 2 hours now. Source code has been changing rapidly. Docs are not able to keep up with it. Binaries of smaples "kinesis_video_gstreamer_sample_app" for some platforms would go a long way in people getting started.

MushMal, 2020-05-20:
@prabhatsharma we are thinking of adding binaries for some platforms.

rtcpersonal2020, 2020-07-23:
+1,000,000
how can anyone evaluate the product with code that won't build according to the docs?

MushMal, 2020-07-23:
These really are two separate issues

1) The docs should certainly be updated to reflect the fast-paced open-source development. In fact, the best would be to simply reference GitHub Readme and other docs that come with the repository.
2) We are thinking of supplying the binaries but it's not a trivial task due to ABI compatibility with many embedded platforms/embedded OS-es and potential licensing issues. 

We understand the pain this can cause for someone who is trying the embedded assets but unfortunately. these are the reasons it's taking longer than anticipated.

jdelapla, 2023-02-23:
Outdated, closing this. Creating binaries for a number of different operating systems is too large of an effort for the team. The README includes build instructions to aide the open source community in building the samples.

closed 2023-02-23 by jdelapla