awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #448: No such element or plugin 'kvssink'

SmitaAhinave, 2020-05-13:
System Information:
          Distributor ID:	Ubuntu
          Description:	Ubuntu 18.04.4 LTS
          Release:	18.04

I'm integrating AWS KVS refering CPP library and following exact same steps mentioned.
Build and everything else was successful.
Please find screenshot attached for more detail.

[image]

Still I'm getting **No such element or plugin 'kvssink' error.**
I also tried solving this with AWS support but still no luck. They have recommended to post this issue here.

Any help is appreciated.

Sean-Der, 2020-05-13:
Hey @SmitaAhinave 

The plugin may be blacklisted, can you deleted your GStreamer plugin registry, it is in ~/.cache

Sean-Der, 2020-05-13:
I also noticed you have copied the libraries into the build folder, that probably will not work. Can your set your LD_LIBRARY_PATH instead.

SmitaAhinave, 2020-05-13:
> I also noticed you have copied the libraries into the build folder, that probably will not work. Can your set your LD_LIBRARY_PATH instead.

Can you please help me to understand why we need it... it's not mentioned here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp

MushMal, 2020-05-13:
@SmitaAhinave it does look odd that GStreamer is unable to load the plugin given the path. Could you perhaps append the path to the kvssink library to the LD_LIBRARY_PATH and to GST_PLUGIN_PATH and check. The GStreamer questions are better answered in GStreamer forums in general.

We have had few customers who set the path from a command line like so: https://stackoverflow.com/questions/2120444/gstreamer-plugin-search-path

Sean-Der, 2020-05-13:
@SmitaAhinave I can confirm the same error.

First do `export LD_LIBRARY_PATH=`pwd`/open-source/local/lib` (will update the docs)
Then do `rm -rf ~/.cache/gstreamer-1.0` (clear the cache, kvssink will be disabled since it failed to load the first time)

SmitaAhinave, 2020-05-13:
> @SmitaAhinave I can confirm the same error.
> 
> First do `` export LD_LIBRARY_PATH=`pwd`/open-source/local/lib `` (will update the docs)
> Then do `rm -rf ~/.cache/gstreamer-1.0` (clear the cache, kvssink will be disabled since it failed to load the first time)

Worked after setting LD_LIBRARY_PATH

Thank you so much for the help

closed 2020-05-13 by SmitaAhinave

Sean-Der, 2020-05-13:
Thanks for using KVS :)

I am going to update the docs now to mention this. I am doing the final touches today.