awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #449: Releases should contain kvssink binaries

palles77, 2020-05-13:
Releases should contain binaries for platforms such as Windows, Linux, etc. It's extremely difficult to get the build process going on Windows. It would be good to have binaries precompiled without the n eed for manual builds.

Sean-Der, 2020-05-13:
Hey @palles77 

We will add binaries in the future, but Windows should be much easier now! Make sure you have GStreamer and GStreamer development install. Then in the `Visual Studio Command Prompt` you should just need to do. 

```
cmake -G "NMake Makefiles" BUILD_GSTREAMER_PLUGIN=TRUE ..
nmake

set GST_PLUGIN_PATH=%CD%\build
set PATH=%PATH%;%CD%\open-source\local\bin;%CD%\open-source\local\lib
```

Sean-Der, 2020-05-13:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/447 has these instructions, and will be merging today. If you get a chance to try it out would love your review!

palles77, 2020-05-14:
Ok - build now works fine. So I suppose I can close the bug for now.

closed 2020-05-14 by palles77