awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #451: Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no KinesisVideoProducerJNI in java.library.path

volatile2018, 2020-05-14:
Hi,
I'm facing below error after following all the steps which are mentioned in the readme file,
Please provide the necessary solution for the same.

DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .so.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .dylib.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library .dll.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.so.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dylib.
DEBUG / KinesisVideo: Unsatisfied link error. Directly loading native library libKinesisVideoProducerJNI.dll.
**ERROR / KinesisVideo: 2020-05-14T14:15Z T1: EXCEPTION: UnsatisfiedLinkError: Unsatisfied link error. Loading native library KinesisVideoProducerJNI failed with java.lang.UnsatisfiedLinkError: no** KinesisVideoProducerJNI in java.library.path: [src/resources/lib/ubuntu]: no KinesisVideoProducerJNI in java.library.path: [src/resources/lib/ubuntu]
Exception in thread "main" java.lang.RuntimeException: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:59)
Caused by: com.amazonaws.kinesisvideo.producer.ProducerException: Failed loading native library StatusCode: 0xd
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.initializeLibrary(NativeKinesisVideoProducerJni.java:1170)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.create(NativeKinesisVideoProducerJni.java:223)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(NativeKinesisVideoProducerJni.java:241)
	at com.amazonaws.kinesisvideo.internal.producer.jni.NativeKinesisVideoProducerJni.createSync(NativeKinesisVideoProducerJni.java:207)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initializeNewKinesisVideoProducer(NativeKinesisVideoClient.java:238)
	at com.amazonaws.kinesisvideo.internal.client.NativeKinesisVideoClient.initialize(NativeKinesisVideoClient.java:119)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:113)
	at com.amazonaws.kinesisvideo.java.client.KinesisVideoJavaClientFactory.createKinesisVideoClient(KinesisVideoJavaClientFactory.java:86)
	at com.amazonaws.kinesisvideo.demoapp.DemoAppMain.main(DemoAppMain.java:42)

[image]

disa6302, 2020-05-14:
@volatile2018 , 

Can you check if the KinesisVideoProducerJNI is present in `src/resources/lib/ubuntu`?

Sean-Der, 2020-05-14:
@volatile2018 also it may be that `LD_LIBRARY_PATH` is not set, if you move the location of the `*.so` and it is linked to things relatively.

You can use `ldd` to diagnose that.

MushMal, 2020-05-20:
@volatile2018 did you get a chance to try the recommendations? I wouldn't want this to go stale and unanswered. Please update or resolve

MushMal, 2020-05-26:
Closing. Please re-open if you need further assistance

closed 2020-05-26 by MushMal