awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #452: Install error

hammacktony, 2020-05-14:
I ran the build instructions as specified in the readme. However, cmake flips out when getting to `Performing configure step for 'project_libautoconf'`.

Here is the full error. 
```
configure: error: cannot find sources (lib/autoconf/autoconf.m4) in /vagrant/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf/build/src/project_libautoconf or ..
CMakeFiles/project_libautoconf.dir/build.make:109: recipe for target 'build/src/project_libautoconf-stamp/project_libautoconf-configure' failed
make[2]: *** [build/src/project_libautoconf-stamp/project_libautoconf-configure] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/project_libautoconf.dir/all' failed
Makefile:83: recipe for target 'all' failed
make[1]: *** [CMakeFiles/project_libautoconf.dir/all] Error 2
make: *** [all] Error 2
CMake Error at CMake/Utilities.cmake:68 (message):
  CMake step for libautoconf failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:81 (build_dependency)

```

Any help would be appreciated.

Sean-Der, 2020-05-14:
Hey @hammacktony. Sorry about that!

What OS (and version) are you running? I can reproduce and fix, not worth asking the individual versions of things. Unless you are installing stuff manually and not through the package manager.

hammacktony, 2020-05-18:
I am installing it manually. I am using Ubuntu 18.04. I am just building from master.

Sean-Der, 2020-05-18:
In the configure logs we get

`checking for GNU M4 that supports accurate traces... configure: error: no acceptable m4 could be found in $PATH` 

Looking into this now!

Sean-Der, 2020-05-18:
@hammacktony after doing a `apt-get install m4` it works! I will update the docs.

PR #460 (2020-05-19): Update docs to mention m4 requirement
Resolves #452
commit 279a5dc5 (2020-05-19): Update docs to mention m4 requirement

Resolves #452
closed 2020-05-19 by Sean-Der