awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #454: Can't use the word "Cam" in stream name

jelling, 2020-05-14:
This is a weird one, but if you name your stream `Webcam` or `MyCam` then Kinesis  / gstreamer will somehow shove the video into another stream (perhaps a default one) not named that. Changing the name to anything else resolves the problem.

I'm using this stock command fwiw:
```
gst-launch-1.0 -v  filesrc location=oneliners.mp4 ! qtdemux name=demux ! queue ! h264parse !  video/x-h264,stream-format=avc,alignment=au ! kvssink name=sink stream-name=Webcam access-key="{removed}" secret-key="{removed}" aws-region="us-east-1" streaming-type=offline demux. ! queue ! aacparse ! sink.
```

MushMal, 2020-05-14:
@jelling could you please elaborate what that other stream is?

Can you include debug logs?

jelling, 2020-05-18:
It didn’t seem to matter what the other stream was. It just would never put the incoming video in the right stream. Not certain what debug logs would show this.

MushMal, 2020-05-18:
@jelling I am not sure I understand what you are describing. Unless we have logs we can't help in this situation, especially, since we can't reproduce this claim.

MushMal, 2020-05-26:
Closing. Please re-open if you have further information

closed 2020-05-26 by MushMal