awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #455: Segmentation fault (core dumped)

usmanm259, 2020-05-16:
Hi,

I am currently facing this issue on ubuntu 16.04, when i try to stream from multiple RTSP sources using Gstreamer kinesis_video_gstreamer_sample_multistream_app.

it works fine and i can see streams from sources in AWS kinesis console, when i run this app 

AWS_ACCESS_KEY_ID=------- AWS_SECRET_ACCESS_KEY=----------- ./kinesis_video_gstreamer_sample_multistream_app MyRtspStreams rtspurlsfile

but after 5 mins process exists with the error message below. 

[ERROR] [15-05-2020 09:28:48:137.398 GMT] postReadCallback(): Failed to get data from the stream with an error: 0x30000004
[WARN ] [15-05-2020 09:28:48:138.928 GMT] curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia
[DEBUG] [15-05-2020 09:28:48:139.027 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStreams_0 with upload handle: 57 exited. http status: 408
[WARN ] [15-05-2020 09:28:48:139.480 GMT] putStreamCurlHandler(): Stream with streamHandle 10923184 has exited without triggering end-of-stream. Service call result: 408
[INFO ] [15-05-2020 09:28:48:139.536 GMT] kinesisVideoStreamTerminated(): Stream 0xa6acb0 terminated upload handle 57 with service call result 408.
[DEBUG] [15-05-2020 09:28:48:139.645 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
[DEBUG] [15-05-2020 09:28:48:139.729 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
[DEBUG] [15-05-2020 09:28:48:139.761 GMT] streamReadyHandler invoked
[DEBUG] [15-05-2020 09:28:48:139.810 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[INFO ] [15-05-2020 09:28:48:139.937 GMT] putStreamResultEvent(): Put stream result event. New upload handle 60
[DEBUG] [15-05-2020 09:28:48:150.090 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[INFO ] [15-05-2020 09:28:48:368.844 GMT] writeHeaderCallback(): RequestId: e731570f-28ff-ed37-b009-341a7846f367
[ERROR] [15-05-2020 09:28:48:369.083 GMT] postReadCallback(): Failed to get data from the stream with an error: 0x30000004
[WARN ] [15-05-2020 09:28:48:370.528 GMT] curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia

[DEBUG] [15-05-2020 09:28:48:370.582 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStreams_0 with upload handle: 58 exited. http status: 408
[WARN ] [15-05-2020 09:28:48:370.713 GMT] putStreamCurlHandler(): Stream with streamHandle 10923184 has exited without triggering end-of-stream. Service call result: 408
[INFO ] [15-05-2020 09:28:48:370.738 GMT] kinesisVideoStreamTerminated(): Stream 0xa6acb0 terminated upload handle 58 with service call result 408.
[DEBUG] [15-05-2020 09:28:48:370.755 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
[DEBUG] [15-05-2020 09:28:48:370.806 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
[DEBUG] [15-05-2020 09:28:48:370.826 GMT] streamReadyHandler invoked
[DEBUG] [15-05-2020 09:28:48:370.874 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[INFO ] [15-05-2020 09:28:48:370.947 GMT] putStreamResultEvent(): Put stream result event. New upload handle 61
[DEBUG] [15-05-2020 09:28:48:381.076 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[INFO ] [15-05-2020 09:28:48:794.762 GMT] writeHeaderCallback(): RequestId: c79248bf-eb95-cb90-90aa-2baa2d1e99aa
[ERROR] [15-05-2020 09:28:48:794.842 GMT] postReadCallback(): Failed to get data from the stream with an error: 0x30000004
[WARN ] [15-05-2020 09:28:48:796.207 GMT] curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia

[DEBUG] [15-05-2020 09:28:48:796.265 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStreams_1 with upload handle: 59 exited. http status: 408
[WARN ] [15-05-2020 09:28:48:796.424 GMT] putStreamCurlHandler(): Stream with streamHandle 10923192 has exited without triggering end-of-stream. Service call result: 408
[INFO ] [15-05-2020 09:28:48:796.469 GMT] kinesisVideoStreamTerminated(): Stream 0xa6acb8 terminated upload handle 59 with service call result 408.
[DEBUG] [15-05-2020 09:28:48:796.581 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
[DEBUG] [15-05-2020 09:28:48:802.966 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
[DEBUG] [15-05-2020 09:28:48:803.267 GMT] streamReadyHandler invoked
[DEBUG] [15-05-2020 09:28:48:803.316 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[INFO ] [15-05-2020 09:28:48:803.412 GMT] putStreamResultEvent(): Put stream result event. New upload handle 62
[DEBUG] [15-05-2020 09:28:48:813.915 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[INFO ] [15-05-2020 09:28:49:364.484 GMT] writeHeaderCallback(): RequestId: c20a7da9-d637-9d9a-9532-1ebf86e2d074
[ERROR] [15-05-2020 09:28:49:364.564 GMT] postReadCallback(): Failed to get data from the stream with an error: 0x30000004
[WARN ] [15-05-2020 09:28:49:365.713 GMT] curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia

[DEBUG] [15-05-2020 09:28:49:365.777 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStreams_0 with upload handle: 60 exited. http status: 408
[WARN ] [15-05-2020 09:28:49:365.869 GMT] putStreamCurlHandler(): Stream with streamHandle 10923184 has exited without triggering end-of-stream. Service call result: 408
[INFO ] [15-05-2020 09:28:49:365.894 GMT] kinesisVideoStreamTerminated(): Stream 0xa6acb0 terminated upload handle 60 with service call result 408.
[DEBUG] [15-05-2020 09:28:49:365.912 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
[DEBUG] [15-05-2020 09:28:49:365.958 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000040
[DEBUG] [15-05-2020 09:28:49:365.980 GMT] streamReadyHandler invoked
[DEBUG] [15-05-2020 09:28:49:366.022 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
[INFO ] [15-05-2020 09:28:49:366.091 GMT] putStreamResultEvent(): Put stream result event. New upload handle 63
[DEBUG] [15-05-2020 09:28:49:377.148 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
[INFO ] [15-05-2020 09:28:49:668.169 GMT] writeHeaderCallback(): RequestId: fd1a5d0a-859a-eee8-aa22-3e1c4311bcd2
[ERROR] [15-05-2020 09:28:49:668.274 GMT] postReadCallback(): Failed to get data from the stream with an error: 0x30000004
[WARN ] [15-05-2020 09:28:49:669.290 GMT] curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia

[DEBUG] [15-05-2020 09:28:49:669.350 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStreams_0 with upload handle: 61 exited. http status: 408
[WARN ] [15-05-2020 09:28:49:669.468 GMT] putStreamCurlHandler(): Stream with streamHandle 10923184 has exited without triggering end-of-stream. Service call result: 408
[INFO ] [15-05-2020 09:28:49:669.490 GMT] kinesisVideoStreamTerminated(): Stream 0xa6acb0 terminated upload handle 61 with service call result 408.
Segmentation fault (core dumped)

MushMal, 2020-05-16:
@usmanm259 It's not possible to tell what's going on with the short log you've attached. In this case the error 0x30000004 is a likely a consequence of bad data that was produced in earlier. Later, it seems that you hit network timeout 408.

Also, there is no information about the crash.

Recommend:

* Using a single RTSP stream for now
* Include detailed DEBUG logs
* Include core dumps for the crash so we can tell what's going on

usmanm259, 2020-05-17:
@MushMal Thanks for the reply. 

1. Using a single RTSP stream:

I have tried the Gstreamer app for single RTSP stream but same response.

2. Include detailed DEBUG logs

Attaching detailed DEBUG logs
[logs.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4640074/logs.txt)

usmanm259, 2020-05-17:
@MushMal Thanks for the reply.

1. Using a single RTSP stream:

I have tried the Gstreamer app for single RTSP stream but same response.

2. Include detailed DEBUG logs

Attaching detailed DEBUG logs

[logs.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4640087/logs.txt)

MushMal, 2020-05-17:
Your logs indicate that at times you have n observed bandwidth of 25Mbps but it’s not actually sufficient for the density of your stream. You get into a pogo-stick effect with the default latency handlers trying to reset the connection and eventually starting to drop frames. 

Try configuring the rtsp source to have lower bitrate.

usmanm259, 2020-05-18:
@MushMal Thanks, i will test with lower bitrate and let you know.

usmanm259, 2020-05-19:
Hi,

So i tested with lower constant bitrate (CBR) (640 kbps) with Stream control configured on camera's sub stream, i successfully ingested stream to kinesis video streams for longer duration (9 hours).

After 9 hours there was some network issue which caused the application to drop stream, so once the network issue was gone and camera was  accessible with ping but the issue was that Gstreamer application was unable to stream again, was stuck in loop and trying again and again for hours.

When i killed the process and started application again, once again i was able to stream to kinesis.

So, if there is anything that can be done to make it work, when there is a network issue and after its gone, then application continues automatically streaming again.

MushMal, 2020-05-19:
The SDK default configuration is to retry. It seems that’s why it was trying to do. Could you please include the debug logs when it we “stuck”? If you are using KVS plugin with GStreamer rtsp plugin then there could be a number of things that might be wrong including the network link to the rtsp source

MushMal, 2020-05-26:
Resolving due to no further correspondence. Please re-open or cut a new issue if you need further assistance.

closed 2020-05-26 by MushMal

usmanm259, 2020-06-04:
**Hi** @MushMal,

Sorry for not responding, i was testing streams with gstreamer kvs_gstreamer_multistream_sample app for longer duration. 
The sample gstreamer app worked for more than 1 week continuously. 

The issue that i am facing is that, when there is a network problem the gstreamer app stops streaming to kinesis video streams and gsteamer app is still running without exiting process.

When network problem is resolved, gstreamer app should continue with streaming from cameras but currently app gets stuck at the state, after restarting app, it again starts streaming.

I am running the app with linux systemd service, so the service status shows app is running for 4 days but when i check logs, app is stuck.

so my question is how can i resolve this issue, If the kvs_gstreamer_multistream_sample

1. After network issue is gone, App should continue with streaming from cameras but currently it gets stuck and no new logs are produced to check the status of app.

**OR**

2. App Doesn't exit process on failure then how can i restart linux systemd service on failure when app is still running at stuck state without streaming to kinesis.

**Debug Logs**

[DEBUG] [03-06-2020 12:44:02:022.915 GMT] fragmentAckReceivedHandler invoked
[ERROR] [03-06-2020 12:44:02:022.950 GMT] getKinesisVideoStreamData(): operation returned status code: 0x5200002f
[DEBUG] [03-06-2020 12:44:02:022.961 GMT] postReadCallback(): Wrote 6583 bytes to Kinesis Video. Upload stream handle: 190
[ERROR] [03-06-2020 12:44:02:023.019 GMT] getKinesisVideoStreamData(): operation returned status code: 0x5200002f
[DEBUG] [03-06-2020 12:44:02:023.030 GMT] postReadCallback(): Pausing CURL read for upload handle: 190
[DEBUG] [03-06-2020 12:44:02:915.284 GMT] Kinesis Video client and stream metrics
        >> Overall storage byte size: 134217728
        >> Available storage byte size: 134046901
        >> Allocated storage byte size: 170827
        >> Total view allocation byte size: 144080
        >> Total streams frame rate (fps): 18
        >> Total streams transfer rate (bps): 661696 (646 Kbps)
        >> Current view duration (ms): 759
        >> Overall view duration (ms): 1999
        >> Current view byte size: 73800
        >> Overall view byte size: 169339
        >> Current frame rate (fps): 18.5827
        >> Current transfer rate (bps): 661696 (646 Kbps)
[DEBUG] [03-06-2020 12:44:03:024.208 GMT] postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 190
[ERROR] [03-06-2020 12:44:03:025.025 GMT] getKinesisVideoStreamData(): operation returned status code: 0x5200002f
[DEBUG] [03-06-2020 12:44:03:025.055 GMT] postReadCallback(): Wrote 10721 bytes to Kinesis Video. Upload stream handle: 190
[ERROR] [03-06-2020 12:44:03:025.100 GMT] getKinesisVideoStreamData(): operation returned status code: 0x5200002f
[DEBUG] [03-06-2020 12:44:03:025.113 GMT] postReadCallback(): Pausing CURL read for upload handle: 190
[DEBUG] [03-06-2020 12:44:03:485.094 GMT] postWriteCallback(): Curl post body write function for stream with handle: MyRtspStream_0 and upload handle: 190 returned: {"EventType":"RECEIVED","FragmentTimec$

[DEBUG] [03-06-2020 12:44:03:485.191 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [03-06-2020 12:44:03:490.203 GMT] postWriteCallback(): Curl post body write function for stream with handle: MyRtspStream_0 and upload handle: 190 returned: {"EventType":"BUFFERING","FragmentTime$

[DEBUG] [03-06-2020 12:44:03:490.246 GMT] fragmentAckReceivedHandler invoked
[DEBUG] [03-06-2020 12:44:03:553.755 GMT] postWriteCallback(): Curl post body write function for stream with handle: MyRtspStream_0 and upload handle: 190 returned: {"EventType":"PERSISTED","FragmentTime$

[DEBUG] [03-06-2020 12:44:03:564.010 GMT] fragmentAckReceivedHandler invoked
**[ERROR]** [03-06-2020 12:44:03:564.073 GMT] getKinesisVideoStreamData(): operation returned status code: 0x5200002f
[DEBUG] [03-06-2020 12:44:03:564.084 GMT] postReadCallback(): Pausing CURL read for upload handle: 190
**[WARN ]** [03-06-2020 12:44:39:620.940 GMT] curlCompleteSync(): curl perform failed for url https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com/putMedia with result Timeout was reached: Operation too s$
**[WARN ]** [03-06-2020 12:44:39:621.264 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com/putMedia
Request Headers:

[DEBUG] [03-06-2020 12:44:39:621.335 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStream_0 with upload handle: 190 exited. http status: 0
**[WARN ]** [03-06-2020 12:44:39:621.531 GMT] putStreamCurlHandler(): Stream with streamHandle 37746056 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [03-06-2020 12:44:39:621.565 GMT] **kinesisVideoStreamTerminated():** Stream 0x23ff588 terminated upload handle 190 with service call result 599.
[DEBUG] [03-06-2020 12:44:39:621.588 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200

MushMal, 2020-06-04:
@usmanm259 not sure what's going on with your case as I don't have any logs. From the snippet of the code it looks like the SDK is continuously retrying and is getting a timeout for PutMedia so there is no networking.

Generally speaking, the sample application will use continuous retry policy. This means that the SDK will just continue retrying to send the buffer. Even when the frames are dropped. The default logic will attempt to reset the connection as it hits the buffer pressures but in your case it won't help as you don't have network at all so eventually the frames will start dropping. The default continuous retry policy will also reset the stream (dump the buffer and retry) when we get any errors.

Your application can listen to the callbacks (like dropped frame, buffer pressure or error) and react accordingly.

usmanm259, 2020-06-04:
@MushMal I agree with you but the issue is if application has continuous retry policy then why it starts working as soon as i restart the application and if i don't restart the app and wait for it to start streaming again then it just doesn't do that.

As you can see these logs are of **June 3, 2020** ,the stream stopped after app produced these logs and i restart the app on **June 4, 2020** , waited for the app to retry streaming but no success.

[DEBUG] [03-06-2020 12:44:03:564.010 GMT] fragmentAckReceivedHandler invoked
[ERROR] [03-06-2020 12:44:03:564.073 GMT] getKinesisVideoStreamData(): operation returned status code: 0x5200002f
[DEBUG] [03-06-2020 12:44:03:564.084 GMT] postReadCallback(): Pausing CURL read for upload handle: 190
[WARN ] [03-06-2020 12:44:39:620.940 GMT] curlCompleteSync(): curl perform failed for url https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com/putMedia with result Timeout was reached: Operation too s$
[WARN ] [03-06-2020 12:44:39:621.264 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com/putMedia
Request Headers:

[DEBUG] [03-06-2020 12:44:39:621.335 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStream_0 with upload handle: 190 exited. http status: 0
[WARN ] [03-06-2020 12:44:39:621.531 GMT] putStreamCurlHandler(): Stream with streamHandle 37746056 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [03-06-2020 12:44:39:621.565 GMT] kinesisVideoStreamTerminated(): Stream 0x23ff588 terminated upload handle 190 with service call result 599.
[DEBUG] [03-06-2020 12:44:39:621.588 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200

usmanm259, 2020-06-04:
These are the logs of June 4, 2020, 

WARN ] [04-06-2020 10:00:20:512.382 GMT] curlCompleteSync(): curl perform failed for url https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
[WARN ] [04-06-2020 10:00:20:513.040 GMT] curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com/putMedia
Request Headers:
    
[DEBUG] [04-06-2020 10:00:20:513.056 GMT] putStreamCurlHandler(): Network thread for Kinesis Video stream: MyRtspStream_0 with upload handle: 6 exited. http status: 0
[WARN ] [04-06-2020 10:00:20:513.212 GMT] putStreamCurlHandler(): Stream with streamHandle 29480328 has exited without triggering end-of-stream. Service call result: 599
[INFO ] [04-06-2020 10:00:20:513.233 GMT] kinesisVideoStreamTerminated(): Stream 0x1c1d588 terminated upload handle 6 with service call result 599.
[DEBUG] [04-06-2020 10:00:20:513.257 GMT] stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200

If there is network issue and the stream entered this state, why it is resolved by restarting app and not by retry policy

MushMal, 2020-06-04:
@usmanm259 It really comes down to libCurl timing out. If I were to guess, there is something wrong with the OS network stack/sockets. Try to troubleshoot your system by checking the available memory, opened sockets perhaps. 
Recommend building debug version of libCurl, capture any output when this happens, open an issue with libCurl GitHub repository perhaps they can give better answers