awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #457: How to send live stream from camera only when a face is detected?

sumanghosh12, 2020-05-17:
Need some pointer on how to send live stream from a camera only when a face is detected in the stream?

With the current design I am sending the camera feed continuously to Kinesis Video Steam via Producer SDK- CPP. Only drawback with the current design is that I am invoking the Amazon Rekognition API continuously through out the day even though only around 30% of the time faces are in the live feed. This is result with high Reckinition cost.

Planning to add an additional program between Camera and Producer SDK which will use some Computer Vision library to only detect presence of person in the live camera feed and then send the stream to Kinesis through Producer SDK?

How to use Producer SDK to send stream to Kinesis only when a face is detected and dont send when there is no person in the feed? With the custom program I am able to detect faces but need some guidance or pointer on how to call ProducerSDK or send stream directly to Kinesis Video Stream?

sumanghosh12, 2020-05-17:
Addition to my previous Question:
As soon as I detect a face in the video frame how do I start the producer SDK to send the stream to Kinesis?
Is there is a way to send the frames with faces to Kineis Video Stream directly?

MushMal, 2020-05-17:
@sumanghosh12 the Producer SDK can send any types of frames. True, for certain types of streams, the SDK can perform further actions like extracting pixel width and height from the CPD and setting the appropriate MKV elements in case the underlying stream is video/h264 for example (with the appropriate flags) or extract audio properties for AAC for example. Other than that, the SDK treats the frames as binary blobs with timestamps and flags.
The backend acts similarly, exception being the console playback, HLS or MPEG-DASH session generation APIs.

Now, getting back to your question. You can simply create a stream as your session starts. This will create and prepare the client stream state machine and move to Ready state after calling some of the control plane APIs. Your application then starts processing frames. As soon as you determine that a frame is of some interest, you call put frame API to output into the SDK. You stop putting frames when there are no longer interesting frames.

Couple of points!

* The start frame should be a key-frame (I-frame for video) to not "skip" it
* The producing should be a series of continuous frames or some "skips" in between but logically, it should be a period which you are streaming.
* If you stop producing frames, the SDK will not yet receive the Persisted ACK as the SDK/backend don't know whether you will be adding more frames to the fragment to close and persist. If the gap between the last frame of this session and the next one is longer than max session timeout (30 seconds) your session will timeout. This is OK in the "intermittent" producer scenario and the SDK will not re-try if it had already sent out the last bits of the last frame. The SDK will re-stream the last fragment with the new session (when your application detects a new face and calls put frame after the current session expired underneath). 
However, if your application requires tight control on the persistence, as soon as you determine that no more frames need to be sent for this session, you could produce (put frame) as specialized EoFR frame in order to tell the SDK and the backend that your fragment is closed.

Please let me know if this made sense.

MushMal, 2020-05-26:
Resolving as stale. Please re-open or cut a new issue if you need further assistance

closed 2020-05-26 by MushMal