awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #458: Media preview screen is black for Kinesis stream, Vivotek FD9187-H camera

wrigsa, 2020-05-18:
I'm trying to set up my FD9187-H camera to stream to a Kinesis Video Stream. I've verified that I'm streaming to the correct region and trying to view the stream in the same region. I've also verified that the timecodes are correct. I'm using these settings on stream 2 on the camera:

- H.264
- 
- Frame size - 640x480
- 
- Maximum frame rate - 15 fps
- 
- Intra frame period - 1s
- 
- Smart Stream III
- 
- Smart codec / Mode - Auto Tracking
- Bit rate control
- 
- Constrained bit rate: Target Quality - Detailed, Maximum bit rate - 3 Mbps, Policy - Image quality priority, Smart Q - On

But it's not streaming to the kinesis video stream. I looked at issue #295 , but their fix didn't work for me.

MushMal, 2020-05-18:
Is there any way you could retrieve the producer side logs to understand what's going on?

You could check the CW logs as described here:
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/monitoring-cloudwatch.html

PutMedia.IncomingFragments should be increasing as you stream.

wrigsa, 2020-05-18:
18 14:15:24.499|ERROR| 5732|ProducerCallback.cpp:46|STATUS_SERVICE_CALL_NOT_AUTHORIZED_ERROR
18 14:15:54.211|ERROR| 5708|DefaultCallbackProvider.cpp:318|createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
18 14:15:54.715|ERROR| 5708|KinesisVideoProducer.cpp:147| Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 14:15:54.721|ERROR| 5708|ProducerBase.cpp:86|createStreamSync fail, reason:  Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 14:15:54.885|ERROR| 5736|KinesisUploader.cpp:172|EStreamCodec::H264
18 14:15:55.580|ERROR| 5760|ProducerCallback.cpp:41|Reporting stream error. Errored timecode 0 with status code 1375731728
18 14:15:55.586|ERROR| 5760|ProducerCallback.cpp:46|STATUS_SERVICE_CALL_NOT_AUTHORIZED_ERROR
18 14:16:25.336|ERROR| 5736|DefaultCallbackProvider.cpp:318|createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
18 14:16:25.837|ERROR| 5736|KinesisVideoProducer.cpp:147| Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 14:16:25.839|ERROR| 5736|ProducerBase.cpp:86|createStreamSync fail, reason:  Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 14:16:25.994|ERROR| 5940|KinesisUploader.cpp:172|EStreamCodec::H264
18 14:16:26.868|ERROR| 5964|ProducerCallback.cpp:41|Reporting stream error. Errored timecode 0 with status code 1375731728
18 14:16:26.868|ERROR| 5964|ProducerCallback.cpp:46|STATUS_SERVICE_CALL_NOT_AUTHORIZED_ERROR
18 14:16:56.599|ERROR| 5940|DefaultCallbackProvider.cpp:318|createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
18 14:16:57.100|ERROR| 5940|KinesisVideoProducer.cpp:147| Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 14:16:57.103|ERROR| 5940|ProducerBase.cpp:86|createStreamSync fail, reason:  Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf

But the Kinesis Video Stream IS created. When I tried deleting the video stream it was created again, but I keep getting the "unable to create kinesis video stream" error. I think the real issue here is the SATUS_SERVICE_CALL_NOT_AUTHORIZED_ERROR. How can I fix that?

MushMal, 2020-05-18:
@wrigsa what do you mean when you say "the stream IS created"? Have you created it in the console or through CLI?

The create state is reached when the describe state returns no streams. Can you verify that you are using the correct region? Also, please include complete logs.

wrigsa, 2020-05-18:
The stream creation is handled by a teleport plugin and a cloudformation stack. When I provision the camera with the correct credentials, the stream is created. 

I am using us-east-1 for all of my resources, and viewing from the same location. 

Complete logs below

 18 13:57:27.384|ERROR| 3094|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:28.485|ERROR| 3110|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:29.485|ERROR| 3111|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:30.485|ERROR| 3112|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:31.485|ERROR| 3113|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:32.485|ERROR| 3114|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:33.484|ERROR| 3115|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:34.484|ERROR| 3116|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:35.475|ERROR| 3117|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:36.475|ERROR| 3118|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:37.475|ERROR| 3119|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:38.475|ERROR| 3137|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:39.475|ERROR| 3138|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:40.475|ERROR| 3139|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:41.465|ERROR| 3140|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:42.466|ERROR| 3141|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:43.495|ERROR| 3142|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:44.495|ERROR| 3144|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:45.497|ERROR| 3145|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:46.495|ERROR| 3146|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:47.495|ERROR| 3147|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:48.495|ERROR| 3165|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:49.495|ERROR| 3166|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:50.495|ERROR| 3167|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:51.495|ERROR| 3168|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:52.495|ERROR| 3169|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:53.487|ERROR| 3170|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:54.494|ERROR| 3171|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:55.486|ERROR| 3172|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:56.484|ERROR| 3173|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:57.485|ERROR| 3174|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:58.485|ERROR| 3192|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:57:59.491|ERROR| 3193|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:00.475|ERROR| 3194|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:01.478|ERROR| 3195|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:02.477|ERROR| 3196|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:03.475|ERROR| 3197|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:04.475|ERROR| 3198|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:05.475|ERROR| 3199|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:06.475|ERROR| 3225|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:07.475|ERROR| 3226|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:08.474|ERROR| 3244|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:09.474|ERROR| 3245|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:10.505|ERROR| 3246|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:11.495|ERROR| 3247|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:12.497|ERROR| 3248|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:13.496|ERROR| 3249|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:14.495|ERROR| 3250|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:15.495|ERROR| 3251|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:16.495|ERROR| 3252|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:17.495|ERROR| 3253|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:18.494|ERROR| 3271|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:19.495|ERROR| 3272|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:20.494|ERROR| 3273|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:21.487|ERROR| 3274|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:22.495|ERROR| 3275|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:23.494|ERROR| 3276|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:24.485|ERROR| 3277|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:25.485|ERROR| 3284|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:26.484|ERROR| 3285|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:27.485|ERROR| 3286|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:28.475|ERROR| 3304|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:29.485|ERROR| 3305|KinesisUploader.cpp:172|EStreamCodec::H264
18 13:58:30.485|ERROR| 3306|KinesisUploader.cpp:172|EStreamCodec::H264
[... 965 lines cut ...]
18 15:45:01.078|ERROR|13903|ProducerCallback.cpp:46|STATUS_SERVICE_CALL_NOT_AUTHORIZED_ERROR
18 15:45:30.834|ERROR|13877|DefaultCallbackProvider.cpp:318|createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
18 15:45:31.335|ERROR|13877|KinesisVideoProducer.cpp:147| Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 15:45:31.338|ERROR|13877|ProducerBase.cpp:86|createStreamSync fail, reason:  Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 15:45:31.495|ERROR|13981|KinesisUploader.cpp:172|EStreamCodec::H264
[... cut ...]
18 15:47:04.685|ERROR|14065|KinesisVideoProducer.cpp:147| Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 15:47:04.688|ERROR|14065|ProducerBase.cpp:86|createStreamSync fail, reason:  Unable to create Kinesis Video stream. 439fdab7-be0f-effe-6d6b-5bbc296ff8cd Error status: 0xf
18 15:47:04.855|ERROR|14095|KinesisUploader.cpp:172|EStreamCodec::H264
18 15:47:05.603|ERROR|14146|ProducerCallback.cpp:41|Reporting stream error. Errored timecode 0 with status code 1375731728
18 15:47:05.604|ERROR|14146|ProducerCallback.cpp:46|STATUS_SERVICE_CALL_NOT_AUTHORIZED_ERROR

MushMal, 2020-05-18:
OK, understood. It seems that you indeed have an auth issue.

* Please describe how you integrate with auth - are you using access key/secret key or using IoT certificate provider integration or anything else?
* Make sure that the account you are using is configured to have the right level of access.

For example: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html

wrigsa, 2020-05-19:
I'm using the onica quick start cloudformation template to automatically create the following

- An AWS IoT policy to associate with connected cameras.
-     An AWS Identity and Access Management (IAM) role for connected cameras to stream to Kinesis Video Streams.
-     An Amazon DynamoDB table to store provisioning keys. Once provisioning is complete, you should remove the keys.
-     AWS Lambda functions to create a provisioning key and a role alias, verify the stack, and provision cameras.
-     Amazon API Gateway to expose provisioning endpoints through HTTPS.
-     Amazon CloudWatch alarms to expose camera streaming status through an Amazon Simple Notification Service (Amazon SNS) topic, and update the associated camera’s IoT thing shadow.

I also have a config app on my app. I provide the stack endpoint and provisioning key, both of which are outputs of the stack. 

When I provision the camera, and iot thing is created with a certificate that can assume a role that has the correct policy and trust relationship. I can change the certificate on the camera by uploading a new certificate, private key, and root ca.

MushMal, 2020-05-19:
It seems that you are missing some permissions as it results in not authorized error.

We are in the middle of updating the readme for the cpp but you can look at webrtc readme here: https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c#setup-iot

There is more info here: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/how-iot.html

MushMal, 2020-05-19:
One more thing.. I see the following in the log:

`18 13:58:43.332|ERROR| 3371|CredentialHandler.cpp:151|curl_easy_perform fail(): 3`

What is CredentialHandler.cpp? This is the first time the error is surfaced.

MushMal, 2020-05-26:
Resolving as stale due to no answer. Please follow-through with the suggestions. Please reach out to the manufacturer of the device for further help.

closed 2020-05-26 by MushMal

dmgardner88, 2021-07-12:
@wrigsa did you ever find a resolution for this? I'm getting the same error and have reached out to Vivotek for support. The quickstart that is referenced by Vivotek's instructions looks to be no longer supported by AWS.