awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #459: putMedia failed after long term upload

farleylai, 2020-05-18:
We have a video stream uploading to KVS without issues over the weekend but this afternoon, 
the KVS upload seems down between 13:51 to 14:30 EDT.

[image]

From the log messages in reverse order, the SDK seemed to retry with repeated requests and errors as follows:

```
May 18 14:10:05 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:10:04 DEBUG   curlCompleteSync(): HTTP Error 0 : Response: (null)
May 18 14:10:05 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:10:04 DEBUG   curlCompleteSync(): curl perform failed for url https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout 
May 18 14:10:05 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:29 DEBUG   postWriteCallback(): Curl post body write function for stream with handle: wyze-farley and upload handle: 301 returned: {"EventType":
May 18 14:10:05 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:29 DEBUG   postReadCallback(): Pausing CURL read for upload handle: 301
May 18 14:10:05 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:29 DEBUG   postReadCallback(): Wrote 60491 bytes to Kinesis Video. Upload stream handle: 301
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:29 DEBUG   writeHeaderCallback(): RequestId: d2943843-5e5f-c0a1-85a8-e9db4f2a30cd
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:29 DEBUG   postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 301
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   notifyDataAvailable(): Failed to un-pause curl with error: 43
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   putStreamResultEvent(): Put stream result event.
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94525807303785
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   getStreamingTokenResultEvent(): Get streaming token result event.
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   getStreamingEndpointResultEvent(): Get streaming endpoint result event.
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amaz
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   writeHeaderCallback(): RequestId: 49325393-4e03-4e02-b70e-eeb238373a09
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   describeStreamResultEvent(): Describe stream result event.
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.589182998353E9,"DataRetentionInHours":6,"De
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   writeHeaderCallback(): RequestId: b4cc5b24-43ec-4cf2-b437-a7e0cc15fe90
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000200, Next state: 0x0000000000000001
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   kinesisVideoStreamTerminated(): Stream terminated event.
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   putStreamCurlHandler(): Stream with streamHandle 94525807303785 has exited without triggering end-of-stream. Service call result: 599
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: 2020-05-18 18:09:28 DEBUG   putStreamCurlHandler(): Network thread for Kinesis Video stream: wyze-farley with upload handle: 300 exited. http status: 0
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     x-amzn-stream-name: wyze-farley
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     x-amzn-producer-start-timestamp: 1589825332.768
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     x-amzn-fragment-timecode-type: ABSOLUTE
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     x-amzn-fragment-acknowledgment-required: 1
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     X-Amz-Date: 20200518T180852Z
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     user-a
gent: AWS-SDK-KVS/2.1.0 GCC/5.4.0 Linux/4.15.0-72-generic x86_64
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     transfer-encoding: chunked
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     host: s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     content-type: application/json
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     connection: keep-alive
May 18 14:09:29 ml17-pc02 kvs.sh[2377]:     Authorization: AWS4-HMAC-SHA256 Credential=AKIA2WG7PM2NWCVGKM5F/20200518/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: Request Headers:
May 18 14:09:29 ml17-pc02 kvs.sh[2377]: Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia
```

I am not sure if it's the client side issue or something else on the AWS side since our ITS reports no network issue that time.
Any ideas?

MushMal, 2020-05-18:
This is really odd @farleylai 

From the logs it seems that your PutMedia session has timed out and the state machine re-tries but somehow call to curl_easy_pause returns CURLE_BAD_FUNCTION_ARGUMENT.

Generally, this means that the function is called with a bad parameter but I really can't understand how we end up with Curl getting into a state when it returns this value,

Is it possible if you could perhaps include a slightly longer version of the logs to understand whether there were other issues leading to the timeout and consequent error in unpausing curl.

Also, it seems that your stream had a significant outage - close to 30 minutes or so. Wonder if this is network related?

farleylai, 2020-05-19:
Sure, the log of the last 8 hours is attached FYI.

[kvs.log.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4647752/kvs.log.zip)

I believe your SDK log timestamps are in UTC and our messages are in EDT.
Whenever `putKinesisVideoFrame(...)` is about to call, a log message in the format of `Sending frame[XXXXXXX]` is print.

After a closer look, the HTTP Error actually happened once in a while and lasted for some time.
The first one occurred after `Sending frame[940326]` for 10 mins or so.
The second HTTP Error happened after `Sending frame[1028269]` for about half an hour.

Our program was deployed as a system service forwarding a video stream from a camera device to KVS.
If for some reason it receives an empty frame from the camera, the program reopens the stream and  connection to KVS that accidentally recovered the timeout issue in this case.
Otherwise, if the network was down, the connection to the camera through internet (not in LAN) would time out in seconds because our program configures `ffmpeg` to read with connection timeout set.
I also checked with our ITS staff who confirmed there was no network down time then.
So it feels like something else going around.
We really hope the issue can be clarified for long term streaming.

MushMal, 2020-05-19:
Here is what I see from the logs:

* Really good streaming which stabilizes around 500Kbps. There is little variation in the "current" buffer and most of the time there is almost zero buffer so your stream keeps up very well. 
* The persisted ACKs come at a regular cadence and your workset is ~1.5 seconds with 1 second fragment duration it makes it less than 500ms behind which is very cool.
* The first instance I see an issue happens is around Upload Handle 251. 

```
2020-05-18 16:01:57 DEBUG   postReadCallback(): Pausing CURL read for upload handle: 251
2020-05-18 16:02:32 DEBUG   curlCompleteSync(): curl perform failed for url https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
2020-05-18 16:02:32 DEBUG   curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIA2WG7PM2NWCVGKM5F/20200518/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=495457958cda5b5c29e67c4d933b7039b9435bf52b919308daa5ec5a3db9062d
    connection: keep-alive
    content-type: application/json
    host: s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com
    transfer-encoding: chunked
    user-agent: AWS-SDK-KVS/2.1.0 GCC/5.4.0 Linux/4.15.0-72-generic x86_64
    X-Amz-Date: 20200518T154634Z
    x-amzn-fragment-acknowledgment-required: 1
    x-amzn-fragment-timecode-type: ABSOLUTE
    x-amzn-producer-start-timestamp: 1589816794.437
    x-amzn-stream-name: wyze-farley
```

This means that either, your network slowed-down/stopped to trigger the curl timeout or, you are actually not producing any bits into the stream which causes the curl to timeout. 

In this case, as the SDK re-tries the control plane APIs just fine, it does seem that your application simply stops producing frames which causes the cycling from timeout -> control plane states -> put media -> timeout.

The curl error 43 which is CURLE_BAD_FUNCTION_ARGUMENT is an internal curl error that's generated when the SDK tries to pause the curl while it's actually about to fail with the Operation Too Slow error which is the timeout.

This lasts for some time and seems that this is the time your application detects the disconnection or something

`[05/18/2020 12:11:26] WARNING: Restarting streaming due to EOS/EOF`

From this moment on, it starts streaming just fine on the new handles.

========================

Conclusion.

Your application at some point starts acting as what we call an "unenlightened intermittent" producer. In this scenario, the application streams for a while and then stops producing frames into the SDK causing the backend to timeout (default 30 seconds - which is what we configure the CURL client as - 30 bytes in 30 second slow timeout). The latest PIC has logic to handle part of this - if the current buffer is empty then it won't prime (or drive) the state machine back to another PutMedia and leave the state machine in the Ready state waiting for the next time a frame is produced. You seem to be using older version of the CPP assets? 

Recommendation:

Recommend understanding your media pipelines behavior. If, for some reason, you have the media pipeline stopped, you can do one of the following (assuming latest codebase).

1) Issue StopSync, FreeStream. Wait until your pipeline needs to stream again, re-create the stream. In fact, the stream can be created earlier and waiting in the Ready state for the first frame. This way, you will have durability guarantees when doing StopSync as it will flush the buffer and await for the last persist ACK before returning.
2) Enlightened intermittent producer scenario: In this case, when your media pipeline stops producing frames, just issue a specialized EoFR frame which will cause flushing of the remainder of the buffer and the backend closing the fragment and issuing the persisted ACK

```
Frame eofr = EOFR_FRAME_INITIALIZER;
kinesis_video_stream->putFrame(eofr);
```

farleylai, 2020-05-19:
Thanks for the comprehensive analysis and recommendations.
Given the network was not down then, I think you are suggesting for some reason, the incoming device stream was sometimes being slow such that our program may not upload timely or sufficiently, and your SDK starts to log timeout events. Since `putKinesisVideoFrame(...)` is non-blocking, it should be straightforward to verify what is going on with reading the device stream then. 

BTW, the SDK version we built is 2.1.0. 
Is it already old?

MushMal, 2020-05-19:
We just released v3.0.0 that has the improvements including bug fixes and support for the intermittent producer scenario which is the likely case you are hitting.

I am not sure I understood your question about reading the device stream? 

Yes, in essence, in the REAL_TIME mode the put frame is a 'prompt' operation - it doesn't block and can/might cause frame to be dropped at the tail if there is no room in the content view (temporal view of the buffer).

In your case, it's kind of odd that you are seeing timeouts but the networking timeouts can happen not only when there is networking outage but also when no frames are being produced and the buffer gets depleted. The new handling of the intermittent producer will just ensure that after the initial timeout, we don't issue another PutMedia call if the buffer is already empty.

A really simple step (if durability is not primary concern for now) is to simply upgrade to v3.0.0 and run your scenario and stop producing frames for over 30 seconds - in this case, the SDK (if the buffer is empty) shouldn't issue another PutMedia. It will just stop at the Ready state. As soon as you issue a frame, it will move to PutStream state and issue PutMedia call.

NOTE: When resuming after the "pause" the frame has to be a key-frame or, otherwise, it will be "skipped" until a new key frame. This is important as the new session has to start with a new cluster.

farleylai, 2020-05-20:
Good to hear that. We will migrate to v3.0.0 soon.

Besides, let me elaborate our conclusion a bit. Our program serves as a forwarder/aggregator that receives the device stream over RTSP to upload to KVS. Given the timeout behavior of your SDK, the remaining mystery to verify is whether reading the device stream suddenly becomes very slow. We have set the read timeout of `ffmpeg` to read the device stream and made sure the read timeout exception would be thrown in seconds by cutting off the power of the device. From the logs, there is no such read timeout in the upload `blackout` or new frame sending messages. Then our guess is that there are still a small number of bytes readable from the device such that no read timeout is triggered but it is still too slow for `ffmpeg` to deliver a complete frame to upload to KVS. We will see if it is the root cause and appreciate your SDK deals with this situation gracefully.

MushMal, 2020-05-20:
Sounds good. @farleylai could you please resolve the issue when you have a chance? Please open a new issue specific to any issues you might see after the upgrade.

farleylai, 2020-05-20:
Sure thing!

closed 2020-05-20 by farleylai