awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #461: Building failing on windows [project_libopenssl]

BurguerJohn, 2020-05-20:
I am following the readme tutorial.
After executing the command `cmake .. -G "NMake Makefiles"`
It goes well until `Performing update step for 'project_libopenssl'`
My first error was that it was unable to find the command `Pearl`
So i installed the [padre-perl-ide](https://code.google.com/archive/p/padre-perl-ide/downloads) and it got me a little far in the instalation, but now it giving me this weird error:

```
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:59 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:78 (build_dependency)
```

Here is the log for my `CMakeOutput.log`

```
The system is: Windows - 10.0.17763 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe 
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) C/C++ Optimizing Compiler Versão 19.16.27032.1 para x86
Copyright (C) Microsoft Corporation. Todos os direitos reservados.

CMakeCCompilerId.c
Microsoft (R) Incremental Linker Version 14.16.27032.1
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:CMakeCCompilerId.exe 
CMakeCCompilerId.obj 

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.exe"

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.obj"

The C compiler identification is MSVC, found in "E:/Kinesis stream cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/3.17.2/CompilerIdC/CMakeCCompilerId.exe"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe 
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) C/C++ Optimizing Compiler Versão 19.16.27032.1 para x86
Copyright (C) Microsoft Corporation. Todos os direitos reservados.

CMakeCXXCompilerId.cpp
Microsoft (R) Incremental Linker Version 14.16.27032.1
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:CMakeCXXCompilerId.exe 
CMakeCXXCompilerId.obj 

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.exe"

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.obj"

The CXX compiler identification is MSVC, found in "E:/Kinesis stream cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/3.17.2/CompilerIdCXX/CMakeCXXCompilerId.exe"

Determining if the C compiler works passed with the following output:
Change Dir: E:/Kinesis stream cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_5240f\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_5240f.dir\build.make /nologo -L                  CMakeFiles\cmTC_5240f.dir\build

Building C object CMakeFiles/cmTC_5240f.dir/testCCompiler.c.obj

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\cl.exe @C:\Users\{my_user}\AppData\Local\Temp\nmF24F.tmp

testCCompiler.c

Linking C executable cmTC_5240f.exe

	"C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_5240f.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\mt.exe --manifests  -- C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_5240f.dir\objects1.rsp @C:\Users\{my_user}\AppData\Local\Temp\nmF3D7.tmp

Detecting C compiler ABI info compiled with the following output:
Change Dir: E:/Kinesis stream cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_03686\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_03686.dir\build.make /nologo -L                  CMakeFiles\cmTC_03686.dir\build

Building C object CMakeFiles/cmTC_03686.dir/CMakeCCompilerABI.c.obj

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\cl.exe @C:\Users\{my_user}\AppData\Local\Temp\nmF945.tmp

CMakeCCompilerABI.c

Linking C executable cmTC_03686.exe

	"C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_03686.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\mt.exe --manifests  -- C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_03686.dir\objects1.rsp @C:\Users\{my_user}\AppData\Local\Temp\nmF9B3.tmp

Determining if the CXX compiler works passed with the following output:
Change Dir: E:/Kinesis stream cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_80ab8\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_80ab8.dir\build.make /nologo -L                  CMakeFiles\cmTC_80ab8.dir\build

Building CXX object CMakeFiles/cmTC_80ab8.dir/testCXXCompiler.cxx.obj

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\cl.exe @C:\Users\{my_user}\AppData\Local\Temp\nmFE94.tmp

testCXXCompiler.cxx

Linking CXX executable cmTC_80ab8.exe

	"C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_80ab8.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\mt.exe --manifests  -- C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_80ab8.dir\objects1.rsp @C:\Users\{my_user}\AppData\Local\Temp\nmFF22.tmp

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: E:/Kinesis stream cpp/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_f1a07\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"  -f CMakeFiles\cmTC_f1a07.dir\build.make /nologo -L                  CMakeFiles\cmTC_f1a07.dir\build

Building CXX object CMakeFiles/cmTC_f1a07.dir/CMakeCXXCompilerABI.cpp.obj

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\cl.exe @C:\Users\{my_user}\AppData\Local\Temp\nm3A5.tmp

CMakeCXXCompilerABI.cpp

Linking CXX executable cmTC_f1a07.exe

	"C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_f1a07.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x86\mt.exe --manifests  -- C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\link.exe /nologo @CMakeFiles\cmTC_f1a07.dir\objects1.rsp @C:\Users\{my_user}\AppData\Local\Temp\nm423.tmp

```

Sean-Der, 2020-05-20:
Hey @BurguerJohn 

Sorry about that! Can you try running `cmake ..` again, I haven't been able to track it down but sometimes we have a race and OpenSSL fails to build on Windows.

For the 'source of truth' this is our [travis.yml](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.travis.yml#L110-L114) and [build_windows.bat](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat) we run this in CI. 

There are lots of doc improvements we can make, so when we get this resolved I will update the README (so other people don't have to deal with it as well)

thanks

BurguerJohn, 2020-05-20:
Thanks Sean, I used the `cmake ..` before installing Pear, so it had stopped at the previous error. Just gave another try and the installation went further than the `cmake .. -G "NMake Makefiles"` command, but still gave me a error before the end: (Sorry, my system is in portuguese)

```
"E:\Strawberry\perl\bin\perl.exe" "ms\uplink-x86_64.pl" auto crypto\uplink-x86_64.asm
  no assembler found on %PATH at ms\../crypto/perlasm/x86_64-xlate.pl line 103.
        ml64 /c /Cp /Cx /Zi /Focrypto\uplink-x86_64.obj "crypto\uplink-x86_64.asm"
  'ml64' não é reconhecido como um comando interno
  ou externo, um programa operável ou um arquivo em lotes.
NMAKE : fatal error U1077: 'ml64' : c¾digo de retorno '0x1' [E:\kin\stream\open-source\libopenssl\project_libopenssl.vcxproj]
  Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx86\x86\nmake.exe"' : c¾digo de retorno '0x2' [E:\kin\stream\open-source\libopenssl\project_libopenssl.vcxproj]
  Stop.
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:59 (message):
  CMake step for libopenssl failed: 1
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:78 (build_dependency)
```

My CMakeOutput.log:
```
The system is: Windows - 10.0.17763 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler:  
Build flags: 
Id flags:  

The output was:
0
Microsoft(R) Build Engine versão 15.9.21+g9802d43bc3 para .NET Framework 
Copyright (C) Microsoft Corporation. Todos os direitos reservados.

Compilação de 20/05/2020 15:29:47 iniciada.
Projeto "E:\kin\stream\build\CMakeFiles\3.17.2\CompilerIdC\CompilerIdC.vcxproj" no nó 1 (destinos padrão).
PrepareForBuild:
  Criando o diretório "Debug\".
  Criando o diretório "Debug\CompilerIdC.tlog\".
InitializeBuildStatus:
  Criando "Debug\CompilerIdC.tlog\unsuccessfulbuild" porque "AlwaysCreate" foi especificado.
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\CL.exe /c /nologo /W0 /WX- /diagnostics:classic /Od /Oy- /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"Debug\\" /Fd"Debug\vc141.pdb" /Gd /TC /analyze- /FC /errorReport:queue CMakeCCompilerId.c
  CMakeCCompilerId.c
Link:
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\link.exe /ERRORREPORT:QUEUE /OUT:".\CompilerIdC.exe" /INCREMENTAL:NO /NOLOGO kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:embed /PDB:".\CompilerIdC.pdb" /SUBSYSTEM:CONSOLE /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:".\CompilerIdC.lib" /MACHINE:X86 /SAFESEH Debug\CMakeCCompilerId.obj
  CompilerIdC.vcxproj -> E:\kin\stream\build\CMakeFiles\3.17.2\CompilerIdC\.\CompilerIdC.exe
PostBuildEvent:
  for %%i in (cl.exe) do @echo CMAKE_C_COMPILER=%%~$PATH:i
  :VCEnd
  CMAKE_C_COMPILER=C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\Hostx86\x86\cl.exe
FinalizeBuildStatus:
  Excluindo o arquivo "Debug\CompilerIdC.tlog\unsuccessfulbuild".
  Tocando "Debug\CompilerIdC.tlog\CompilerIdC.lastbuildstate".
Projeto de compilação pronto "E:\kin\stream\build\CMakeFiles\3.17.2\CompilerIdC\CompilerIdC.vcxproj" (destinos padrão).

Compilação com êxito.
    0 Aviso(s)
    0 Erro(s)

Tempo Decorrido 00:00:01.14

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CompilerIdC.exe"

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CompilerIdC.vcxproj"

The C compiler identification is MSVC, found in "E:/kin/stream/build/CMakeFiles/3.17.2/CompilerIdC/CompilerIdC.exe"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler:  
Build flags: 
Id flags:  

[... 99 lines cut ...]

  cmTC_44dfd.vcxproj -> E:\kin\stream\build\CMakeFiles\CMakeTmp\Debug\cmTC_44dfd.exe

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: E:/kin/stream/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/MSBuild/15.0/Bin/MSBuild.exe cmTC_1a5fd.vcxproj /p:Configuration=Debug /p:Platform=Win32 /p:VisualStudioVersion=15.0 /v:m && Microsoft(R) Build Engine versão 15.9.21+g9802d43bc3 para .NET Framework 

Copyright (C) Microsoft Corporation. Todos os direitos reservados.

  Microsoft (R) C/C++ Optimizing Compiler Versão 19.16.27032.1 para x86

  Copyright (C) Microsoft Corporation. Todos os direitos reservados.

  

  cl /c /Zi /W3 /WX- /diagnostics:classic /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_1a5fd.dir\Debug\\" /Fd"cmTC_1a5fd.dir\Debug\vc141.pdb" /Gd /TP /analyze- /errorReport:queue "C:\Program Files\CMake\share\cmake-3.17\Modules\CMakeCXXCompilerABI.cpp"

  CMakeCXXCompilerABI.cpp

  

  cmTC_1a5fd.vcxproj -> E:\kin\stream\build\CMakeFiles\CMakeTmp\Debug\cmTC_1a5fd.exe

```

Sean-Der, 2020-05-20:
@BurguerJohn Can you install NASM and then add it to your path? The follow command might be a little wrong, not at a Windows machine to try. But this is kind of what we do in travis.

```
choco install nasm
set PATH=C:\Program Files\NASM:%PATH%
```

BurguerJohn, 2020-05-20:
Alright, installed nasm and giving it a try, since it take a while to run I want to report that using the [build_window.bat](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat) got me really far in the build, I was positive it would work, but still gave me a error:

```
-- Finished building dependencies.
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
CMake Error at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
  Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:45 (find_package_handle_standard_args)
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:111 (find_package)
```

I can see in the Readme that pkg-config is a requirement, but i'm having a hard time finding on how to add it to windows. will try it with Nasm and later on see if i can get pkg-config on a windows.

Where is the log just in case:
```
The system is: Windows - 10.0.17763 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe 
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) C/C++ Optimizing Compiler Versão 19.16.27032.1 para x64
Copyright (C) Microsoft Corporation. Todos os direitos reservados.

CMakeCCompilerId.c
Microsoft (R) Incremental Linker Version 14.16.27032.1
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:CMakeCCompilerId.exe 
CMakeCCompilerId.obj 

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.exe"

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.obj"

The C compiler identification is MSVC, found in "E:/kin/stream/build/CMakeFiles/3.17.2/CompilerIdC/CMakeCCompilerId.exe"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.16.27023/bin/Hostx64/x64/cl.exe 
Build flags: 
Id flags:  

The output was:
0
Microsoft (R) C/C++ Optimizing Compiler Versão 19.16.27032.1 para x64
Copyright (C) Microsoft Corporation. Todos os direitos reservados.

CMakeCXXCompilerId.cpp
Microsoft (R) Incremental Linker Version 14.16.27032.1
Copyright (C) Microsoft Corporation.  All rights reserved.

/out:CMakeCXXCompilerId.exe 
CMakeCXXCompilerId.obj 

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.exe"

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.obj"

The CXX compiler identification is MSVC, found in "E:/kin/stream/build/CMakeFiles/3.17.2/CompilerIdCXX/CMakeCXXCompilerId.exe"

Determining if the C compiler works passed with the following output:
Change Dir: E:/kin/stream/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_8f200\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"  -f CMakeFiles\cmTC_8f200.dir\build.make /nologo -L                  CMakeFiles\cmTC_8f200.dir\build
[... 32 lines cut ...]

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe @C:\Users\Gabriel\AppData\Local\Temp\nm7BD6.tmp

testCXXCompiler.cxx

Linking CXX executable cmTC_4e576.exe

	"C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_4e576.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\cmTC_4e576.dir\objects1.rsp @C:\Users\Gabriel\AppData\Local\Temp\nm7C64.tmp

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: E:/kin/stream/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_c3466\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"  -f CMakeFiles\cmTC_c3466.dir\build.make /nologo -L                  CMakeFiles\cmTC_c3466.dir\build

Building CXX object CMakeFiles/cmTC_c3466.dir/CMakeCXXCompilerABI.cpp.obj

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe @C:\Users\Gabriel\AppData\Local\Temp\nm8089.tmp

CMakeCXXCompilerABI.cpp

Linking CXX executable cmTC_c3466.exe

	"C:\Program Files\CMake\bin\cmake.exe" -E vs_link_exe --intdir=CMakeFiles\cmTC_c3466.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100177~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\cmTC_c3466.dir\objects1.rsp @C:\Users\Gabriel\AppData\Local\Temp\nm8117.tmp

```

Sean-Der, 2020-05-20:
@BurguerJohn I think we get `pkg-config` from `choco install strawberryperl` 

It is a hard requirement for GStreamer plugin, but we could put that behind a 'are you building GStreamer' check maybe!

BurguerJohn, 2020-05-20:
Ok, i installed `nasm / strawberryperl /  gstreamer `
Following by `gstreamer-devel --version=1.16.2`
Just like the file you linked. The build worked for about half-hour, but still ended in error:

```
CMake Warning (dev) at CMakeLists.txt:135 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Checking for module 'gstreamer-app-1.0'
--
CMake Error at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:467 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.17/Modules/FindPkgConfig.cmake:647 (_pkg_check_modules_internal)
  CMakeLists.txt:211 (pkg_check_modules)
```

CMakeError.log:
```
Determining if the include file pthread.h exists failed with the following output:
Change Dir: E:/kin/stream/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake /nologo cmTC_cd9bc\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"  -f CMakeFiles\cmTC_cd9bc.dir\build.make /nologo -L                  CMakeFiles\cmTC_cd9bc.dir\build

Building C object CMakeFiles/cmTC_cd9bc.dir/CheckIncludeFile.c.obj

	C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe @C:\Users\Gabriel\AppData\Local\Temp\nmABA.tmp

CheckIncludeFile.c

E:\kin\stream\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Não é possível abrir arquivo incluir: 'pthread.h': No such file or directory

NMAKE : fatal error U1077: 'C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx64\x64\cl.exe' : c¾digo de retorno '0x2'

Stop.

NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : c¾digo de retorno '0x2'

Stop.
```

I don't want to be rude and I really appreciate the help, it may be my fault since I just started getting into the streaming services, but i never hit so much walls on other services than trying to make Amazon Kinesis work, I can't even make a simple test to tell my bosses if this service is good for us.

I managed to make it work with [BrainPower](https://github.com/brain-power/Brain-Power-Amazon-Fidgetology) after a lot of struggle. But it really is not the same as saying that i was able to use this service.

Looking at the Kinesis page it seen that this project is the entry point to start to work with the Kinesis stream, but I had trouble with this and the Java project, and i just want to make a successful test to tell my bosses if this look ok for own needs.

I don't want to lash out on nobody here, I'm also a programmer and i know how this kinda of stuff is, but my suggestion is that Amazon priorize a little this service, since I'm almost giving it up and using another service outside of Amazon.

Sean-Der, 2020-05-20:
@BurguerJohn totally understand why you are frustrated, it is annoying when stuff just doesn't work!

In the future I want to distribute binary builds so you don't have to build stuff, but for now all I can do is help fix the build.

It looks like GStreamer failed to be discovered via `package-config` I am not exactly sure why though. If you run `C:\gstreamer\1.0\x86_64\bin\pkg-config.exe --libs gstreamer-app-1.0` does that work?

BurguerJohn, 2020-05-21:
Sorry for the delay @Sean-Der my shift had ended.
You was right about my error. The problem was that I had installed gstreamer in my E:/
So I moved it to my C:/ and the build worked! Thanks.

After that it was missing a few DLL once i tried any of the .exe files, so I had to add a gstream and the open-source folder to my path and now the .exe open (and instantly close) without any error.
Are the .exe files are suppose to open and close like that?

This command` cmd.exe /c cmake -G "NMake Makefiles" ..` still give me the same error from the first post `NMAKE : fatal error U1073: Don't know how to create '"crypto\aes\aes_cfb.c"'` but cmake -G `"NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..` made the build work.

Just a few notes to help others with similar problems:
- Make sure nmake and cmake are working corretly.
- Install choco and do all instalation from [travis](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.travis.yml#L110-L114) and use this [bat](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat)
- Make sure gstreamer is at `C:\gstreamer\1.0\`

BurguerJohn, 2020-05-21:
Following the readme and [this link](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html)
I did try to add a few folders to my path and call.
`gst-inspect-1.0 kvssink` and this command `gst-launch-1.0 rtspsrc location="rtsp://YourCameraRtspUrl" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="YourStreamName" storage-size=512  access-key="YourAccessKey" secret-key="YourSecretKey" aws-region="YourAWSRegion"`

Both show me this error:
[image]

Perhaps I try again the Java version since now i'm able to build the dll files.

aeperaltan, 2020-05-21:
Hello, we have similar issues for both Windows and Mac. We are using a .NET environment and followed the procedure:

git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git 
mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build
cmake .. -G "NMake Makefiles"

As for windows, we get errors related to openssl. For Mac we get issues related to liblog4cplus.

I have attached both files.
[kinesis_log_windows.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4663144/kinesis_log_windows.txt)
[cmake_log_mac.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4663146/cmake_log_mac.txt)

Please advise,
Thanks!

lpawlaczyk, 2020-05-21:
You should use .github/build_windows.bat 
In my case I changed the first line to:
call "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Auxiliary\Build\vcvars64.bat"
Another thing is to make sure you have downloaded and populated directories in:
open-source/amazon-kinesis-video-streams-pic and open-source/amazon-kinesis-video-streams-producer-c. This should easily build what you need.

amilaDulanjana, 2020-05-25:
I am also having this issue. I used [build_windows.bat](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat), but i changed the call to "C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Auxiliary\Build\vcvars64.bat". 

The issue is related to ` libopenssl`

error : 
` C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe  /I "." /I "crypto\include" /I "include" -DOPENSSL_USE_APPLINK -DDSO_WIN32 -DNDEBUG -DOPENSSL_THREADS -DOPENSSL_NO_STATIC_ENGINE -DOPENSSL_PIC "-DENGINESDIR=\"D:\\Projects\\iot-cam\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\\lib\\engines-1_1\"" "-DOPENSSLDIR=\"D:\\Projects\\iot-cam\\amazon-kinesis-video-streams-producer-sdk-cpp\\open-source\\local\"" -W3 -wd4090 -Gs0 -GF -Gy -nologo -DOPENSSL_SYS_WIN32 -DWIN32_LEAN_AND_MEAN -DL_ENDIAN -D_CRT_SECURE_NO_DEPRECATE -DUNICODE -D_UNICODE /MD /O2 /Zi /Fdossl_static -c /Focrypto\aes\aes_cbc.obj "crypto\aes\aes_cbc.c"
NMAKE : fatal error U1073: don't know how to make '"crypto\aes\aes_cfb.c"'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bin\Hostx64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: 'echo' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:59 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:78 (build_dependency)`

Please find error logs

[windows_error_logs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4678489/CMakeOutput.log)

MushMal, 2020-05-25:
@amilaDulanjana for some odd reason building open ssl on Windows at times errs out like you've described. However, running the build command again seems to work. Could you please retry building and let us know whether it solves? Don't clean up or anything.

nfvelado, 2020-05-26:
Hi, on my Windows 7 PC using Visual Studio 2019, I've followed the steps above in combination with the build_windows.bat (that I modified to point to my VS19 bat)...

git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
build_windows.bat

Initially, I was having the same openSSL build issues before using the build_windows.bat file.  I'm now stuck at the same gstreamer point mentioned above...

I've tried installing both the VS and MinGW versions of GStreamer from their site, into c:\gstreamer as suggested.  In both cases, build produces this:

-- Checking for module 'gstreamer-app-1.0'
--   No package 'gstreamer-app-1.0' found
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/FindPkgConfig.cmake:458 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindPkgConfig.cmake:637 (_pkg_check_modules_internal)
  CMakeLists.txt:211 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "C:/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log"
See also "C:/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

Microsoft (R) Program Maintenance Utility Version 14.24.28316.0
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1064: MAKEFILE not found and no target specified
Stop.

Running this from your question above...
C:\gstreamer\1.0\x86_64\bin\pkg-config.exe --libs gstreamer-app-1.0

C:\AWS\amazon-kinesis-video-streams-producer-sdk-cpp\build>c:\gstreamer\1.0\x86_64\bin\pkg-config.exe --libs gstreamer-app-1.0
Package gstreamer-app-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `gstreamer-app-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'gstreamer-app-1.0' found

Sean-Der, 2020-05-26:
@nfvelado You will also need to run the `GStreamer development installer` to get the headers and pkg-config files required to build the GStreamer extension.

Sean-Der, 2020-05-26:
@aeperaltan For your Mac build it seems to be using a broken CXX compiler, not sure why though! Can you try doing

```
export CC=/usr/bin/clang 
export CXX=/usr/bin/clang++
```

and try again with a brand new build directory?

nfvelado, 2020-05-26:
@Sean-Der appreciate the clarification there.  I've installed the msvc development installer, and am getting a different gstreamer related error now:

-- Checking for module 'gstreamer-app-1.0'
--
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/FindPkgConfig.cmake:458 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindPkgConfig.cmake:637 (_pkg_check_modules_internal)
  CMakeLists.txt:211 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "C:/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "C:/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

Microsoft (R) Program Maintenance Utility Version 14.24.28316.0
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1064: MAKEFILE not found and no target specified
Stop.

Is installing gstreamer with the typical settings enough, and with the msvc build?  Is there some sort of environment variable that should be set, or does the build assume it's present in the C:/gstreamer directory?  What is the gstreamer version that you guys have tested with (I installed 1.16.2)?  I noticed burgeor john reported gstreamer errors once he got the build to succeed, but want to ensure I'm reproducing the exact environment that has been tested.  Thanks,

Nick

amilaDulanjana, 2020-05-26:
@MushMal thanks it works. But now i didn't able to stream through my web cam 

The command i used 
`gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name="video_stream" access-key="XXXXXXX" secret-key="XXXXXXXXXXX"`

I am getting following out put. Please help me I need to solve this.

Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2020-05-26 23:56:00 [19136] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2020-05-26 23:56:00 [19136] INFO - heapInitialize(): Creating AIV heap.
2020-05-26 23:56:00 [19136] INFO - heapInitialize(): Heap is initialized OK
2020-05-26 23:56:00 [19136] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-05-26 23:56:00 [19136] DEBUG - getSecurityTokenHandler invoked
2020-05-26 23:56:00 [19136] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 15905175604846254 Expiration: 0
2020-05-26 23:56:00 [19136] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-05-26 23:56:00 [19136] INFO - createDeviceResultEvent(): Create device result event.
2020-05-26 23:56:00 [19136] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-05-26 23:56:00 [19136] DEBUG - clientReadyHandler invoked
2020-05-26 23:56:00 [19136] INFO - try creating stream
2020-05-26 23:56:00 [19136] INFO - Creating Kinesis Video Stream iot_camera_stream
2020-05-26 23:56:00 [19136] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Stream name: iot_camera_stream
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Streaming type: STREAMING_TYPE_REALTIME
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Content type: video/h264
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Max latency (100ns): 600000000
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Fragment duration (100ns): 20000000
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Key frame fragmentation: Yes
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Use frame timecode: Yes
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Absolute frame timecode: Yes
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Nal adaptation flags: 0
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Average bandwith (bps): 4194304
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Framerate: 25
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Buffer duration (100ns): 1200000000
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Replay duration (100ns): 400000000
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Connection Staleness duration (100ns): 600000000
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Store Pressure Policy: 1
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    View Overflow Policy: 1
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Segment UUID: NULL
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Frame ordering mode: 0
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo(): Track list
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Track id: 1
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Track name: kinesis_video
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Codec id: V_MPEG4/ISO/AVC
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Track type: TRACK_INFO_TYPE_VIDEO
2020-05-26 23:56:00 [19136] DEBUG - logStreamInfo():    Track cpd: NULL
2020-05-26 23:56:00 [19136] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-05-26 23:56:00 [19136] INFO - writeHeaderCallback(): RequestId: 45fcc09a-307b-4ae2-825b-8cb42d0e4351
2020-05-26 23:56:01 [9768] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.590515361706E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:490744837685:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:490744837685:stream/iot_camera_stream/1590515361706","StreamName":"iot_camera_stream","Version":"kGXITi65soKYFPpHOvlS"}}
2020-05-26 23:56:01 [9768] INFO - describeStreamResultEvent(): Describe stream result event.
2020-05-26 23:56:01 [9768] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2020-05-26 23:56:01 [9768] INFO - writeHeaderCallback(): RequestId: 24edaf4c-d112-4e88-8c22-3cdabac62e10
2020-05-26 23:56:03 [24704] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
2020-05-26 23:56:03 [24704] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2020-05-26 23:56:03 [24704] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
[... cut ...]
2020-05-26 23:56:13 [11036] DEBUG - postReadCallback(): Wrote 37599 bytes to Kinesis Video. Upload stream handle: 0
2020-05-26 23:56:13 [11036] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 23:56:13 [11036] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2020-05-26 23:56:13 [11036] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: iot_camera_stream and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":1590517568884,"FragmentNumber":"91343852333181447247962534053423835832617768593"}

MushMal, 2020-05-26:
Folks, it seems that this thread is quite long and for some odd reason I don't see some of the replies any longer - perhaps GitHub has a limit on the length of the issue.

Just some comments (as I see in my email inbox but not here).

@amilaDulanjana the small snippet of log shows no problems at all with streaming. We had a printing bug/issue which is since fixed in PIC but not propagated to the CPP layer where the getStreamData returns 0x5200002f as an error but in fact, this is a notification only - it means no more data to send. This is good as your networking is keeping up just fine.

While I don't understand why you are unable to preview your stream, here are couple of suggestions:

* Make sure you are streaming and viewing in the same Region.
* Make sure you have video that's "playable" by the console - it should be h264 or h265
* Make sure you either have AvCC format for the video stream. Most encoders produce Annex-B format so there are flags to adapt this to AvCC format. The KVS plugin will do that for you - same with the samples.
* Cut a separate issue to track the playback.

Regarding Windows build issues. While we won't be able to solve all of the open source component builds on Windows, we can help sharing the knowledge but I would also strongly encourage folks to check the corresponding open source community pages.

amilaDulanjana, 2020-05-26:
@MushMal  thank you very much for your supoort. Finally it works.

nfvelado, 2020-05-26:
@MushMal regarding the questions and issues I'm still having related to GStreamer, the assumptions the build is making is not clear to me at all.  I see comments above stating gstreamer should be in the c:\gstreamer directory, from downloading it separately from the gstreamer website.  I was under the impression this should get pulled down automatically during the build process, so I'm guessing that manually downloading it is a work-around.  If so, what needs to change in the build to accommodate that? 

There is a build_windows.bat mentioned above that I pulled in from the link provided, but the documentation doesn't seem to be updated with the latest methods to really get this build to work under Windows.  The build scripts don't get pulled in automatically as the documentation states, or they're no longer in the repo as they were at one point, so there is a bit of a manual process to make this work.

I'm not clear on what version of gstreamer this has been tested with, what development version of gstreamer you're expecting (msvc or mingw), what, if anything, should be changed if the gstreamer development build is downloaded an installed into c:\gstreamer.

Basically, what exactly are the steps required to build this with gstreamer under windows?  Here's exactly what I tried:

1) Get the repo:
git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build

1) Downloaded (tried with mingw and vs) development builds taking defaults from here (assuming 1.16 is the right version):  https://gstreamer.freedesktop.org/download/

2) Included the build_windows.bat file and modified it to look like this:
call "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvars64.bat"
mkdir build
cd build
cmd.exe /c cmake -G "NMake Makefiles" ..
cmake -G "NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..
nmake

Should -DBUILD_GSTREAMER_PLUGIN=TRUE be in here given the work-around with the manual download?  Is there anything else that needs to get set or unset, the environment or elsewhere?

Merely trying to piece together the build assumptions that are being made here.  I've collectively spent about 3 months off and on trying to build this under Windows in my spare time with no luck.  Thanks

aeperaltan, 2020-05-28:
Please confirm if there will be another thread on this topic. Thanks

nfvelado, 2020-05-28:
Guys, just a quick update on this also...  -DBUILD_GSTREAMER_PLUGIN=TRUE does seem to be necessary in order to build the kvssink part of the application, from the readme documentation.

It appears that choco is needed in order to install gstreamer via the packages method, however, this still requires some additional step that I haven't been able to solve.  I'm guessing the build doesn't automatically install gstreamer like the documentation seems to infer, so this is an additional step needed by doing the following...and again, this is from guessing and playing around with this.

1) Install choco referened by Sean above (which I've never heard of before, but apparently this is chocolatey.org) by following this youtube video, then it'll be accessible via command prompt:

https://www.google.com/search?q=windows+install+choco&oq=windows+install+choco&aqs=chrome..69i57j0l7.7711j0j9&sourceid=chrome&ie=UTF-8#kpvalbx=_JRDPXp6gMoyO9PwP2fKQ4AM41

2) Install gstreamer via choco (there is a note above that nasm is also needed, but I don't know that it really is.  Also, I manually installed strawberry perl, but it appears you can do this via choco also)
choco install gstreamer-devel --version=1.16.2

When it finishes installing into your c:\ root, you'll see the following message...

PATH environment variable does not have %GSTREAMER_1_0_ROOT_X86_64%\bin in it. Adding...
  gstreamer-devel may be able to be automatically uninstalled.

But, it doesn't actually seem to set GSTREAMER_1_0_ROOT_X86_64 to anything, and I don't know that this matters.

3) Install pkg-config via choco too, since pkg-config is what's getting called by perl to locate the gstreamer plugin (I'm guessing from the build output shown below)
choco -install pkg-config

This is where I'm getting stuck.  The build calls pkg-config on gstreamer-app-1.0.  This lives in C:\gstreamer\1.0\x86_64\lib\pkgconfig, however, doesn't seem to be referenced via the GSTREAMER_1_0_ROOT_X86_64 environment variable.  It's not like the build seems to have any knowledge of where gstreamer was installed by choco, nor does installing via choco seems to set any relevant path variables that the build uses(????)

CMake seems to rely on pkg-config, which according to google, seems to use a path variable named PKG_CONFIG_PATH.  So, if you run set PKG_CONFIG_PATH=%PKG_CONFIG_PATH%;c:\gstreamer\1.0\x86_64\lib\pkgconfig, you can verify that running pkg-config --list-all now sees gstreamer-app-1.0, whereas it didn't before this variable was set.  However, running build_windows.bat at this point still encounters the error:

-- Checking for module 'gstreamer-app-1.0'
--
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/FindPkgConfig.cmake:458 (message):
  A required package was not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindPkgConfig.cmake:637 (_pkg_check_modules_internal)
  CMakeLists.txt:211 (pkg_check_modules)

So, there's got to be some sort of variable that needs to be set somewhere to get cmake to see this, I just haven't been able to find the missing link still.

disa6302, 2020-06-15:
@aeperaltan , 
You can open a new issue with the details.

closed 2020-06-15 by disa6302

mggunman, 2020-06-24:
> Following the readme and [this link](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html)
> I did try to add a few folders to my path and call.
> `gst-inspect-1.0 [...]

So I managed to fix this issue as I started a discusion on the mailinglist for GStreamer. The issue was that I had installed numerous copies of GStreamer both the runtime and development version and apparently each time you switch between MinGW or MSVC version or download a new copy of GStreamer you need to refresh the cache for GStreamer by just deleting it, I couldn't locate the cache on AppData for GStreamer as it moves around and at times is in a different location for others.
What I did was uninstall all GStreamer components including the MinGW, MSVC and the chocolately version, restarted the machine and installed only the MSVC runtime and development msi installers and got GStreamer to work. Another useful tip would be to move the GStreamer bin and lib path directories above strawberry perls path in the PATH variable as I was told by one the devs the glib files could be getting picked up from strawberry perl instead of GStreamer.