awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #462: Build Failed in Ubuntu 18.04

timetxt, 2020-05-23:
$ uname -a
Linux  5.3.0-46-generic #38~18.04.1-Ubuntu SMP Tue Mar 31 04:17:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

`CMake Error at CMakeLists.txt:95 (add_subdirectory):
  The source directory

    /home/jason/kvs4/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-producer-c

  does not contain a CMakeLists.txt file.

CMake Error at CMakeLists.txt:106 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.`

Looks like CMAKE cannot find cURL lib. But I checked cURL lib has been installed 

$ sudo apt-get install libcurl4-openssl-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libcurl4-openssl-dev is already the newest version (7.58.0-2ubuntu3.8).

Any suggestion how to resolve this? Thanks

MushMal, 2020-05-23:
have you cloned the dependent repository as in readme?

`git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git`

Seems that the directory is empty

timetxt, 2020-05-29:
> have you cloned the dependent repository as in readme?
> 
> `git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git`
> 
> Seems that the directory is empty

I think the option --recursive was the cause. Compile successfully now

closed 2020-05-29 by timetxt