awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #463: Device '/dev/video0' is busy

joyuiyeong, 2020-05-25:
Hello, guys.

I have problem when running 'kvs_gstreamer_sample'.
when I excute ``` ./kvs_gstreamer_sample SomeStreamName```, it stop after few seconds with following errors

```
[DEBUG] [25-05-2020 05:16:46:466.232 GMT] streamReadyHandler invoked
[DEBUG] [25-05-2020 05:16:46:466.449 GMT] Stream is ready
[INFO ] [25-05-2020 05:16:46:539.252 GMT] Streaming from live source
[DEBUG] [25-05-2020 05:16:46:539.588 GMT] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
0:00:00.078980691 20478 0xaaaaae4e4f80 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "vtenc_h264_hw"!
0:00:00.079067894 20478 0xaaaaae4e4f80 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:456:gst_element_factory_make: no such element factory "omxh264enc"!
0:00:00.100732715 20478 0xaaaaae4e4000 WARN                 default v4l2-utils.c:189:gst_v4l2_error:<source> error: Device '/dev/video0' is busy
0:00:00.100790863 20478 0xaaaaae4e4000 WARN                 default v4l2-utils.c:192:gst_v4l2_error:<source> error: Call to S_FMT failed for H264 @ 640x480: Device or resource busy
0:00:00.100898732 20478 0xaaaaae4e4000 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<source> error: Internal data stream error.
0:00:00.100929528 20478 0xaaaaae4e4000 WARN                 basesrc gstbasesrc.c:3055:gst_base_src_loop:<source> error: streaming stopped, reason not-negotiated (-4)
Error received from element source: Device '/dev/video0' is busy
Debugging information: gstv4l2object.c(3754): gst_v4l2_object_set_format_full (): /GstPipeline:live-kinesis-pipeline/GstV4l2Src:source:
Call to S_FMT failed for H264 @ 640x480: Device or resource busy
[INFO ] [25-05-2020 05:16:46:568.530 GMT] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000aaaaae283b28.
[DEBUG] [25-05-2020 05:16:46:568.769 GMT] streamClosedHandler invoked for upload handle: 18446744073709551615
```

Error message says, `/dev/video0' is busy. but I don't know how to handle this.

Please help me. Thank you.

disa6302, 2020-05-25:
@joyuiyeong ,

Can you check if have another such process running which might be using the device? The device can only be accessed exclusively by only one process at a time.

joyuiyeong, 2020-05-25:
@disa6302 
A script is running with kvs_gstreamer_sample.

Script getting sensor data like humidify, atmosphere with extention board(I use this as importing DFRobot_RaspberryPi_Extension_Board) is running on my RPi.

And also I send that data to AWS IoT core with AWSIoTPythonSDK.

Does that python script cause this? I don't think it access the device..

But, when I didn't run that script, ```./kvs_gstreamer_sample``` was worked properly

How can I found trouble?

MushMal, 2020-05-25:
@joyuiyeong per your debug/troubleshooting I think the other script (not sure what it is) might be accessing the camera and thus the GStreamer is unable to get a lock.

I suggest opening an issue with whoever owns the script.

I am going to resolve this as it's not related to KVS

closed 2020-05-25 by MushMal