awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #465: What is the name of the gstreamer audio track?

ajhool, 2020-05-26:
When trying to forward the audio track to Transcribe, we are unable to find the correct name. It also does not seem to be a parameter for kvssink. We are using this command to execute gstreamer:

```
gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,width=640,height=480,framerate=30/1,format=I420 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name="testing-kvs-to-transcribe" alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.
```
Ubuntu laptop
us-east-1
Audio + Video is working in the KVS console

lherman-cs, 2020-05-26:
@ajhool Do you mind to clarify what you mean by "we are unable to find the correct name. It also does not seem to be a parameter for kvssink"? 

Also, what was the output from the command?

ajhool, 2020-05-26:
@lherman-cs 

> Do you mind to clarify what you mean by "we are unable to find the correct name. It also does not seem to be a parameter for kvssink"?

We are trying to send a KVS stream to AWS Transcribe. The closest ref arch that we could find was: https://github.com/amazon-connect/amazon-connect-realtime-transcription 

That ref arch uses AWS connect with names the audio track ("FROM_CUSTOMER") ("TO_CUSTOMER"). However, we would like to use the gstreamer producer library to produce the video. Then, we are using a lambda function with the Kinesis Video Parser Library to (attempt to) forward the audio track to AWS transcribe.

We were able to see the video with audio in the Kinesis Video Console. However, the lambda function and the Kinesis Video Parser Library are checking for an audio track by name and are unable to find one. I don't see a way to name the audio track and am not sure what the default audio track name would be. Thus, Transcribe is returning an error because it is not receiving audio for 15 seconds.

> Also, what was the output from the command

The output of gstreamer was a long log while the video was recording. Here is a sample:

```
2020-05-26 15:09:05 [139987186075392] DEBUG - fragmentAckReceivedHandler invoked
2020-05-26 15:09:05 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:05 [139987186075392] DEBUG - postReadCallback(): Wrote 766 bytes to Kinesis Video. Upload stream handle: 0
2020-05-26 15:09:05 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:05 [139987186075392] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2020-05-26 15:09:05 [139987186075392] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: testing-kvs-to-transcribe-may26-02 and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":1590520143880,"FragmentNumber":"91343852333181432932423919757287009865442279106"}

2020-05-26 15:09:05 [139987186075392] DEBUG - fragmentAckReceivedHandler invoked
2020-05-26 15:09:05 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:05 [139987186075392] DEBUG - postReadCallback(): Wrote 3419 bytes to Kinesis Video. Upload stream handle: 0
2020-05-26 15:09:05 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:05 [139987186075392] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2020-05-26 15:09:05 [139987186075392] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 134217728
	>> Available storage byte size: 134099426
	>> Allocated storage byte size: 118302
	>> Total view allocation byte size: 576080
	>> Total streams frame rate (fps): 25579
	>> Total streams transfer rate (bps): 686048 (669 Kbps)
	>> Current view duration (ms): 743
	>> Overall view duration (ms): 1477
	>> Current view byte size: 47875
	>> Overall view byte size: 113070
	>> Current frame rate (fps): 25579.1
	>> Current transfer rate (bps): 686048 (669 Kbps)
2020-05-26 15:09:06 [139987613009664] DEBUG - postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2020-05-26 15:09:06 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:06 [139987186075392] DEBUG - postReadCallback(): Wrote 16827 bytes to Kinesis Video. Upload stream handle: 0
2020-05-26 15:09:06 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:06 [139987186075392] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2020-05-26 15:09:06 [139987186075392] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: testing-kvs-to-transcribe-may26-02 and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":1590520145372,"FragmentNumber":"91343852333181432937375679914428531422785831960"}

2020-05-26 15:09:07 [139987186075392] DEBUG - fragmentAckReceivedHandler invoked
2020-05-26 15:09:07 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:07 [139987186075392] DEBUG - postReadCallback(): Wrote 5063 bytes to Kinesis Video. Upload stream handle: 0
2020-05-26 15:09:07 [139987186075392] ERROR - getKinesisVideoStreamData(): operation returned status code: 0x5200002f
2020-05-26 15:09:07 [139987186075392] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2020-05-26 15:09:07 [139987186075392] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: testing-kvs-to-transcribe-may26-02 and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":1590520146860,"FragmentNumber":"91343852333181432942327440071570052828111090384"}
```

lherman-cs, 2020-05-26:
Did you get the video from our kvssink gstreamer plugin? If yes, the audio track name is going to be called "audio", https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L380.

But, if you used the producer sdk directly, you can configure the audio track's name programmatically, as shown in the example, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_audio_video_sample.cpp#L682.

ajhool, 2020-05-26:
@lherman-cs Great, "audio" is the track name, thank you! Is there a place to find some of the information about the stream produced by the gstreamer plugin? For example, I had trouble finding the track name and am now having trouble finding the audio sample rate? The video seems to be making its way into transcribe but no transcription events are occurring

lherman-cs, 2020-05-26:
Yes, I would say the gstreamer folder is the best source, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/src/gstreamer, especially the [gstkvssink.cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp) file. That's how we define the plugin to gstreamer.

For the audio sample rate, it's defined here, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L343.

MushMal, 2020-05-26:
To add to what @lherman-cs states. There is a set of audio specific EBML elements in MKV specification that the producer SDK specifies if the track is Audio and is of a known type and has CPD (codec private data). This is done automatically for you by the kvs sink and packaged in PIC. If this is what you are after then check this out:

Here is where it extracts:
https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/src/MkvGenerator.c#L1844

Here is where it actually packages audio specific bits
https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/mkvgen/src/MkvGenerator.c#L127

ajhool, 2020-05-27:
@MushMal @lherman-cs Thanks for the responses and references. Part of my confusion was that I don't think h264 encoding has a defined sample rate in code. When I review the video after KVS ingestion, I see the sample rate is at 44100Hz.

I am still having difficulty determining information about the audio stream and am unable to feed it into AWS Transcribe. Transcribe does not throw an error, but no real transcription occur. When I set the sample rate to 8000, Transcribe outputs "hmm. hmmm. hmmm." as the transcription, which I only mention to show that some streaming data is being fed into Transcribe and it is returning correctly. However, clearly the audio data is not in the correct format or with incorrect parameters.

Is there an example of unpacking the audio, like the opposite of mkvgen? The relevant Parser code in the lambda is spread across these classes: `https://github.com/amazon-connect/amazon-connect-realtime-transcription/tree/master/src/main/java/com/amazonaws/transcribestreaming`

My goal is to stream KVSProducer -> KVS -> Lambda (with KVS Consumer) -> AWS Transcribe (realtime) endpoint. I can edit this issue title to be more of a question about audio format, rather than just the name?

lherman-cs, 2020-05-27:
@ajhool Do you mind to create a new issue regarding unpacking audio stream and close this? I think there's no more issue about the audio track's name. But, I'd like to preserve this issue in case somebody else also stumbled into the same problem.

lherman-cs, 2020-05-27:
@ajhool Yes, we have a parser library, https://github.com/aws/amazon-kinesis-video-streams-parser-library, it allows you to parse mkv format, which will give you information about the audio sample rate.

ajhool, 2020-05-28:
@lherman-cs, agreed this audio track issue is closed. I would, however, recommend adding the track name to documentation somewhere (if it is not already there or if this is a standard).

closed 2020-05-28 by ajhool

lherman-cs, 2020-05-28:
> @lherman-cs, agreed this audio track issue is closed. I would, however, recommend adding the track name to documentation somewhere (if it is not already there or if this is a standard).

Got it. I'll take a note on this. Thanks!