awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #466: Backwards on stream sending using Docker image

FlorianRuen, 2020-05-27:
Hi KVS Team,

I recently open a ticket on AWS Premium Support and your colleague said that I can get more help using your Github repository.

I'm using the Docker image for Raspberry PI to send video stream to a KVS endpoint with a particular configuration: my RTSP/network camera and internet access arren't on the same network.

I'm explaining.

My camera is connected to my RPI using one RJ45 port (for example IP is 10.10.24.60) and my RPI is connected to internet on a second RJ45 port (for example IP is 192.168.1.60, connected to my router and to internet, ping works perfectly)

But when I'm sending stream, there is some backwards on it. If I put a chronometer on the front of my camera, we can see 1:20:21, 1:20:21, 1:20:22 and 1:20:19 (for example) and it's the same all time long.

So your colleague on my ticket ask me to send some logs. So I add GST_DEBUG:=6 before gst-launch ...

**And the issue doesn't appear with this GST_DEBUG. Very strange solution. To confirm this, I remove it, and the backwards are back on the video stream. So it seems, just added something to log actions, can correct a stream issue ?**

I can confirm that the performance (CPU/RAM) of my RPI isn't very hight when I send the stream (whatever if the GST_DEBUG is set or not). So I don't think the problem came from a lack of performances.

This is a 1 minute clip to see exactly what happen (check the timer at the top of the video, this timer is write by my network camera): https://eyesr-files.s3.eu-central-1.amazonaws.com/5n4pdno1ztfenzcsc5qi4pn8mcyy8xblbv9vkewmrtoaext2kg(1).mp4

Do you have some tips or ideas to correct this ? Because I can't think that GST_DEBUG will correct this all the time, and I think the problem will be back.

Thanks for your help,
Florian.

lherman-cs, 2020-05-27:
@FlorianRuen Hi, I'm sorry for the issue. This is definitely a very strange issue. Do you mind to share your logs after running the pipeline (without GST_DEBUG being set)? Could you also make sure that the encoded video is in the right format? (Probably try adding `video/x-h264, format=avc,alignment=au` in between `rtph264depay` and `h264parse`)

To eliminate some variables, could you also try to feed the video stream to `autovideosink`?

For more context of the issue:

From the information that I received from my colleague, your network camera is configured to send H264 stream with 5FPS using a maximun bitrate of 800 ko/s (with a resolution of 720p).

Following is the pipeline:
`rtspsrc location={current_camera.rtsp_url} short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name={current_camera.remote_id} storage-size=256 access-key={self.access_key} secret-key={self.secret_key} aws-region=eu-central-1 fragment-duration=1000 buffer-duration=10'`.

MushMal, 2020-05-27:
To add to @lherman-cs I would also ask you to give more details about your application and what you asset you are basing your application on. The video you showed doesn't have the clock backward - you simply have out-of-order frames or an extra I-frame at times. Having as much information as possible including GST debug logs (assuming it's gstreamer resources your application is using) will help

FlorianRuen, 2020-05-28:
Hello @MushMal @lherman-cs 

Thanks for your answers

Your colleague on my ticket give me some answers to get the logs, but I can't found them. I'm using the Docker Image you provide.

Can you give me more details on how I can get the log file ? (For your informations, I'm on RPI using Python to run the Docker image with GST pipeline in parameters)

Thanks again for your help and keep safe;

MushMal, 2020-05-28:
@FlorianRuen the logs are the producer SDK side logs.

Are you able to run the samples out-of-the-box?

FlorianRuen, 2020-05-29:
@MushMal Maybe but not with the same configuration, because It seems the SDK can't run on RPI without Docker

MushMal, 2020-05-29:
SDK can run directly on the RPi.

FlorianRuen, 2020-05-29:
@MushMal So If I understand, I need to run the gst launch ... (full pipeline) without docker to get some logs ? I will try it

FlorianRuen, 2020-05-29:
@MushMal Today I try to run the SDK directly on RPI, but there is an error during the Cmake process which is:

`CMake Error at CMakeLists.txt:106 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/eyesr_root/amazon-kinesis-video-streams-producer-sdk-cpp/amazon-kinesis-video-streams-producer-c/build/CMakeFiles/CMakeOutput.log".
See also "/home/eyesr_root/amazon-kinesis-video-streams-producer-sdk-cpp/amazon-kinesis-video-streams-producer-c/build/CMakeFiles/CMakeError.log".
`

MushMal, 2020-05-29:
Hi @FlorianRuen I am not sure yet why CMake fails to build libCurl (it's an open source dependency).

Could you please confirm that you have clean installation directory (new git clone --recursive) and are following the readme?

disa6302, 2020-05-29:
@FlorianRuen , 

Maybe the cmake is not located in the directory find_package() is trying to search in. Can you do a search for this specific cmake file and run cmake with --CMAKE_PREFIX_PATH=</path/to/folder/containing/CURLConfig.cmake>

Also, please ensure you have installed `libcurl4-openssl-dev`.

chehefen, 2020-06-01:
@FlorianRuen can you provide us a clip that comes directly from the RTSP camera? You can generate it by going into the docker container and run 
`
rtspsrc location={current_camera.rtsp_url} short-header=TRUE ! rtph264depay ! h264parse ! matroskamux ! filesink location=test.mkv
`

FlorianRuen, 2020-06-02:
@disa6302 I can confirm that `libcurl4-openssl-dev` is installed on my RPI (also installed on my Ubuntu computer where the error message is the same)

But I can't find one of this files (using locate command, the first has no results, and the second return only files from Bazel, as you can see)

```
/home/florian/.cache/bazel/_bazel_florian/4cc773bd3c97e7d0e87919a234706bed/external/curl/CMake/curl-config.cmake.in
/home/florian/.cache/bazel/_bazel_root/4cc773bd3c97e7d0e87919a234706bed/external/curl/CMake/curl-config.cmake.in
```

@chehefen Already trying to test using directly the SDK, but I can't build your SDK with success

Here is a sample directly from the RTSP camera (on S3 storage): https://eyesr-files.s3.eu-central-1.amazonaws.com/test_video.mkv

disa6302, 2020-06-08:
@FlorianRuen , 

I am not sure why the build is failing for you. One thing I can suggest is remove the existing curl installations in your platform and retry installing curl from scratch.

MushMal, 2020-06-08:
@FlorianRuen have you been able to run natively on RPi? Can you check whether the samples run OK? Can you run the gst pipeline with autovideo sink to eliminate the KVS all together?

From the video/description (as I mentioned before) it seems that your RTSP source produces key-frames which are out-of-order. This points to some issues with the UDP/networking

chehefen, 2020-06-08:
@FlorianRuen I streamed the clip using ./kvs_gstreamer_file_uploader_sample and I didnt see the timestamp go backward. Are you using producer timestamp mode or server timestamp mode when viewing the stream?

FlorianRuen, 2020-06-08:
@chehefen I try both of them. But I can see the same problem when I watch the feed on my application which displays an HLS feed provided by Kinesis Video

chehefen, 2020-06-09:
can you export KVS_DEBUG_DUMP_DATA_FILE_DIR=/path/to/dir, and run the kvs producer? It will save an mkv file there which is exactly what has been sent to kvs. Can you send us that mkv file?

FlorianRuen, 2020-06-10:
Can I do this using Docker image ? (just by adding the export to the docker command), because I've still the issue with C/C++ SDK for now

disa6302, 2020-06-15:
@FlorianRuen , 

You should be able to export in Docker image.

FlorianRuen, 2020-06-15:
@disa6302 It seems not works

I run my app with the KVS_DEBUG_DUMP_DATA_FILE_DIR=test_debug.mkv using my docker container, but there is no way to copy the file from container to host after (the docker cp doesn't work on the docker container with KVS image)

Can you explain the way to do this ?

Thanks in advance,

chehefen, 2020-06-15:
@FlorianRuen  it should be a path to a directory. export KVS_DEBUG_DUMP_DATA_FILE_DIR=/path/to/dir

FlorianRuen, 2020-06-15:
@chehefen Ok I will try, so it will be

- Start a container using the KVS image
- Exec this command inside container
`KVS_DEBUG_DUMP_DATA_FILE_DIR=/ gst-launch-1.0 rtspsrc location={current_camera.rtsp_url} short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name={current_camera.remote_id} storage-size=256 access-key={self.access_key} secret-key={self.secret_key} aws-region=eu-central-1 `

- And I need to make a docker cp to get the video from container to host

It's ok ? **(Maybe you are available for a live chat to speak about this ?)**

chehefen, 2020-06-16:
yes this is ok.

FlorianRuen, 2020-06-16:
@chehefen Tonight I tried something as we discussed:

- I try with 2 networks (my camera not on the same network that the computer which send the video to KVS, with the export KVS_DEBUG ...), and the problem doesn't appear (I have the video file, but I don't think it's necessary to send to you because there is nothing strange)

- But when I try the exact same configuration using my application (which just get data from an RDS, build the pipeline and launch it in a docker container), **the backwards/loops appear on KVS !**

So I try the second point with the KVS_DEBUG, and nothing seem happenning (no streams are present on Kinesis Dashboard, and there is no network bandwidth used, so maybe the stream isn't send), so in this case I can't grab the video file as expected.

Tomorrow, I will try to use a container created by my application to run the command directly inside, but it use the same image that the container we used together, the exact same command with the same parameters for the pipeline.

MushMal, 2020-06-26:
@FlorianRuen is there any update on this issue? From what I understand this is an issue with the media pipeline you have?

FlorianRuen, 2020-06-26:
@MushMal Yes it is, as I can said, when I run the pipeline with my application loops appear, but loops doesn't appear when I run the command directly in container. But it's exactly the same command i run using my app or the command inside the container

I don't take more time to make more test on it (due to others things to do), but I will try to go deep into this case on the next week, even if I don't understand very well what happen, there is no update on KVS side that can cause this ? (on docker image or something else ?)

MushMal, 2020-06-26:
Thanks for the update Florian. I don't see this particular thread serving more purpose. I am resolving this issue. Please cut a new issue with more details as you discover those.

closed 2020-06-26 by MushMal

FlorianRuen, 2020-07-09:
@MushMal Hi there, I'm comming back to you, because I retry to found the problem on this case. After many investigations, I try to pull your docker image again, and to keep exactly the same pipeline that work when camera and internet are the same network, there is still backward when camera and internet aren't the same networks.

I run this inside a docker container directly (without using my app, to try if the problem came from this application), but the problem still the same.

So I can't be sure the problem is from the pipeline or from the docker image that you provide. You haven't any way to try this configuration on your side ?

Thanks again