awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #467: Cannot view the streamed MP4 file in the console.

ravidetroit, 2020-05-28:
I have built the c++ kinesis video producer. And i can stream a mp4 file. However i cannot playback the media in the console. I get a no media found error.

MushMal, 2020-05-28:
Please include more information on what the experience is? Are you seeing black screen? Are you using the same region? Can you include debug producer logs

ravidetroit, 2020-05-28:
[kvs.log.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4697203/kvs.log.zip)

I get a no media found error.

MushMal, 2020-05-28:
@ravidetroit looks like the file is uploaded just fine. If the file is "short" it will finish before you get the playback. Try to stream, go to the console. Make sure you select the same region. Select server-side timestamp. Select from some time past before the time you started streaming for the playback to pick up the fragments that langed.

ravidetroit, 2020-05-28:
[output.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4697653/output.zip)

@MushMal, attached the output of get-inspect-1.0 kvssink. When i execute 
./build/kvs_gstreamer_file_uploader_sample /home/ubuntu/environment/samplevideos/60.mp4

 Failed to construct pipeline: no element "qtdemux"
[ERROR] kvs_gstreamer_file_uploader_sample: Failed to initialize gstreamer pipeline. Have you set GST_PLUGIN_PATH properly?

     For example: export GST_PLUGIN_PATH=<YourSdkFolderPath>/build:$GST_PLUGIN_PATH

echo $GST_PLUGIN_PATH
/home/ubuntu/environment/amazon-kinesis-video-streams-producer-sdk-cpp/build

MushMal, 2020-05-28:
Seems that it's unable to find the plugins. Could you execute

```
export GST_PLUGIN_PATH=`pwd`/build
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
```

from the root directory to set the correct paths?

ravidetroit, 2020-05-28:
echo $GST_PLUGIN_PATH
/home/ubuntu/environment/amazon-kinesis-video-streams-producer-sdk-cpp/build, the path variables are set
echo $LD_LIBRARY_PATH
/home/ubuntu/environment/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib

MushMal, 2020-05-28:
Looks correct. Seems that the demux plugin is missing for some reason.

ravidetroit, 2020-05-28:
Is that a shared library, i can include that explicitly

Sean-Der, 2020-05-28:
Hey @ravidetroit 

You also need to install the GStreamer 'good plugins'. If you are using homebrew it would be `brew install gst-plugins-good` and for Debian/Ubuntu it would be `apt-get install gstreamer1.0-plugins-good`

ravidetroit, 2020-05-28:
I now get the following error. Building gstreamer pipeline
[ERROR] kvs_gstreamer_file_uploader_sample: Failed to construct pipeline: no element "h264parse"

MushMal, 2020-05-28:
This is in bad plugins https://gstreamer.freedesktop.org/documentation/videoparsersbad/h264parse.html?gi-language=c

Try to install that one too 

`apt-get install gstreamer1.0-plugins-bad`

I am guessing that you already have the base plugins installed but give this a try. I think for file uploader this should be it - you wouldn't require plugins-ugly

ravidetroit, 2020-05-28:
I installed gst-plugins-bad and the errors are now different

log4cplus:ERROR could not open file ./kvs_log_configuration
ERROR - Unable to create Rotating Credential provider. Error status: 0x9
[ERROR] kvs_gstreamer_file_uploader_sample: Unable to set the pipeline to the playing state.

MushMal, 2020-05-28:
First issue is due to log4cpus config file not being there or the path not being there. Perhaps, copy  the config file from the samples to build directory

The second is complaining about the auth integration. Make sure you set the auth as described in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/linux.md

chehefen, 2020-06-01:
This issue has been resolved after working with customer offline.

closed 2020-06-01 by chehefen