awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #469: Gstreamer failing for Windows producer

aeperaltan, 2020-05-29:
Hello,

We have followed the following steps to build the library:
call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars64.bat"
mkdir build64
cd build64
set PKG_CONFIG_PATH=C:\gstreamer\1.0\x86_64\lib\pkgconfig
cmd.exe /c cmake -G "NMake Makefiles" ..
cmake -G "NMake Makefiles" -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..
nmake

Please notice that in order for the compilation process for X64 on Debug to finish correctly, we needed to include some DLLs generated in a different folder: "open-source\local\bin". Specifically, the files that we  had to copy were "libcurl-d.dll" and "log4cplusD.dll". After copying both files to the build folder the samples could be called without any exception. 

At this point,  we have detected that the gstreamer pipeline crashes using the default settings. You can find attached the log file called 'win64 run test audio_video_sample.txt'

I would appreciate your help on this issue. Thanks!
[win64 run test audio_video_sample.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4702770/win64.run.test.audio_video_sample.txt)

MushMal, 2020-05-29:
@aeperaltan I am not sure why your GStreamer errors out.

(kvs_gstreamer_audio_video_sample.exe:12472): GLib-GObject-CRITICAL **: 15:37:02.800: g_object_set: assertion 'G_IS_OBJECT (object)' failed

You will need to get some debug logs from GStreamer by setting GST_DEBUG=4 environment variable.

JuanCarlos-Arce, 2020-06-01:
Hi @MushMal ,

I have tried this situation with GST_DEBUG=4 and the errors seem to be related with some of the elements created for the pipeline.

Here is the log file:

[gstreamer_run_error.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4710654/gstreamer_run_error.txt)

I have installed the development version of gstreamer 1.16.2

Hope it helps.

MushMal, 2020-06-01:
@JuanCarlos-Arce you need to install plugin base as well as some of the plugins like good and bad

https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gstreamer-plugins/html/gstreamer-plugins-capsfilter.html

JuanCarlos-Arce, 2020-06-01:
Hi @MushMal 

I have reinstalled gstreamer for windows from https://gstreamer.freedesktop.org/download/ with a complete installation but these plugins are not available. Is there any external package that we need to install after gstreamer ?

Sorry if it's an obvious question, but I haven't seen any documentation about how to install those plugins.

Sean-Der, 2020-06-01:
Hey @JuanCarlos-Arce can you re-run both installers and choose 'Full Installation'

I will update the docs to call that out also!

MushMal, 2020-06-01:
I believe you need to download the MSI executable for Windows (dev) and should choose an option of install all. @Sean-Der bit me to it!

JuanCarlos-Arce, 2020-06-02:
Hi @Sean-Der and @MushMal .

I have installed again both installers with "full installation" and after building again the library, the results are the same. I have used gstreamer v1.16.2.

I don't know if it's related to this issue, but for this test I have set the environment variable "GSL_PLUGIN_PATH" for the folder "C:\gstreamer\1.0\x86_64\lib\gstreamer-1.0"

aeperaltan, 2020-06-08:
Please provide guidance on how to proceed. Thanks!!

MushMal, 2020-06-08:
OK, what's the latest on this issue? 

* Has the GStreamer for Windows been installed? Recommend installing the latest dev with FULL installation as that's what's installing the plugins too
* Set the GST_PLUGIN_PATH 
* The gstreamer_run_error.txt contains a single failure of not finding capsfilter filter. This points to the plugins and plugins base not being installed.

While I understand that this is a pressing issue, we are also trying to find some remedies from GStreamer forums when something fails. Generally, on our Windows installation we can't reproduce some of the install issues.

JuanCarlos-Arce, 2020-06-09:
Hi @MushMal ,

At the moment I have reinstalled both installers with "Full Installation" and I have several environment variables pointing to GStreamer folders:

- GST_PLUGIN_PATH: C:\gstreamer\1.0\x86_64\lib\gstreamer-1.0
- GSTREAMER_1_0_ROOT_X86_64: C:\gstreamer\1.0\x86_64\
- GSTREAMER_INCDIR: C:\gstreamer\1.0\x86_64\include\gstreamer-1.0
- GSTREAMER_LIBDIR: C:\gstreamer\1.0\x86_64\lib

And I have generated a complete list with the files installed on gstreamer folder, maybe you can check if there is any difference between your installation and ours.

[gstreamer_files.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4750508/gstreamer_files.txt)

MushMal, 2020-06-09:
@JuanCarlos-Arce I don't have a Windows machine handy right now to check. Meanwhile, could you please take a look at the way the TravisCI runs install Windows specific GStreamer: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.travis.yml#L107-L115

The build can provide some more insights: 
https://travis-ci.com/github/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/jobs/343049270#L52-L139

disa6302, 2020-06-15:
@aeperaltan , @JuanCarlos-Arce , 

What is the latest on the issue?

JuanCarlos-Arce, 2020-06-15:
Hi @disa6302 , @MushMal ,

I have installed gstreamer from chocolatey as your CI machine does and I have updated the environment variable GST_PLUGIN_PATH to point the build folder as the documentation in [here ](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html) suggests, and the compilation process continues working fine.

I also have checked the information that @MushMal has sent me about the travis-ci configuration and I have checked that all the tests run by the CI-Machine fail and throw errors.
Also the CI machine compiles the example **kvs_gstreamer_audio_video_sample** but it doesn't run it to check if it works or not ( it's the one that I have used to test the library )

Please, could you review if this example runs correctly on this CI Machine ?

MushMal, 2020-06-15:
@JuanCarlos-Arce if you are getting the same error then something is odd with the pipeline. Could you perhaps try gst-launch with your pipeline but without kvs in it. You might want to stabilize the GStreamer pipeline first. Sorry, I am unable to reproduce this locally.

JuanCarlos-Arce, 2020-06-17:
Hi @MushMal ,

I have found the issue with GStreamer, the library generates a bin file with the register of all the available plugins, it seems that there was an issue with the first installation and I had to erase that file to generate it again with the complete installation and now all the plugins works as expected.

On the another hand, as I wrote previously, I have checked that your CI Machine is throwing errors on the CMake configuration step.

https://travis-ci.com/github/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/jobs/343049270#L52-L139

Thanks

MushMal, 2020-06-17:
@JuanCarlos-Arce this particular issue was opened to track build issues on the Windows platform which seem to be OK for you. I am not sure what the TravisCI issues you are having - perhaps this is something on your environment?

JuanCarlos-Arce, 2020-06-18:
Hi @MushMal,

Please, check that the URL that I have posted is your own travis machine. You are having the same building issue on your Windows machine since a long time. The building process is failing with several compilation and linking errors but your TravisCI Machine returns that the process has been completed correctly.

MushMal, 2020-06-18:
I have the following in the link you've provided:

$ choco install nasm strawberryperl gstreamer
Chocolatey v0.10.15
Installing the following packages:
nasm;strawberryperl;gstreamer
By installing you accept licenses for the packages.
Progress: Downloading nasm 2.14.02... 100%
nasm v2.14.02 [Approved]
nasm package files install completed. Performing other installation steps.
Downloading nasm 64 bit
  from 'https://www.nasm.us/pub/nasm/releasebuilds/2.14.02/win64/nasm-2.14.02-installer-x64.exe'
Progress: 100% - Completed download of C:\Users\travis\AppData\Local\Temp\chocolatey\nasm\2.14.02\nasm-2.14.02-installer-x64.exe (933.77 KB).
Download of nasm-2.14.02-installer-x64.exe (933.77 KB) completed.
Hashes match.
Installing nasm...
nasm has been installed.
 The install of nasm was successful.
  Software installed as 'exe', install location is likely default.
Progress: Downloading StrawberryPerl 5.30.2.1... 100%
StrawberryPerl v5.30.2.1 [Approved]
strawberryperl package files install completed. Performing other installation steps.
Downloading strawberryperl 64 bit
  from 'http://strawberryperl.com/download/5.30.2.1/strawberry-perl-5.30.2.1-64bit.msi'
Progress: 100% - Completed download of C:\Users\travis\AppData\Local\Temp\chocolatey\StrawberryPerl\5.30.2.1\strawberry-perl-5.30.2.1-64bit.msi (101.61 MB).
Download of strawberry-perl-5.30.2.1-64bit.msi (101.61 MB) completed.
Hashes match.
Installing strawberryperl...
strawberryperl has been installed.
  strawberryperl may be able to be automatically uninstalled.
Environment Vars (like PATH) have changed. Close/reopen your shell to
 see the changes (or in powershell/cmd.exe just type `refreshenv`).
 The install of strawberryperl was successful.
  Software installed to 'C:\Strawberry\'
Progress: Downloading gstreamer 1.16.2.20200527... 100%
gstreamer v1.16.2.20200527 [Approved]
gstreamer package files install completed. Performing other installation steps.
Downloading gstreamer 64 bit
  from 'https://gstreamer.freedesktop.org/data/pkg/windows/1.16.2/gstreamer-1.0-msvc-x86_64-1.16.2.msi'
Progress: 100% - Completed download of C:\Users\travis\AppData\Local\Temp\chocolatey\gstreamer\1.16.2.20200527\gstreamer-1.0-msvc-x86_64-1.16.2.msi (91.27 MB).
Download of gstreamer-1.0-msvc-x86_64-1.16.2.msi (91.27 MB) completed.
Hashes match.
Installing gstreamer...
gstreamer has been installed.
PATH environment variable does not have %GSTREAMER_1_0_ROOT_X86_64%\bin in it. Adding...
  gstreamer may be able to be automatically uninstalled.
Environment Vars (like PATH) have changed. Close/reopen your shell to
 see the changes (or in powershell/cmd.exe just type `refreshenv`).
 The install of gstreamer was successful.

Perhaps we are talking about something that was broken prior or something? Do you have the latest commits? Clean?

JuanCarlos-Arce, 2020-06-18:
Hi @MushMal ,

Please, scroll down on the same page until the building process are run (line 1334)  and you can check that the windows platform on your CI Machine throws errors like:

`LINK : fatal error LNK1181: cannot open input file 'crypto\aes\aes_cbc.obj'
-- Configuring incomplete, errors occurred!
See also "C:/Users/travis/build/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
-- Kinesis Video PIC path is C:/Users/travis/build/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-pic/
-- Kinesis Video C Producer path is C:/Users/travis/build/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/amazon-kinesis-video-streams-producer-c/
-- Kinesis Video WebRTC Client path is C:/Users/travis/build/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is C:/Users/travis/build/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/travis/build/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus
Scanning dependencies of target project_log4cplus
[ 11%] Creating directories for 'project_log4cplus'
[ 22%] Performing download step (git clone) for 'project_log4cplus'
NMAKE : fatal error U1077: 'link' : return code '0x1'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: 'echo' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
CMake Error at open-source/amazon-kinesis-video-streams-producer-c/CMake/Utilities.cmake:59 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  open-source/amazon-kinesis-video-streams-producer-c/CMakeLists.txt:78 (build_dependency)
`

There are more errors reported and all the tests fail because the binaries are not created, but it seems that the CI machine doesn't detect this errors in the report resume.

MushMal, 2020-06-18:
Thanks, indeed, just saw this linker error. I am now wondering how the final library gets linked together. We will certainly take a deeper look at this.

MushMal, 2020-06-18:
NOTE to the reader (as this thread is quite long). Seems that the OpenSSL crypto library is failing to link.

`LINK : fatal error LNK1181: cannot open input file 'crypto\aes\aes_cbc.obj'`

* Need to understand what's causing OpenSSL build to fail on Windows platform
* Need to understand how we end up with linking the c producer library at the end.

aeperaltan, 2020-06-26:
Should I expect an answer or any help on this? I opened a support case and was instructed to use GitHub... used GitHub and nothing!!!

MushMal, 2020-06-26:
@aeperaltan as these not assets owned by us (these are third party open-source components that fail building on Windows) we can't give any answers. Will update ASAP - the builds are running now

MushMal, 2020-10-09:
Can you give this a try? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/584/commits/8c17d27f2d7d6b47a70d489e76bf1124eb8a019c

MushMal, 2020-10-10:
Resolving due to inactivity

closed 2020-10-10 by MushMal