awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #470: Compilation errors on Release for Windows

JuanCarlos-Arce, 2020-06-01:
Hi,

I have tried to compile the library on Release mode and I have found that there are some compiling errors.

The steps that I have done to build the library on Release are:

`
mkdir build
cd build
set PKG_CONFIG_PATH=C:\gstreamer\1.0\x86_64\lib\pkgconfig
cmake -G "NMake Makefiles" -DCMAKE_BUILD_TYPE=Release ..
nmake
`

You can find attached the complete log file.

Hope it helps

[kinesis_x64_release.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4710191/kinesis_x64_release.txt)

Sean-Der, 2020-06-01:
Hey @JuanCarlos-Arce 

It looks like the test suite is failing to build, can you also add `-DBUILD_TEST=FALSE` to the CMake command and try again?

JuanCarlos-Arce, 2020-06-01:
Hi,

I have tried again with this command:

`cmake -G "NMake Makefiles" -DBUILD_TEST=FALSE -DBUILD_GSTREAMER_PLUGIN=TRUE -DCMAKE_BUILD_TYPE=Release ..`

And the compiling errors continue. Here is the complete log with the tests disabled.

[kinesis_release2.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4710439/kinesis_release2.txt)

It seems that there are compiling errors on the libopenssl project, but I don't know if they are related to the library itself or how we try to compile it with these flags.

disa6302, 2020-06-15:
@JuanCarlos-Arce , 

Can you try what is posted here?

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/461#issuecomment-632044826

JuanCarlos-Arce, 2020-06-15:
Hi @disa6302 ,

I have followed that thread to make the library built on debug ( the default configuration ), but it seems that the release version needs anything else on the cmake files.

MushMal, 2020-06-17:
@JuanCarlos-Arce seems that you are reporting an unrelated issue with openSSL. We had had a few stability issues with OpenSSL (NOTE: We don't own that component) which clean build resolves. Make sure you start fresh or use `git clean -ddfx` command. There have been reports that the cmake fails with some build race condition on open ssl initially. You should try to run it consequently. Also, try to also seek some help from open ssl forums.

mggunman, 2020-06-19:
So I'm getting the same issue as stated in the previous thread and I'm stuck at the same point as many others who have tried to build this SDK on windows.
I've installed git, chocolately, nasm, strawberryperl, gstreamer both versions, visual studio 2019+its dev tools, openssl, pkg-config, mingw, msvs2 and cmake and many more - I've installed this same SDK into a Ubuntu docker container and it worked flawlessly the only issue is docker connecting to the webcam so I've reverted back to this.
I'm doing this for work and working on a work machine, I've got the whole AWS side working and the kinesis stream just needs a video stream/data stream to complete it.
I've attached the error as a snippet of CMD and also a snippet of my environment variable PATH.
----------
[image]
----------
[image]

MushMal, 2020-06-19:
Apologies for this issue, we are trying to reproduce and come up with some solution - no ETA yet.

Meanwhile, search the issues - there are some questions asked about USB camera access under docker.

mggunman, 2020-06-20:
@MushMal,
I've tried the docker version and it builds and runs perfectly both the Java and C++ producer SDK versions, the only issue is passing through the built-in webcam from the laptop running windows to the docker container running on ubuntu - I've searched around but there is no real decent method, its either build and install a bunch of programs and create another container with it or your main machine should be Linux based.

From what I can take from the issue and the previous issues faced is that I either have not set up my build environment correctly or I might have missed a few environment variables or GStreamer is not correctly installed or the path links in cmakelist are pointing to the wrong places/missing.

MushMal, 2020-06-22:
@mggunman understood, indeed, it will be very hard to expose the hardware resources such as webcam through a container on a vastly different OS-es.

We are trying to reproduce the Windows build issue to try to see how we can get some remedy for the 3rd party open-source components that we rely on. Will update as soon as we get more info

mggunman, 2020-06-22:
@MushMal,
To reproduce the errors I would install a fresh copy of windows, then install visual studio 2019 + CMake dev tools, git, chocolately, nasm, strawberryperl, GStreamer and try from there. But I also downloaded CMake separately and OpenSSL and I've been using the NASM cli.

aeperaltan, 2020-06-26:
@MushMal, we have tried to make this work for a long time and you are not helping. Please advise!

MushMal, 2020-06-26:
@aeperaltan these are third party components which have nothing to do with KVS. As you see, we are interested in supporting Windows platform in general and are trying to understand what causes these build issues. No updates yet

closed 2020-10-10 by MushMal