awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #471: Build the library for x86

JuanCarlos-Arce, 2020-06-01:
Hi,

I have tried to build the library for Win32. To do that I'm executing these commands.

`call "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvars32.bat"`
`mkdir build32`
`cd build32`
`set PKG_CONFIG_PATH=C:\gstreamer\1.0\x86_64\lib\pkgconfig`
`cmake -G "NMake Makefiles" -A Win32 -DBUILD_TEST=TRUE -DBUILD_GSTREAMER_PLUGIN=TRUE ..`
`nmake`

And cmake returns this error:
`CMake Error at CMakeLists.txt:8 (project):
  Generator
    NMake Makefiles
  does not support platform specification, but platform
    Win32
  was specified`

If I remove the "-A Win32" parameter, but I set the x86 compilers cmake works for a while, but the third party libraries will try to be compiled for x64 platforms. Here is the build log

[kinesis_x86.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4710753/kinesis_x86.txt)

Please, could you help me to find how can I build the library for these platforms ?

Thanks

lherman-cs, 2020-06-01:
Do you mind to try to change `VC-WIN64A` to `VC-WIN32`, https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMake/Dependencies/libopenssl-CMakeLists.txt#L7?

JuanCarlos-Arce, 2020-06-01:
Hi @lherman-cs ,

I have tried changing the suggested file, but it doesn't fix the issue. Here is the log file for that compilation.

[kinesis_win32_2.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/4712782/kinesis_win32_2.txt)

JuanCarlos-Arce, 2020-06-17:
Hi @lherman-cs,

Kindly Reminder.

lherman-cs, 2020-06-17:
@JuanCarlos-Arce Do you mind to resend the log in English? I missed some of the context. Thanks

MushMal, 2020-06-17:
@JuanCarlos-Arce this is exactly the same issue as I commented on https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/470#issuecomment-645659607

Sorry that this is hard going but we don't own these projects and as such it's hard to debug building them with different configs on various platforms.

JuanCarlos-Arce, 2020-06-18:
Hi @MushMal ,

As I have seen on the cmake files.... the issue is not related with OpenSSL library itself. The problem is the current CMake files that doesn't support x86 platforms because they set x64 by default ( as @lherman-cs said previously in this thread). 

In my view, to support several platforms, some of the cmake files must check the selected platform from the command line arguments and configure the third-party projects with it.

MushMal, 2020-06-18:
Oh, sorry, I was confused. Yes, this does seem to be a missing feature. I am labeling this as an enhancement. This should be addressed.

aeperaltan, 2020-06-26:
@MushMal, can we get an update on this?

MushMal, 2020-10-10:
This is a dup of #446

closed 2020-10-10 by MushMal