awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #473: LibCurl compiles in debug mode

lpawlaczyk, 2020-06-02:
LibCur should compile in Release mode by default.
To fix it you can add: 
		-DCMAKE_BUILD_TYPE=Release 
to libcurl-CMakeLists.txt

MushMal, 2020-06-02:
@lpawlaczyk thanks for opening this issue. Sounds like by default it builds debug which is rather odd. We should also check the rest of the open source components.

Meanwhile, we are welcoming any contribution. Do you mind submitting a PR? All you need is to is create a branch, make the changes, push the branch upstream and it will give you a link that you can use to submit the PR. Let us know

disa6302, 2020-06-09:
@lpawlaczyk , 

The libcurl will now compile in release mode. Fixed it in PR #487 . Feel free to reach out if you have any questions!

closed 2020-06-09 by disa6302